espressif/esp-idf #13580: Getting compiler errors with snmp.h

0xFEEDC0DE64, 2024-04-10:
Using latest master of esp-idf,

I get compiler errors when including `lwip/apps/snmp.h`:

```
<firmware-path>/esp-idf/components/lwip/include/apps/esp_sntp.h:287:9: error: 'uint8_t sntp_getreachability(uint8_t)' was declared 'extern' and later 'static' [-fpermissive]
  287 | uint8_t sntp_getreachability(uint8_t idx)
      |         ^~~~~~~~~~~~~~~~~~~~
<firmware-path>/esp-idf/components/lwip/lwip/src/include/lwip/apps/sntp.h:62:6: note: previous declaration of 'u8_t sntp_getreachability(u8_t)'
   62 | u8_t sntp_getreachability(u8_t idx);
      |      ^~~~~~~~~~~~~~~~~~~~
<firmware-path>/esp-idf/components/lwip/include/apps/esp_sntp.h:293:26: error: conflicting declaration of C function 'esp_sntp_operatingmode_t sntp_getoperatingmode()'
  293 | esp_sntp_operatingmode_t sntp_getoperatingmode(void)
      |                          ^~~~~~~~~~~~~~~~~~~~~
<firmware-path>/esp-idf/components/lwip/lwip/src/include/lwip/apps/sntp.h:52:6: note: previous declaration 'u8_t sntp_getoperatingmode()'
   52 | u8_t sntp_getoperatingmode(void);
      |      ^~~~~~~~~~~~~~~~~~~~~
```

david-cermak, 2024-04-10:
Hi @0xFEEDC0DE64 

This is probably a regression from c30f2825d5 (added a forwarding layer for some potentially thread unsafe lwip API to espressif wrappers).
This is probably causing trouble when building your project with some custom lwip features (not selected by Kconfig options), I wasn't able to reproduce it, though. I'd like to know how you're building and enabling the SNMP module.

We protect this compat layer with `ESP_LWIP_COMPONENT_BUILD` macro, so it's not taken into account when building lwip component. I guess you can workaround this issue by defining this macro for the component using SNMP (or adding the SNMP build to the lwip library).
Adding this line to the component makefile (maybe `main` component?) may fix the issue:
```cmake
    target_compile_definitions(${COMPONENT_LIB} PRIVATE ESP_LWIP_COMPONENT_BUILD)
```

david-cermak, 2024-05-29:
@0xFEEDC0DE64 Could you please share more details on how you're building your project with `snmp` module?

this seems unrelated to c30f2825d554dfe25afa450bb6aaa3aeb58782ef since you're also getting errors on `sntp_getoperatingmode()` API

david-cermak, 2024-10-30:
Closing due to short of feedback, feel free to reopen.

closed 2024-10-30 by david-cermak

CircuitGuy, 2025-03-23:
I've gone down the rabbit hole on this for a few days and have been really lost. This is the most relevant chat I've seen on this.

OP's original problem are solved by adding something like this after the idf_component_register (or probably in it would work too).
```
idf_build_set_property(COMPILE_DEFINITIONS
    "LWIP_SNMP=1"
    "MIB2_STATS=1"
    "SNMP_USE_NETCONN=1"
    "SNMP_LWIP_MIB2=1"
    APPEND)
```

I'm still lost in linker purgatory after enabling this. This allows standalone .c's to build, but they fail with linker errors 
 like `(.text.initialize_snmp+0x20): undefined reference to `snmp_mib2_set_syscontact'` after main() references them.

I can probably boil down a simple reproduction case if needed, but an example of "enable LWIP SNMP" would go a long way. I'm still stumbling through this, and so will update when I figure it out.

CircuitGuy, 2025-03-23:
Solved! I think.
I couldn't figure out the idf tools that let me do overrides. After digging into the component though; it's clear that ESP is neglecting the LWIP apps.
 
So there's two parts to my solution.
1. Add the apps/snmp/ source and header files into idf_componenent_register
2. Override the build flags outside of esp-idf's control with a set(CMAKE_C_FLAGS...) in cmake.
3. <Other> Ignore LWIP's and some of the other google search results that talk about a local lwiopts.h file. Of course you can override other settings in that or another file, but as far as I can tell, ESP's component 'wins' by compiling first unless you override with (1).

This snippet of CMakeLists.txt does both of those.

```
#Add LWIP as a source
# Glob all *.c files under the lwip api directory
file(GLOB_RECURSE SNMP_SOURCES "${IDF_PATH}/components/lwip/lwip/src/apps/snmp/*.c" )

idf_component_register(
    SRCS ${app_sources} ${SNMP_SOURCES} 
    INCLUDE_DIRS "." "${IDF_PATH}/components/lwip/lwip/src/apps/snmp/"
)

#LWIP Options override
set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -DMIB2_STATS=1 -DLWIP_SNMP=1 -DSNMP_USE_RAW=0 -DSNMP_USE_NETCONN=1 -DSNMP_LWIP_MIB2=1")
```

Would be a nice-to-have feature request if menuconfig had snmp as a build option.

chegewara, 2025-03-24:
Hi,
i probably dont understand the nature of this issue, but i can use sntp client without any problems or extra steps.
This is just a part of my code to show which API i am using:
```
#include "esp_sntp.h"
#include <time.h>
#include <sys/time.h>

...
    sntp_set_time_sync_notification_cb(time_sync_notification_cb);
    sntp_setoperatingmode(SNTP_OPMODE_POLL);
    if(s)
    {
        server = std::string(s);
        ip_addr_t server_ip;
        ip4_addr_set_u32(&server_ip.u_addr.ip4, ipaddr_addr(server.c_str()));
        sntp_setserver(0, &server_ip);
    } else {
        sntp_setservername(0, server.c_str());
    }
    sntp_init();
```

I hope this helps anyone to solve the problem.

CircuitGuy, 2025-03-24:
> Hi, i probably dont understand the nature of this issue, but i can use sntp client without any problems or extra steps. This is just a part of my code to show which API i am using:

> #include "esp_sntp.h"

That's `sntp` not `snmp` ;)

P.S. - for any others, I just kicked the can to the next problem, which is the LWIP apps aren't ESP thread-safe. I found the docs with the limitation, but I'm not seeing how to work around it.

chegewara, 2025-03-24:
Ok, my bad.
I was misleaded by this in OP
```
<firmware-path>/esp-idf/components/lwip/lwip/src/include/lwip/apps/sntp.h:52:6: note: previous declaration 'u8_t sntp_getoperatingmode()'
   52 | u8_t sntp_getoperatingmode(void);
```
Sorry

CircuitGuy, 2025-03-24:
Sorry, I too messed up. The thread title is snmp; and you absolutely will get the same type of errors out of snmp; but OP is clearly showing sntp errors. There are a few unsolved threads and github issues floating around.

FWIW - yes, *sntp* works well.

Thanks!

CircuitGuy, 2025-03-25:
Just to close this thread and hopefully help someone out. The workaround above is holding.

https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/lwip.html
ESP-IDF docs state that officially something like this:
```
idf_component_get_property(lwip lwip COMPONENT_LIB)
target_compile_definitions(${lwip} PRIVATE "-DETHARP_SUPPORT_VLAN=1")
```

can be used to modify LWIP. For whatever reason, I couldn't get that to work. I'm sure it's my issue, but it seems to be an order of operations issue. The CMake hammer above worked just fine.
 
I'm still having some issues, and might open another bug report or PR, but I can't figure out if it's LWiP or ESP's modifications to it that are the problem. SNMP *works* technically, but I'm getting a kernel panic if I request certain invalid OIDs. As far as I can tell, the ESP options for LWiP are configured correctly.

Again, hopefully that's helpful for anybody in the future just because there's not much that comes up on search engines, and this issue thread is highly ranked.

david-cermak, 2025-03-25:
> can be used to modify LWIP. For whatever reason, I couldn't get that to work.

Maybe you'd need to use `PUBLIC` instead of `PRIVATE` type? you'd also need to add the relevant sources (either to the lwip lib, or any other and define a link dependency)

But please open a feature request for SNMP (a new issue, this one's already mixing sntp and thread safety).

Do not see much demand for SNMP, but maybe others would comment and upvote..? Already had this PR some time ago: https://github.com/espressif/esp-idf/pull/5155, but couldn't accept it