espressif/esp-idf #13581: OTA update via Android

edbek, 2024-04-10:
### Is your feature request related to a problem?

Is it possible to OTA update my software in new, just purchased modules (for example, ESP32-C3-MINI) via BLE or WiFi, as is done by NORDIC via Android?
So that there is no need to specially configure such modules via AT commands or pull their pins.

CharlesMcBaker, 2024-04-13:
No you need firmware on the devices to support updates, so you would need to flash by wire fist which defeats the purpose of updating later on with OTA.

OTA update via BLE is very very slow.

mickeyl, 2024-04-16:
FWIW, OTA via BLE is only slow, if you're going the GATT route. If you transfer the data via L2CAP COC, it's pretty fast. (Yes, WiFi will always be faster, but less convenient).

hmalpani, 2024-04-22:
Hello @edbek 
OTA is possible in ESP-IDF via [Wi-Fi](https://github.com/espressif/esp-idf/tree/master/examples/system/ota) and [BLE](https://components.espressif.com/components/espressif/ble_ota) both. But if I understand properly, you want to perform OTA on a newly purchased devkit. A devkit should have a firmware which will be able to perform OTA. I think you need to flash a firmware which will be able to perform OTA and then you can update the firmware via BLE of Wi-Fi.

hmalpani, 2024-05-07:
Hello @edbek 
As mentioned before, I would recommend you to flash a firmware which is capable of performing OTA on the new devkit.
Closing this issue. Please reopen or you can also post this on [esp32 forum](https://www.esp32.com/viewforum.php?f=13)

closed 2024-05-07 by hmalpani