espressif/esp-idf #13582: Bluetooth Mesh node reception issue

sonwa, 2024-04-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Using the 'onoff_server' example, I have set up two nodes, A and B. Both nodes are configured to be part of the same network via an app. The app controls the switch on node A. How can I ensure that node B prints out the control information from node A? It seems that the official implementation filters out the element_address, so if it's not the current node's control element_address, it won't output any information. I have registered both `esp_ble_mesh_register_generic_client_callback` and `esp_ble_mesh_register_generic_server_callback`, but when I control node A, node B doesn't output any control information. My goal is to have node B print out the control information (element_address, functionality operation, etc.) whenever node A is controlled.

forx157, 2024-08-01:
Hi, @sonwa 
Sorry for the long delay in getting back to you and many thanks for being patient.
Perhaps you can put Node A and Node B into the same group address, which allows Node A to receive a message while Node B also receives the message, and then output the information.

sonwa, 2024-08-01:
Hi, @forx157 
To join the same group address, it needs to be controlled through the group address, right? But I want to achieve controlling a single device address, and have another device receive and print the message.

forx157, 2024-08-01:
@sonwa 
Why not let A send the message to B after A receives the message in the first place?

sonwa, 2024-08-07:
@forx157 
For example, address A is 0x2 and address B is 0x3. When I send a switch command to A(0x2) through the app, B (0x3) should also receive this information, right? It's just that at the lower level, a match is made with A(0x2) to execute the action. The device at B (0x3) determines that the address is not A(0x2), so it does not execute. But can B (0x3) log this operation?

forx157, 2025-04-07:
Hi, @sonwa 
Sorry for the long delay in getting back to you and many thanks for being patient.
you could replaced `https://gitlab.espressif.cn:6688/espressif/esp-idf/-/blob/master/components/bt/esp_ble_mesh/core/net.c?ref_type=heads#L1947` to
```
    printf(
           "\nNetRecv: ctl: %d, src: %d, dst: %d, ttl: %d, data: 0x%s",
           rx->ctl, rx->ctx.addr, rx->ctx.recv_dst, rx->ctx.recv_ttl,
           bt_hex(buf.data + BLE_MESH_NET_HDR_LEN, buf.len - BLE_MESH_NET_HDR_LEN));

```c

closed 2025-06-30 by Alvin1Zhang