espressif/esp-idf #13584: scan ap list error when mqtt is connect

gl-chenxiaosheng, 2024-04-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

IDF: release v5.1.2
module: esp32s3

I use wifi connect to network. And the mqtt client is be created to pub/sub. When i use `esp_wifi_scan_start(&scan_cfg, false); ` to scan the ap list, sometimes have errors as follow.（The probability is very high）.
It will disconnect the wifi and reconnect it. What is wrong with it?
```c
I (751655) wifi:bcn_timeout,ap_probe_send_start
W (757215) transport_base: Poll timeout or error, errno=Success, fd=48, timeout_ms=10000
E (757215) mqtt_client: Writing didn't complete in specified timeout: errno=0
I (757215) gl_mqtt_cfg: MQTT_EVENT_DISCONNECTED
W (757225) mqtt_client: Publish: Losing qos0 data when client not connected
I (757385) wifi:bcn_timeout,ap_probe_send_start
I (757415) wifi:bcn_timeout,ap_probe_send_start
I (757455) wifi:bcn_timeout,ap_probe_send_start
I (757495) wifi:bcn_timeout,ap_probe_send_start
I (757555) wifi:bcn_timeout,ap_probe_send_start
I (757655) wifi:bcn_timeout,ap_probe_send_start
I (757715) wifi:bcn_timeout,ap_probe_send_start
I (757735) wifi:bcn_timeout,ap_probe_send_start
I (757815) wifi:bcn_timeout,ap_probe_send_start
I (757845) wifi:bcn_timeout,ap_probe_send_start
I (757925) wifi:bcn_timeout,ap_probe_send_start
I (758005) wifi:bcn_timeout,ap_probe_send_start
I (758245) wifi:bcn_timeout,ap_probe_send_start
I (758975) wifi:bcn_timeout,ap_probe_send_start

```

gl-chenxiaosheng, 2024-04-12:
When i disconnect the mqtt and not upload data to topic, wifi scan work fine.
What should i do? Can make data upload and wifi scan work at the meantime.

mhdong, 2024-04-26:
Hi @gl-chenxiaosheng 
Could you provide a complete log? If convenient, could you provide a reproduce demo so that we can try local reproduce.

AxelLin, 2024-05-09:
@gl-chenxiaosheng 
Any update for comment https://github.com/espressif/esp-idf/issues/13584#issuecomment-2078565179 ?

closed 2024-06-19 by Sherry616