espressif/esp-idf #13586: [Question] Is there a jig / hardware adapter for programming program raw esp32 modules?

chipweinberger, 2024-04-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Is there a jig I can buy or build to program raw esp32 modules?

[image]

[image]

apologies for asking on github. I've always preferred Github's UI.

chipweinberger, 2024-04-10:
I've found this for esp32, but nothing for s3

[image]

chipweinberger, 2024-04-10:
Looking at the pinout, it looks to be *mostly compatible* with s3. Just missing connections for GPIO3 & GPIO45 (which are strapping pins so most people don't use those anyways.)

**esp32:**
[image]

**esp32s3:**
[image]

closed 2024-04-11 by chipweinberger

dobairoland, 2024-04-11:
@chipweinberger Here is a link: https://www.espressif.com/en/products/equipment/production-testing-equipment/overview

For example, clicking on the first, you can find the list of compatible modules: https://docs.espressif.com/projects/espressif-esp-dev-kits/zh_CN/latest/other/esp-module-prog-1-r/user_guide.html#id1