espressif/esp-idf #13589: Can't get RGB LED of ESP32-H2/C6 to work

jorisvervuurt, 2024-04-11:
The `blink` example works (LED blinks white), but the LEDC examples don't work (they should according to the documentation).

Are the pins in the examples incorrect for ESP32-H2/C6?

suda-morris, 2024-04-11:
@jorisvervuurt  The RGB LED on the ESP32-C6-DevKit is also called Addressable RGB LED. We can't drive it by LEDC. Check the schematic here: https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/esp32c6/esp32-c6-devkitc-1/user_guide.html#getting-started

The blink example in esp-idf is using a component called [led strip](https://components.espressif.com/components/espressif/led_strip) to drive it. It's based on RMT peripheral, instead of LEDC.

jorisvervuurt, 2024-04-11:
Aha, thanks. I'll check `led_strip` later today. 😄

SolidStateLEDLighting, 2024-04-11:
I have created a component for that ws2812 LED.    My version closes down the RMT driver after each use to reduce power consumption.   I use that code on S3 and C6 dev kits already with pretty good success.

You can check out what I have done here: https://github.com/SolidStateLEDLighting/indication_ws2812_5.2

Here is another repository which will show you how to use that library in a large project:  https://github.com/SolidStateLEDLighting/esp32-S3-idf-advanced-template-alpha-5.2

These projects are bit on the advanced side, but it does everything that Espressif recommends.   So, you'll learn a lot and have a full understanding of how it all fits together.

closed 2024-07-25 by suda-morris