espressif/esp-idf #13590: Closing the USB serial port/JTAG on the PC causes the ESP32C3 to restart.

2023wangyin2023, 2024-04-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF4 and ESP-IDF5

### Espressif SoC revision.

ESP32C3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-c3-wroom-02

### Power Supply used.

USB

### What is the expected behavior?

The USB serial port/JTAG controller of ESP32C3 is used to communicate with the PC side. The PC side uses a universal serial port debugger. There is no problem with the communication. The expectation is to open the serial port and close the serial port communication just like normal UART serial port communication. The operation of other tasks will not be affected during the task.

### What is the actual behavior?

The USB serial port/JTAG controller of ESP32C3 is used to communicate with the PC side. The PC side uses a universal serial port debugger. There is no problem with communication. However, when clicking to close the serial port, the ESP32C3 program will restart. I took a look. After ESP-IDF technical documentation, it and RTS\DTR serial port line status, I want to know how to prevent the program from re-running after closing the serial port.

### Steps to reproduce.

1. ESP32C3’s USB serial port/JTAG operates normally;
2. After configuring the serial port information on the PC, connect normally and open the serial port;
3. If the communication is normal, close the serial port and the ESP32C3 device will restart.

BenjamimKrug, 2025-07-17:
I'm also having this issue, did you find any solution?

BenjamimKrug, 2025-07-18:
Found this issue that seems to be related and a very bad sign: https://github.com/espressif/esp-idf/issues/9762, if I understood correctly the C3 implements this behaviour via hardware, which means we either get the serialport to stop toggling the RTS(which makes the ESP restart) or we use another chip that doesn't have this behaviour. That's a bummer, cause in my use case the web serialport library in JS doesn't expose the controls to RTS in a way that allows to circumvent this issue on the PC side