espressif/esp-idf #13591: ESP32 SPI通信非特定引脚怎么配置，任意引脚配置！

i-FC19, 2024-04-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP32 SPI通信非特定引脚怎么配置，任意引脚配置！通信失败

suda-morris, 2024-04-11:
Could you share your SPI initialization code that can help us reproduce your issue?

i-FC19, 2024-04-12:
void spi_master_init(spi_host_device_t host_id, int dma_chan, uint32_t max_tran_size, gpio_num_t miso_io_num, gpio_num_t mosi_io_num, gpio_num_t clk_io_num)
{
    esp_err_t ret;
    // 配置 MISO、MOSI、CLK、CS 的引脚，和DMA最大传输字节数
    spi_bus_config_t buscfg={
        .miso_io_num=miso_io_num,              // MISO引脚定义
        .mosi_io_num=mosi_io_num,              // MOSI引脚定义
        .sclk_io_num=clk_io_num,               // CLK引脚定义
        .quadwp_io_num=-1,
        .quadhd_io_num=-1,
        .max_transfer_sz=max_tran_size,        // 最大传输字节数
    };

    // 初始化SPI总线
    ret=spi_bus_initialize(host_id, &buscfg, dma_chan);
    ESP_ERROR_CHECK(ret);
}

i-FC19, 2024-04-12:
static uint16_t spi_send_and_recv_uint16(spi_device_handle_t spi, uint16_t senddata, gpio_num_t cs_io_num)
{
    uint8_t temp = 0;
    esp_err_t ret;
    spi_transaction_t t;
    memset(&t, 0, sizeof(t));       // 清空传输结构体

    uint8_t p_data[2];              // 由于ESP32大小端的问题，传输uint16数据需要进行转换调换uint8_t的顺序
    p_data[0] = senddata>>8;
    p_data[1] = senddata;

    // 软件CSn
    gpio_set_level(cs_io_num, 0);

    t.length=8*2;                   // SPI发送AS5047P的CMD长度：16Bit。2个字节。
    t.tx_buffer=p_data;             // 命令是经过偶运算的addr，直接&addr会引起大小端错误
    //  t.tx_buffer=senddata;             // 命令是经过偶运算的addr，直接&addr会引起大小端错误
    t.flags = SPI_TRANS_USE_RXDATA;
    //t.user=(void*)0;                // D/C 线电平为0，传输命令
    ret=spi_device_polling_transmit(spi, &t);  // 开始传输
    assert(ret==ESP_OK);            // 应该没有问题

    // 软件CSn
    gpio_set_level(cs_io_num, 1);

    // 将接收到的数据按大小端重新排列
    temp = *t.rx_data;
    *t.rx_data = *(t.rx_data+1);
    *(t.rx_data+1) = temp;

    return *(uint16_t*)t.rx_data; // 返回经过重新大小端排列的数据
}

suda-morris, 2024-04-12:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gpio.html#gpio-summary

> SPI0/1: GPIO6-11 and GPIO16-17 are usually connected to the SPI flash and PSRAM integrated on the module and therefore should not be used for other purposes.

i-FC19, 2024-04-12:
飞线测试连接，测量时钟没有波形

// ESP32默认的SPI2引脚。使用默认引脚可以使用IOMUX将GPIO翻转速率提高到80MHz，GPIO矩阵不建议超过26MHz
#define SPI2_DEF_PIN_NUM_MISO    12  // SPI2默认的MISO引脚
#define SPI2_DEF_PIN_NUM_MOSI    13  // SPI2默认的MOSI引脚
#define SPI2_DEF_PIN_NUM_CLK     14  // SPI2默认的CLK引脚
#define ATT7022_CS_PIN 		32

wanckl, 2024-04-18:
@i-FC19 你是指使用 13/15/17 时候没有波形输出吗？一条都没有波形吗

任意脚就是正常这么配的，没什么不一样，驱动里会配好的。

另外，esp32 iomux 只能跑到16M, gpio不超过10M,否则不保证安全的。

i-FC19, 2024-04-18:
[spi.txt](https://github.com/espressif/esp-idf/files/15022095/spi.txt)
是的没有波形

wanckl, 2024-04-18:
@i-FC19 不可能一条都没有，像楼上说的，你只是碰巧用到了内部引脚 17，随便换一个别的应该是有输出的

```c
//非ESP32默认的SPI引脚。
#define SPI_MOSI_PIN 15
#define SPI_MISO_PIN 18
#define SPI_SCLK_PIN 13
#define SPI_CS_PIN 32

void app_main(void)
{
    spi_device_handle_t dev0;
    uint32_t master_send;
    uint32_t master_receive;
    uint32_t master_expect;

    spi_bus_config_t buscfg = {
        .sclk_io_num = SPI_SCLK_PIN,
        .miso_io_num = SPI_MISO_PIN,
        .mosi_io_num = SPI_MOSI_PIN,
    };
    spi_device_interface_config_t devcfg = SPI_DEVICE_TEST_DEFAULT_CONFIG();
    devcfg.spics_io_num = SPI_CS_PIN;
    spi_bus_initialize(TEST_SPI_HOST, &buscfg, SPI_DMA_CH2);
    spi_bus_add_device(TEST_SPI_HOST, &devcfg, &dev0);

    spi_transaction_t trans_cfg = {
        .tx_buffer = &master_send,
        .rx_buffer = &master_receive,
        .user = & master_expect,
        .length = sizeof(uint32_t) * 8,
    };

    for (int i = 0; i < 100; i++) {
        master_send += i;
        spi_device_transmit(dev0, &trans_cfg);
    }

    spi_bus_remove_device(dev0);
    spi_bus_free(TEST_SPI_HOST);
}
```

i-FC19, 2024-04-19:
这个芯片是其他人用的，借鉴的程序

closed 2024-04-19 by i-FC19