espressif/esp-idf #13592: espnow发送存在延时该如何处理

kkkintama, 2024-04-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

用串口接收数据然后调用espnow透传，发现从调用esp_now_send之后到系统调用注册的espnow_send_cb之间总是有一个随机的延时存在，从10ms-50ms不等。整个例程代码一共只有串口接收和espnow接收两个任务，从串口接收到的数据立即会立即发送，发送器log如下：
[0;31mE (1956376) uart_port: dt = 244[0m
[0;31mE (1956379) my_espnow2: dt = 236[0m
[0;31mE (1956620) uart_port: dt = 244[0m
[0;31mE (1956628) my_espnow2: dt = 248[0m
[0;31mE (1956864) uart_port: dt = 244[0m
[0;31mE (1956866) my_espnow2: dt = 238[0m
[0;31mE (1957108) uart_port: dt = 244[0m
[0;31mE (1957114) my_espnow2: dt = 248[0m
[0;31mE (1957352) uart_port: dt = 244[0m
[0;31mE (1957359) my_espnow2: dt = 245[0m
[0;31mE (1957596) uart_port: dt = 244[0m
[0;31mE (1957653) my_espnow2: dt = 294[0m
[0;31mE (1957840) uart_port: dt = 244[0m
[0;31mE (1957842) my_espnow2: dt = 189[0m
[0;31mE (1958084) uart_port: dt = 244[0m
串口发送间隔是244ms。这个发送间隔计算写于串口接收之后、esp_now_send之前；espnow的发送完成间隔计算写于espnow_send_cb中，接收器的espnow接收任务中也计算了接收间隔，与发送端是匹配的。理想情况下，发送间隔是244ms，那么发送完成间隔也应该是244ms，接收间隔也应该是244ms，到底什么事件会产生这样一个延时？望指点。

zhangyanjiaoesp, 2024-05-17:
@kkkintama 调用 `esp_now_send()` 发送完数据之后会产生一个事件，通知wifi task 发送完成，然后 wifi task 会调用注册的回调函数 `espnow_send_cb()` 返回应用层，这之间的时间差主要是由于 wifi task 调度导致的随机延迟，其实数据是立即发送出去的。

kkkintama, 2024-05-17:
@zhangyanjiaoesp 我也有从接收器的接收间隔角度观察过，发送器的数据明显是没有立即发出的，如果发送器那端能立即发出接收器这边的接收间隔应该是相同的。但实测下来接收器也有个随机的接收延时，这个随机延时和发送器的延时完全是对得上的。
我直接用的两个板子和官方例程做的测试，这很容易就能实现，你们也可以观察一下。
从这些测试结果来看我认为数据直到wifi task调度到才会被发送出去，否则一直是存在缓存里的。这种机制对实时性要求比较高的系统来说很不友好，比较头疼。

zhangyanjiaoesp, 2024-05-20:
@kkkintama 你的 case 里面只有ESP-NOW 数据吗？如果 STA 连接了 AP（或者 AP 连接了 STA），并且两者之间有通信，是会影响 ESP-NOW 的收发包的。

kkkintama, 2024-05-21:
@zhangyanjiaoesp https://github.com/espressif/esp-idf/blob/4523f2d67465373f0e732a3264273a8e84a1a6d1/examples/wifi/espnow/main/espnow_example_main.c#L45
当时是直接用这个例程做的测试，仅在收发处获取了一下systick计算时间间隔，看wifi配置模块是做了STA但是没连啥AP，应该是只有ESP-NOW数据的。

zhangyanjiaoesp, 2024-05-21:
@kkkintama 你用的什么 IDF 分支？什么芯片？我本地测试一下。

kkkintama, 2024-05-24:
> @kkkintama 你用的什么 IDF 分支？什么芯片？我本地测试一下。

硬件是ESP32-C3-DevKitM-1开发板，IDF分支是用vscode插件下载的release v5.1.3版本

zhangyanjiaoesp, 2024-05-27:
> @kkkintama 调用 `esp_now_send()` 发送完数据之后会产生一个事件，通知wifi task 发送完成，然后 wifi task 会调用注册的回调函数 `espnow_send_cb()` 返回应用层，这之间的时间差主要是由于 wifi task 调度导致的随机延迟，其实数据是立即发送出去的。

@kkkintama 和环境是有关系的，如果周围环境比较差，ESP-NOW 的发包比较困难，那么到回调函数的返回时间就会变长。

kkkintama, 2024-05-28:
看来还是得从应用层处理这个问题，和应用环境有关那是避不开这个延迟了。

closed 2024-05-28 by kkkintama

kkkintama, 2024-05-28:
@zhangyanjiaoesp esp32有2.4g物理层的API接口么，就是说不用wifi协议，自己用收发API写个私有协议啥的，这种能整么？

zhangyanjiaoesp, 2024-05-28:
@kkkintama 没有这样的 API. 只有下面的这样一个 API，可以发送一些用户指定的包。https://github.com/espressif/esp-idf/blob/003f3bb5dc7c8af8b71926b7a0118cfc503cab11/components/esp_wifi/include/esp_wifi.h#L1144