espressif/esp-idf #13593: `esp_tls_conn_new_async` does not work if real time is set

ngc7293, 2024-04-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

master

### Espressif SoC revision.

ESP32-S2

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S2-DevKitM-1

### Power Supply used.

USB

### What is the expected behavior?

`esp_tls_conn_new_sync` waits up to `timeout_ms` milliseconds before returning (set in the passed `esp_tls_cfg_t`)

### What is the actual behavior?

`esp_tls_conn_new_sync`  returns immediately and logs a warning "Failed to open new connection in specified timeout"

### Steps to reproduce.

1. Set the proper time (I used esp_sntp)
2. Call `esp_tls_conn_new_sync` with `timeout_ms = INT32_MAX`

### More Information.

The issue here is that the function's body uses `uint32_t` to store the epoch timestamp in milliseconds. When the "system" time has not been set this works fine, but if you have properly set the time the current epoch timestamp exceeds the holding capacity of `uint32_t`.

Additionally, there could be a case where the time is set _during_ the connection process which would also break the timeout check.

The reliance on `gettimeofday` should probably be replaced with an internal counter, like FreeRTOS's tick count or esp_timer's  `esp_timer_get_time`

AdityaHPatwardhan, 2024-04-23:
Hi @ngc7293 Thanks for finding the issue.
The use of `uint32_t` type for 64bit time seems wrong. Will make the relevant changes and update here.
Thanks,
Aditya

AxelLin, 2024-05-03:
@AdityaHPatwardhan 
Do you have ETA for the fix? (I think this needs urgent fix for release branches)

AdityaHPatwardhan, 2024-05-03:
@AxelLin It is been merged in the internal code-base. Should be available on GitHub in the release branches in a few days.

closed 2024-05-16 by espressif-bot

AxelLin, 2024-06-08:
@AdityaHPatwardhan @mahavirj 
While checking v5.3-beta2 chances, I found below fix is not in v5.2 branch. Could you check if this fix is required for v5.2 branch?
Initialize mbedtls RNG context prior to setting client config in ESP-TLS, this fixes ECC key parsing issue for specific cases ( a02b096)

AdityaHPatwardhan, 2024-06-14:
Hi @AxelLin Yes, we have backported the fix. Should be visible in a few days on v5.2.

AxelLin, 2024-07-02:
@AdityaHPatwardhan 
v5.1 branch still needs backport fix.

AdityaHPatwardhan, 2024-07-02:
Hi @AxelLin, have backported this fix to `v5.1`, and `v5.0` as well. Should be visible in a few days.