espressif/esp-idf #13596: Occasionally a watchdog is triggered when the MQTT is connected

maenkai, 2024-04-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.3.1

### Espressif SoC revision.

ESP32D-WROVE

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32D-WROVE

### Power Supply used.

USB

### What is the expected behavior?

Does not trigger watchdog

### What is the actual behavior?

Once started, the connected mqtt occasionally triggers a watchdog.

### Steps to reproduce.

1. chip starup
2. connect mqtt
  mqtt_connecting = true;
  esp_mqtt_client_config_t mqtt_cfg = {
      .uri = MZ_MQTT_URI,
      .username = mz_get_device_id(),
      .password = mz_get_mqtt_login_password(),
      .client_id = mz_get_device_id(),
      .keepalive = MQTT_KEEPALIVE_S,
      .cert_pem = (const char *)mz_get_meizu_cert(),
      .network_timeout_ms = 0,
  };

  mz_mqtt_client = esp_mqtt_client_init(&mqtt_cfg);
  if (mz_mqtt_client == NULL) {
    mqtt_connecting = false;
    mz_mqtt_client_need_start = true;
      MZ_ERROR(TAG, "mqtt not initialize fail !");
    return false;
  }
  
  mqtt_connecting = true;
  mz_mqtt_client_need_start = false;
  
  event_msg_receive_t queue = {
    .event = MZ_MQTT_EVENT_INIT_DONE,
  };
  esp_mqtt_client_register_event(mz_mqtt_client, ESP_EVENT_ANY_ID,
                                 mz_mqtt_event_handler, mz_mqtt_client);
  
  esp_err_t err = esp_mqtt_client_start(mz_mqtt_client);
  if (err != ESP_OK) {
    return false;
  }

3. subject some topic

### Debug Logs.

```plain
E (176656) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (176656) task_wdt:  - IDLE (CPU 1)
E (176656) task_wdt: Tasks currently running:
E (176656) task_wdt: CPU 0: IDLE
E (176656) task_wdt: CPU 1: mqtt_task
E (176656) task_wdt: Aborting.

abort() was called at PC 0x40193ff6 on core 0

Backtrace:0x400910e6:0x3ffbf310 0x40091a09:0x3ffbf330 0x40098bba:0x3ffbf350 0x40193ff6:0x3ffbf3c0 0x40083109:0x3ffbf3e0 0x401f4cf7:0x3ffbd860 0x400d99b9:0x3ffbd880 0x40092f67:0x3ffbd8a0

0x400910e6: panic_abort at /home/zy/esp/esp-idf/components/esp_system/panic.c:379
0x40091a09: esp_system_abort at /home/zy/esp/esp-idf/components/esp_system/system_api.c:112
0x40098bba: abort at /home/zy/esp/esp-idf/components/newlib/abort.c:46
0x40193ff6: task_wdt_isr at /home/zy/esp/esp-idf/components/esp_common/src/task_wdt.c:182 (discriminator 1)
0x40083109: _xt_lowint1 at /home/zy/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105
0x401f4cf7: cpu_ll_waiti at /home/zy/esp/esp-idf/components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at /home/zy/esp/esp-idf/components/esp_pm/pm_impl.c:827
0x400d99b9: esp_vApplicationIdleHook at /home/zy/esp/esp-idf/components/esp_common/src/freertos_hooks.c:63
0x40092f67: prvIdleTask at /home/zy/esp/esp-idf/components/freertos/tasks.c:3839
```

### More Information.

my mqtt configure
CONFIG_MQTT_PROTOCOL_311=y
CONFIG_MQTT_TRANSPORT_SSL=y
CONFIG_MQTT_TRANSPORT_WEBSOCKET=y
CONFIG_MQTT_TRANSPORT_WEBSOCKET_SECURE=y
# CONFIG_MQTT_MSG_ID_INCREMENTAL is not set
# CONFIG_MQTT_SKIP_PUBLISH_IF_DISCONNECTED is not set
# CONFIG_MQTT_REPORT_DELETED_MESSAGES is not set
CONFIG_MQTT_USE_CUSTOM_CONFIG=y
CONFIG_MQTT_TCP_DEFAULT_PORT=1883
CONFIG_MQTT_SSL_DEFAULT_PORT=8883
CONFIG_MQTT_WS_DEFAULT_PORT=80
CONFIG_MQTT_WSS_DEFAULT_PORT=443
CONFIG_MQTT_BUFFER_SIZE=2048
CONFIG_MQTT_TASK_STACK_SIZE=6144
CONFIG_MQTT_TASK_PRIORITY=5
# CONFIG_MQTT_TASK_CORE_SELECTION_ENABLED is not set
# CONFIG_MQTT_CUSTOM_OUTBOX is not set
	CONFIG_MQTT_OUTBOX_EXPIRED_TIMEOUT_MS=30000
# end of ESP-MQTT Configurations

AxelLin, 2024-04-12:
@maenkai 
esp-idf-v4.3 branch is already EOL.
Could you try reproduce this issue with recent esp-idf versions?

euripedesrocha, 2024-04-12:
@maenkai as mentioned, could you try to reproduce in newer idf version?

To be able to help you identify any misconfigurations and suggest a workaround, I will need more information on your log, do you have any logs from the MQTT client?

maenkai, 2024-04-15:
> @maenkai as mentioned, could you try to reproduce in newer idf version?
> 
> To be able to help you identify any misconfigurations and suggest a workaround, I will need more information on your log, do you have any logs from the MQTT client?

I'm sorry, it only appeared once so far, and there are no logs

maenkai, 2024-04-15:
> @maenkai esp-idf-v4.3 branch is already EOL. Could you try reproduce this issue with recent esp-idf versions?
I will try it

AxelLin, 2024-06-05:
> > @maenkai esp-idf-v4.3 branch is already EOL. Could you try reproduce this issue with recent esp-idf versions?
> > I will try it

@maenkai Any update?

closed 2024-06-18 by maenkai