espressif/esp-idf #13597: Compile error if define LWIP_USE_EXTERNAL_MBEDTLS 1 and build with PPP eanbled

AxelLin, 2024-04-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.3-416-gd23b7a0361

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

It should pass compilation with LWIP_USE_EXTERNAL_MBEDTLS=1.

### What is the actual behavior?

"#define LWIP_USE_EXTERNAL_MBEDTLS 1" with below config:
CONFIG_LWIP_PPP_SUPPORT=y
CONFIG_LWIP_PPP_NOTIFY_PHASE_SUPPORT=y
CONFIG_LWIP_PPP_PAP_SUPPORT=y
CONFIG_LWIP_PPP_CHAP_SUPPORT=y
CONFIG_LWIP_PPP_MSCHAP_SUPPORT=y
CONFIG_LWIP_PPP_MPPE_SUPPORT=y

I got below error:
```
In file included from /home/axel/esp/esp-idf/components/lwip/lwip/src/include/netif/ppp/mppe.h:42,
                 from /home/axel/esp/esp-idf/components/lwip/lwip/src/include/netif/ppp/ppp.h:174,
                 from /home/axel/esp/esp-idf/components/lwip/lwip/src/include/netif/ppp/ppp_impl.h:53,
                 from /home/axel/esp/esp-idf/components/lwip/lwip/src/core/init.c:62:
/home/axel/esp/esp-idf/components/lwip/lwip/src/include/netif/ppp/pppcrypt.h:128:27: error: unknown type name 'mbedtls_arc4_context'
  128 | #define lwip_arc4_context mbedtls_arc4_context
      |                           ^~~~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/lwip/lwip/src/include/netif/ppp/mppe.h:155:9: note: in expansion of macro 'lwip_arc4_context'
  155 |         lwip_arc4_context arc4;
      |         ^~~~~~~~~~~~~~~~~
```

Set
CONFIG_LWIP_PPP_CHAP_SUPPORT=n
CONFIG_LWIP_PPP_MSCHAP_SUPPORT=n

I got below error:
```
In file included from /home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/magic.c:83:
/home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/magic.c: In function 'magic_churnrand':
/home/axel/esp/esp-idf/components/lwip/lwip/src/include/netif/ppp/pppcrypt.h:108:26: error: unknown type name 'mbedtls_md5_context'; did you mean 'lwip_md5_context'?
  108 | #define lwip_md5_context mbedtls_md5_context
      |                          ^~~~~~~~~~~~~~~~~~~
```

### Steps to reproduce.

1. In components/lwip/lwip/src/include/netif/ppp/ppp_opts.h
    #define LWIP_USE_EXTERNAL_MBEDTLS       1
2. Set below config to enable PPP settings
CONFIG_LWIP_PPP_SUPPORT=y
CONFIG_LWIP_PPP_NOTIFY_PHASE_SUPPORT=y
CONFIG_LWIP_PPP_PAP_SUPPORT=y
CONFIG_LWIP_PPP_CHAP_SUPPORT=y
CONFIG_LWIP_PPP_MSCHAP_SUPPORT=y
CONFIG_LWIP_PPP_MPPE_SUPPORT=y

david-cermak, 2024-04-12:
Hi @AxelLin 

It looks like lwip is missing mbedtls headers.
Could you please share your lwip/mbedtls configuration? (makefiles, custom flags, changes to IDF)
I'm only interested how did you tell lwip to use mbedtls? (using `LWIP_HAVE_MBEDTLS`, or with `ALTCP`, or maybe setting `LWIP_MBEDTLS_INCLUDE_DIRS` ?)

david-cermak, 2024-04-12:
Could you please try to remove these polarssl sources from the lwip build?

https://github.com/espressif/esp-idf/blob/5a40bb8746633477c07ff9a3e90016c37fa0dc0c/components/lwip/CMakeLists.txt#L132-L136

while keeping `LWIP_USE_EXTERNAL_MBEDTLS` disabled?

Then you'll probably have to add
```cmake
        idf_component_optional_requires(PRIVATE mbedtls)
````
to the end of the lwip makefile for the linker to locate these symbols

AxelLin, 2024-04-12:
Hi @david-cermak 

I made below changes base on above comment:

```
diff --git a/components/lwip/CMakeLists.txt b/components/lwip/CMakeLists.txt
index 671da81e20..866e2caff7 100644
--- a/components/lwip/CMakeLists.txt
+++ b/components/lwip/CMakeLists.txt
@@ -125,12 +125,7 @@ if(CONFIG_LWIP_PPP_SUPPORT)
         "lwip/src/netif/ppp/pppos.c"
         "lwip/src/netif/ppp/upap.c"
         "lwip/src/netif/ppp/utils.c"
-        "lwip/src/netif/ppp/vj.c"
-        "lwip/src/netif/ppp/polarssl/arc4.c"
-        "lwip/src/netif/ppp/polarssl/des.c"
-        "lwip/src/netif/ppp/polarssl/md4.c"
-        "lwip/src/netif/ppp/polarssl/md5.c"
-        "lwip/src/netif/ppp/polarssl/sha1.c")
+        "lwip/src/netif/ppp/vj.c")
 endif()
 
 if(NOT ${target} STREQUAL "linux")
@@ -207,3 +202,4 @@ if(${target} STREQUAL "linux")
     find_package(Threads REQUIRED)
     target_link_libraries(${COMPONENT_LIB} PRIVATE Threads::Threads)
 endif()
+idf_component_optional_requires(PRIVATE mbedtls)

```
I got below build error:

/home/axel/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/lwip/liblwip.a(chap_ms.c.obj):(.literal.NTPasswordHash+0x0): undefined reference to `md4_starts'
/home/axel/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/lwip/liblwip.a(chap_ms.c.obj):(.literal.NTPasswordHash+0x4): undefined reference to `md4_update'
/home/axel/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/lwip/liblwip.a(chap_ms.c.obj):(.literal.NTPasswordHash+0x8): undefined reference to `md4_finish'
/home/axel/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/lwip/liblwip.a(chap_ms.c.obj):(.literal.ChallengeResponse+0x0): undefined reference to `des_setkey_enc'
/home/axel/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/lwip/liblwip.a(chap_ms.c.obj):(.literal.ChallengeResponse+0x4): undefined reference to `des_crypt_ecb'
/home/axel/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/lwip/liblwip.a(chap_ms.c.obj):(.literal.Set_Start_Key+0x0): undefined reference to `sha1_starts'
/home/axel/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/lwip/liblwip.a(chap_ms.c.obj):(.literal.Set_Start_Key+0x4): undefined reference to `sha1_finish'
/home/axel/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/lwip/liblwip.a(chap_ms.c.obj): in function `NTPasswordHash':
/home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/chap_ms.c:576: undefined reference to `md4_starts'
/home/axel/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/chap_ms.c:577: undefined reference to `md4_update'
/home/axel/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/chap_ms.c:578: undefined reference to `md4_finish'
/home/axel/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/lwip/liblwip.a(chap_ms.c.obj): in function `ChallengeResponse':
/home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/chap_ms.c:513: undefined reference to `des_setkey_enc'
/home/axel/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/chap_ms.c:514: undefined reference to `des_crypt_ecb'
/home/axel/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/chap_ms.c:517: undefined reference to `des_setkey_enc'
/home/axel/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/chap_ms.c:519: undefined reference to `des_crypt_ecb'
/home/axel/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/chap_ms.c:523: undefined reference to `des_setkey_enc'
/home/axel/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/chap_ms.c:525: undefined reference to `des_crypt_ecb'
[... cut ...]
/home/axel/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/lwip/liblwip.a(magic.c.obj): in function `magic_random_bytes':
/home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/magic.c:169: undefined reference to `md5_starts'
/home/axel/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/magic.c:172: undefined reference to `md5_finish'

david-cermak, 2024-04-12:
I'm sorry, I misread the previous error message. The symbol duplication comes from the libssh -- an external library. I'm afraid I cannot provide any quick workaround (other that running `objcopy` to namespace the symbols in SSH with a given prefix).
And it looks like the mbedtls alternative in PPP was left long unmaintained...

I'll take a look into it.

AxelLin, 2024-04-13:
Hi @david-cermak 

FYI, I made some progress to compile with LWIP_USE_EXTERNAL_MBEDTLS=1.

I can compile with below settings:
CONFIG_LWIP_PPP_SUPPORT=y
CONFIG_LWIP_PPP_NOTIFY_PHASE_SUPPORT=y
CONFIG_LWIP_PPP_PAP_SUPPORT=y
CONFIG_LWIP_PPP_CHAP_SUPPORT=y
CONFIG_LWIP_PPP_MSCHAP_SUPPORT=n
CONFIG_LWIP_PPP_MPPE_SUPPORT=n

With below changes in esp-idf and lwip:

// Changes in esp-idf
```
diff --git a/components/lwip/CMakeLists.txt b/components/lwip/CMakeLists.txt
index 671da81e20..4d7fb022ac 100644
--- a/components/lwip/CMakeLists.txt
+++ b/components/lwip/CMakeLists.txt
@@ -207,3 +207,4 @@ if(${target} STREQUAL "linux")
     find_package(Threads REQUIRED)
     target_link_libraries(${COMPONENT_LIB} PRIVATE Threads::Threads)
 endif()
+idf_component_optional_requires(PRIVATE mbedtls)

```

// Changes in lwip
```
diff --git a/src/include/netif/ppp/ppp_opts.h b/src/include/netif/ppp/ppp_opts.h
index 479a006d..3b5dc69c 100644
--- a/src/include/netif/ppp/ppp_opts.h
+++ b/src/include/netif/ppp/ppp_opts.h
@@ -362,7 +362,7 @@
  * LWIP_USE_EXTERNAL_MBEDTLS: Use external mbed TLS library
  */
 #ifndef LWIP_USE_EXTERNAL_MBEDTLS
-#define LWIP_USE_EXTERNAL_MBEDTLS       0
+#define LWIP_USE_EXTERNAL_MBEDTLS       1
 #endif
 
 /*
diff --git a/src/include/netif/ppp/pppcrypt.h b/src/include/netif/ppp/pppcrypt.h
index c0230bbc..9e750279 100644
--- a/src/include/netif/ppp/pppcrypt.h
+++ b/src/include/netif/ppp/pppcrypt.h
@@ -97,6 +97,9 @@ extern "C" {
  * Map hashes and ciphers functions to mbed TLS
  */
 #if LWIP_USE_EXTERNAL_MBEDTLS
+#include "mbedtls/des.h"
+#include "mbedtls/md5.h"
+#include "mbedtls/sha1.h"
 
 #define lwip_md4_context mbedtls_md4_context
 #define lwip_md4_init mbedtls_md4_init

```

Also note, I hit compile error if 
CONFIG_LWIP_PPP_MSCHAP_SUPPORT=y
CONFIG_LWIP_PPP_MPPE_SUPPORT=y

// if CONFIG_LWIP_PPP_MPPE_SUPPORT=y
/home/axel/esp/esp-idf/components/lwip/lwip/src/include/netif/ppp/pppcrypt.h:130:27: error: unknown type name 'mbedtls_arc4_context'
  130 | #define lwip_arc4_context mbedtls_arc4_context
      |                           ^~~~~~~~~~~~~~~~~~~~

// if CONFIG_LWIP_PPP_MSCHAP_SUPPORT=y
/home/axel/esp/esp-idf/components/lwip/lwip/src/include/netif/ppp/pppcrypt.h:103:26: error: unknown type name 'mbedtls_md4_context'; did you mean 'mbedtls_md5_context'?
  103 | #define lwip_md4_context mbedtls_md4_context
      |                          ^~~~~~~~~~~~~~~~~~~

mbedtls_arc4_context and mbedtls_md4_context are not exist in mbedtls-3.x.
see https://github.com/Mbed-TLS/mbedtls/blob/development/docs/3.0-migration-guide.md#remove-md2-md4-rc4-blowfish-and-xtea-algorithms .
I'm not sure how to addess this build error.

david-cermak, 2024-04-15:
Thanks for the update! It seems that you only need to add the optional dependency to `mbedtls`, as both the define and includes can be provided via the build system or hook files.

> mbedtls_arc4_context and mbedtls_md4_context are not exist in mbedtls-3.x.
> I'm not sure how to addess this build error.

You can probably report the problem to lwip upstream, but I think that the assesment would be that `MPPE` and `MSCHAP` won't be supported with mbedtls, and mbedtls `> v3.0`

AxelLin, 2024-04-18:
> Thanks for the update! It seems that you only need to add the optional dependency to `mbedtls`, as both the define and includes can be provided via the build system or hook files.

Is it possible to achieve this without touching esp-idf ? (Is it possible to just change it in my application's CMakeLists.txt? )

david-cermak, 2024-04-18:
Yes, it's possible to make the changes in the project makefile:
1) enable `LWIP_USE_EXTERNAL_MBEDTLS` using https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/lwip.html#customized-lwip-options-from-esp-idf-build-system
2) pre-include mbedTLS headers via the lwip hook file: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/lwip.html#customized-lwip-hooks
  - this might cause some issues order due to include order, like missing types, etc, so you might need to include other generic headers, or worst case just forward declare the mbedtls structs
3) dependency on mbedtls -- this may sound a bit *hacky*, but should work, too
```cmake
# idf_component_get_property(lwip lwip COMPONENT_LIB) -- already done in step 1), 2)
idf_component_get_property(mbedtls mbedtls COMPONENT_LIB)
target_link_libraries(${lwip} PRIVATE ${mbedtls})
```

but I think we should provide a cleaner way for the points `2)` and `3)` at some point in IDF.

AxelLin, 2024-04-21:
> 2. pre-include mbedTLS headers via the lwip hook file: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/lwip.html#customized-lwip-hooks

This does not work, I still got build error:
```
/home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/chap-md5.c: In function 'chap_md5_make_response':
/home/axel/esp/esp-idf/components/lwip/lwip/src/include/netif/ppp/pppcrypt.h:108:26: error: unknown type name 'mbedtls_md5_context'; did you mean 'lwip_md5_context'?
  108 | #define lwip_md5_context mbedtls_md5_context
      |                          ^~~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/chap-md5.c:98:9: note: in expansion of macro 'lwip_md5_context'
   98 |         lwip_md5_context ctx;
      |         ^~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/lwip/lwip/src/include/netif/ppp/pppcrypt.h:109:23: error: implicit declaration of function 'mbedtls_md5_init' [-Werror=implicit-function-declaration]
  109 | #define lwip_md5_init mbedtls_md5_init
      |                       ^~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/chap-md5.c:105:9: note: in expansion of macro 'lwip_md5_init'
  105 |         lwip_md5_init(&ctx);
      |         ^~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/lwip/lwip/src/include/netif/ppp/pppcrypt.h:110:25: error: implicit declaration of function 'mbedtls_md5_starts' [-Werror=implicit-function-declaration]
  110 | #define lwip_md5_starts mbedtls_md5_starts
      |                         ^~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/chap-md5.c:106:9: note: in expansion of macro 'lwip_md5_starts'
  106 |         lwip_md5_starts(&ctx);
      |         ^~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/lwip/lwip/src/include/netif/ppp/pppcrypt.h:111:25: error: implicit declaration of function 'mbedtls_md5_update' [-Werror=implicit-function-declaration]
  111 | #define lwip_md5_update mbedtls_md5_update
      |                         ^~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/chap-md5.c:107:9: note: in expansion of macro 'lwip_md5_update'
  107 |         lwip_md5_update(&ctx, &idbyte, 1);
      |         ^~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/lwip/lwip/src/include/netif/ppp/pppcrypt.h:112:25: error: implicit declaration of function 'mbedtls_md5_finish' [-Werror=implicit-function-declaration]
  112 | #define lwip_md5_finish mbedtls_md5_finish
      |                         ^~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/chap-md5.c:110:9: note: in expansion of macro 'lwip_md5_finish'
  110 |         lwip_md5_finish(&ctx, &response[1]);
      |         ^~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/lwip/lwip/src/include/netif/ppp/pppcrypt.h:113:23: error: implicit declaration of function 'mbedtls_md5_free' [-Werror=implicit-function-declaration]
  113 | #define lwip_md5_free mbedtls_md5_free
      |                       ^~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/chap-md5.c:111:9: note: in expansion of macro 'lwip_md5_free'
  111 |         lwip_md5_free(&ctx);
      |         ^~~~~~~~~~~~~

```

AxelLin, 2024-04-21:
According to the comment in 
https://github.com/espressif/esp-lwip/blob/2.1.2-esp/src/include/netif/ppp/pppcrypt.h#L41-L44

This means I need to change the code in esp-idf again.. (or maybe you can check if this should be fix in esp-idf upstream?)
```
diff --git a/components/lwip/port/esp32xx/include/arch/cc.h b/components/lwip/port/esp32xx/include/arch/cc.h
index 486e2782bd..f2b0ff0248 100644
--- a/components/lwip/port/esp32xx/include/arch/cc.h
+++ b/components/lwip/port/esp32xx/include/arch/cc.h
@@ -96,6 +96,12 @@ typedef int sys_prot_t;
 
 #endif /* NDEBUG */
 
+#if LWIP_USE_EXTERNAL_MBEDTLS
+#include "mbedtls/des.h"
+#include "mbedtls/md5.h"
+#include "mbedtls/sha1.h"
+#endif
+
 #ifdef __cplusplus
 }
 #endif

```

david-cermak, 2024-05-24:
Hi @AxelLin 

You were right about the include injection, it's not possible to add them to the hook file, as it comes a bit later after processing the `pppots.h`.

I think it makes sense to support this option (lwip to use mbedTLS) on IDF level as well, as it may also be used in other modules (like snmp or altcp) and helps reducing build time/binary size a little. Although I'd prefer not adding those dependencies to `cc.h`, since that would require `PUBLIC` dependency on mbedTLS (rather than private). I'll address it directly in `pppcrypt.h` together with explanation of deprecating some auth methods.
Reopening the issue.

As of now, I can offer this workaround:
```
LWIP_USE_EXTERNAL_POLARSSL=1
LWIP_INCLUDED_POLARSSL_MD5=1
LWIP_INCLUDED_POLARSSL_SHA1=1
...
```

and removing the polarssl sources from lwip:
```cmake
idf_component_get_property(lwip lwip COMPONENT_LIB)
idf_component_get_property(lwip_dir lwip COMPONENT_DIR)
get_target_property(srcs ${lwip} SOURCES)
list(REMOVE_ITEM srcs "${lwip_dir}/lwip/src/netif/ppp/polarssl/md5.c")
...
set_property(TARGET ${lwip} PROPERTY SOURCES ${srcs})
```

(you may also need to add the `idf_component_optional_requires()` to the libssh to fix potential unresolved symbols...)

AxelLin, 2024-05-26:
Hi @david-cermak 
I'm not sure if MSCHAP and MPPE are still used or not nowdays.
Just wondering is it safe to remove(disable) CONFIG_LWIP_PPP_MSCHAP_SUPPORT and CONFIG_LWIP_PPP_MPPE_SUPPORT ?

AxelLin, 2024-05-27:
> Hi @david-cermak I'm not sure if MSCHAP and MPPE are still used or not nowdays. Just wondering is it safe to remove(disable) CONFIG_LWIP_PPP_MSCHAP_SUPPORT and CONFIG_LWIP_PPP_MPPE_SUPPORT ?

Never mind, we don't use MSCHAP/MPPE.

closed 2024-07-24 by espressif-bot