espressif/esp-idf #13600: Adding custom lwIP based interface layers to ESP-NETIF

tomasgareau, 2024-04-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

# Background

I am working on creating an RFC7668 (LoWPAN6-BLE) connection on an ESP-32 using the NimBLE bluetooth stack.

I managed to get this working directly with lwIP by:

<details>
<summary>Adding the missing `lowpan6*` sources to the build</summary>

```cmake
idf_component_register(
  SRCS
    src/main.c
    "$ENV{IDF_PATH}/components/lwip/lwip/src/netif/lowpan6.c"
    "$ENV{IDF_PATH}/components/lwip/lwip/src/netif/lowpan6_ble.c"
    "$ENV{IDF_PATH}/components/lwip/lwip/src/netif/lowpan6_common.c"
    # etc
)
```
</details>

<details>
<summary>Establishing a GAP connection to a device advertising support for the Internet Protocol Support Service</summary>

```c
static int on_gap_event(struct ble_gap_event* event, void* arg)
{
    switch (event->type)
    {
    case BLE_GAP_EVENT_DISC:
        if (device_supports_ipss(event->disc))
        {
            uint8_t own_addr_type;
            ble_hs_id_infer_auto(0, &own_addr_type);
            ble_gap_connect(own_addr_type, &event->disc.addr, 10000, NULL, on_gap_event, NULL);
        }
    // etc...
    }
}
```
</details>

<details>
<summary>Establishing an L2CAP connection to said device after GAP connection</summary>

```c
static int on_gap_event(struct ble_gap_event* event, void* arg)
{
    switch (event->type)
    {
    case BLE_GAP_EVENT_CONNECT:
    {
        struct os_mbuf* sdu_rx = os_mbuf_get_pkthdr(&m_mbuf_pool, 0);
        ble_l2cap_connect(event->connect.conn_handle, 0x0023, 1280, sdu_rx, on_l2cap_event, NULL);
    }
        // etc...
    }
}
```
</details>
  
<details>
<summary>Creating an lwIP netif that uses lwIP's `rfc7668_if_init` and `rfc7668_input` functions for initialization and receiving input, respectively</summary>

```c
struct netif my_netif;
netif_add(&my_netif, NULL, NULL, NULL, NULL, rfc7668_if_init, rfc7668_input);
```

* setting that netif instance to use a custom `linkoutput` function that outputs data using the L2CAP connection, above:
```c
my_netif.linkoutput = my_l2cap_linkoutput_function;
```
</details>

<details>
<summary>And finally hooking up the netif's input/output functions to their respective L2CAP events</summary>

```c
static int on_l2cap_event(struct ble_l2cap_event* event, void* arg)
{
    switch (event->type)
    {
    case BLE_L2CAP_EVENT_COC_CONNECTED:
        struct ble_gap_conn_desc desc;
        ble_gap_conn_find(event->l2cap_event->connect.conn_handle, &desc);

        // NimBLE stores addresses in _reverse_ order. We need to reverse these
        // before doing the EUI64 conversion, otherwise we get incorrect
        // addresses in our IPv6 headers.
        uint8_t addr_buf[6];
        for (int i = 0; i < 6; ++i)
        {
            addr_buf[i] = desc.peer_id_addr.val[6 - 1 - i];
        }
        rfc7668_set_peer_addr_mac48(&rfc7668_netif, addr_buf, 6, 1);

        for (int i = 0; i < 6; ++i)
        {
            addr_buf[i] = desc.our_id_addr.val[6 - 1 - i];
        }
        rfc7668_set_local_addr_mac48(&rfc7668_netif, addr_buf, 6, 1);

        // Assign link-local IPv6 address based on our BLE address per RFC7668 requirements
        ip6_addr_t lladdr;
        ipv6_create_link_local_from_mac48(addr_buf, &lladdr, 1);
        ip_addr_copy_from_ip6(rfc7668_netif.ip6_addr[0], lladdr);
        ip6_addr_assign_zone(ip_2_ip6(&rfc7668_netif.ip6_addr[0]), IP6_UNICAST, &rfc7668_netif);
        netif_ip6_addr_set_state(&rfc7668_netif, 0, IP6_ADDR_PREFERRED);

        // Finally set the link up
        netif_set_up(&rfc7668_netif);
        netif_set_link_up(&rfc7668_netif);
        break;

    case BLE_L2CAP_EVENT_COC_DISCONNECTED:
        break;

    case BLE_L2CAP_EVENT_COC_DATA_RECEIVED:
        // Copy NimBLE's os_mbuf into the pbuf lwIP expects
        size_t rx_len = (size_t)OS_MBUF_PKTLEN(event->l2cap_event->receive.sdu_rx);
        struct pbuf* p = pbuf_alloc(PBUF_RAW, rx_len, PBUF_POOL);
        os_mbuf_copydata(event->l2cap_event->receive.sdu_rx, 0, rx_len, p->payload);
        p->len = rx_len;

        // Pass the data to our RFC7668 netif
        my_netif.input(p, &my_netif);

        // Free the os_mbuf and provide NimBLE with another for the next recv
        os_mbuf_free_chain(event->receive.sdu_rx);
        struct os_mbuf* next = os_mbuf_get_pkthdr(&m_mbuf_pool, 0);
        ble_l2cap_recv_ready(event->receive.chan, next);

        break;

    }
}
```
</details>

# Problem

This worked mint on some test firmware that did nothing but establish this 6LoWPAN connection: I was able to successfully pass UDP packets between some test devices.

I then added a Wi-Fi interface with ESP-NETIF and started seeing odd failures -- in particular, I'd find that the source addr would change in my udp_recv callbacks:

```c
static void
recv(void *arg, struct udp_pcb *upcb, struct pbuf *p, const ip_addr_t *addr, u16_t port)
{
    ip_addr_debug_print(addr);  // the IPv6 address from my 6lowpan peer, good
    // later in the same function...
    
    ip_addr_debug_print(addr);  // some IPv4 address from the Wi-Fi network I'm connected to
}
```

Which I suspect is because lwIP's `udp_*` API is not thread-safe. Now that I'm running the ESP-NETIF Wi-Fi event loop, it's processing udp input outside of the lwIP stuff I set up for RFC7668 communications.

# Attempted solution

This sounded like a good fit for an [ESP-NETIF Custom I/O Driver](https://docs.espressif.com/projects/esp-idf/en/release-v4.3/esp32/api-reference/network/esp_netif_driver.html). I'd just need to pull apart my proof of concept code to configure:
* all the NimBLE GAP/L2CAP configuration stuff as my I/O driver
* lwIP's RFC7668 functions as a network stack layer (e.g., I need to tell ESP-NETIF to use `rfc7668_if_init` as the init func instead of `wlanif_init`)

My thinking here is if I can wrap my code above using a custom I/O driver and a custom network stack layer, all the processing would happen in the ESP-NETIF event loop and I wouldn't have to worry about concurrent access to lwIP's non-threadsafe APIs.

I thought I'd define my own `esp_netif_netstack_config` to point to the RFC7668 functions:

```c
static const struct esp_netif_netstack_config my_rfc7668_config = {
        .lwip = {
            .init_fn = rfc7668_if_init,
            .input_fn = my_rfc7668_input_wrapper,
        }
};
```

However, I can't customize the network stack from my application code because `esp_netif_netstack_config` is only exposed outside of `esp_netif_lwip*` as an opaque pointer:

```c
    esp_netif_config_t cfg = {
      .base = MY_CUSTOM_INHERENT_BASE,
      .driver = NULL,
      .stack = ???  // I can't define my own version of this
    };
    esp_netif_t* esp_netif = esp_netif_new(&cfg);
```

The [latest ESP-IDF docs for custom I/O drivers](https://docs.espressif.com/projects/esp-idf/en/v5.2.1/esp32/api-reference/network/esp_netif_driver.html) suggest that this _is_ possible:

> In rare cases, expert users might want to define custom lwIP based interface layers; it is possible, but an explicit dependency to lwIP needs to be set.

but I'm not sure how to just add a single lwIP based interface layer. I don't want to get rid of all the existing layers (as I'd like to leverage the existing Wi-Fi layer at the same time!).

# Question
Is this the correct approach for what I'm attempting here? If so, how are we intended to add support for "custom lwIP based interface layers" to ESP-IDF?

david-cermak, 2024-04-15:
Hi @tomasgareau 

Yes, this is the correct approach. You just need to include:
```cpp
#include "lwip/esp_netif_net_stack.h"
```

You can also refer to this example, which creates a custom lwip netif:

https://github.com/espressif/esp-protocols/blob/bd6e12050928464b4357015c7c8c1b215432d580/examples/esp_netif/slip_custom_netif/components/slip_modem/library/slip_modem_netif.c#L161-L166

---
PS: Very interesting project, btw. If you succeed in implementing LoWPAN6-BLE on ESP-32, would you be willing to share some portions of your code?

tomasgareau, 2024-04-15:
Aha -- we're on an old version of ESP-IDF so I could not find that file. I'll take a look at upgrading to v5+ and take another shot at it.

----------------

RE: sharing code -- it's certainly something I'm open to but it is a work project. I'll discuss it with my employer and see where I get with the ESP-NETIF integration!

david-cermak, 2024-04-16:
yes, registering custom lwip IO functions wasn't possible in IDF < v5.0.
(still possible, if you redefine the opaque structs, like [this](https://github.com/david-cermak/eth-ap-nat/blob/2279344e18a0b98b5368999aac9441c59871e6fa/eth-ap-idf4.3/main/ethernet_example_main.c#L90-L96) )

tomasgareau, 2024-05-09:
Hey @david-cermak -- we were able to get this working in both IDF < v5.0 and >5.0, thanks! We can probably close this issue if you'd like.

----------------
RE: sharing the code -- my employer has OK'd sharing the resulting component. It's available here: https://github.com/geonavo/lowpan6_ble. It should roughly follow the structure of the `slip_custom_netif` example you linked above.

Our implementation is only really used for a pretty specific use-case (we only use it to connect to a non-ESP32 device and transfer UDP packets between the two). It is probably missing some more "general" use-case features (especially around the events bubbled up by the lowpan6_ble driver). However, we have added a UDP client/server echo example that does demonstrate accepting a LoWPAN6 BLE connection, too.

david-cermak, 2024-05-10:
> RE: sharing the code

Thanks so much for sharing this! Very helpful, even if it's tailored for your specific use case and :+1: for adding the server-client example!

---

I'm closing this issue (although I'm aware that developing such custom driver is not well documented in the netif docs; we've  already planned on updating and cleaning up the docs in this respect...)