espressif/esp-idf #13601: 为什么这里是两个esp_ble_gap_config_adv_data？

FlyyingPiggy2020, 2024-04-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

[gatts_demo](https://github.com/espressif/esp-idf/blob/5a40bb8746633477c07ff9a3e90016c37fa0dc0c/examples/bluetooth/bluedroid/ble/gatt_server/main/gatts_demo.c#L330-L336)
貌似确实是这样子写才可以工作，我自作聪明改成esp_ble_gap_set_scan_params反而不能工作了。。

zhp0406, 2024-04-15:
@FlyyingPiggy2020 
调用两次，一次设置 adv data，另一次设置 scan rsp data
[image]
[image]

zhp0406, 2024-04-15:
[image]

zhp0406, 2024-04-15:
只有当 adv data 和 scan rsp data 都设置完成，才可以广播。
你设置 esp_ble_gap_set_scan_params 想做什么？

FlyyingPiggy2020, 2024-04-15:
好的 ，了解了。

closed 2024-04-15 by FlyyingPiggy2020