espressif/esp-idf #13603: SSE Server-Sent-Events

CharlesMcBaker, 2024-04-13:
### Is your feature request related to a problem?

I tried creating Server-Sent-Events SSE from the examples withouth any succes.

shraiwi, 2024-04-21:
How did you end up implementing it? I'm currently trying to use SSE as well.

AdityaHPatwardhan, 2024-04-23:
Hi @CharlesMcBaker Thanks for the issue.
Based on your comment you have already successfully implemented the example.
Is it possible for you to share your code by raising a pull request in esp-idf.
It would be helpful for others who are interested in the same feature.
You can follow the [guidelines](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/contribute/index.html) for contributing to esp-idf. Please let me know if you need any help for raising up a PR.

AdityaHPatwardhan, 2024-11-18:
@gudvinr, Thanks for your response. 
To share an update from the last time, We have already implemented the SSE handler, the MR is in progress to be merged internally.
Thanks for your inputs.

closed 2024-12-13 by espressif-bot

chegewara, 2025-04-23:
Hi,
it seems to work nicely, event reconnecting, just i have one question:
- when for some reason connection is lost and data are still pushed to event source request (i am testing with P4 and usb-netif now and i unplug usb cable to simulate connection lost), the data is buffered and sent after reconnecting
- the question is for how long data are buffered, how connection timeout is detected or handled, how to prevent memory issues in such case

Thanks