espressif/esp-idf #13605: ESP-IDF v4.3.7 - BLE Crashes during connection parameter update

expresspotato, 2024-04-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.3.7

### Espressif SoC revision.

ESP32-Pico-V3-02

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

LyraT

### Power Supply used.

USB

### What is the expected behavior?

Not crash

### What is the actual behavior?

Crashes

### Steps to reproduce.

1. Build a co-ex project
2. Connect via HFP, A2D, AVRC, BLE
3. Change the connection parameters, relatively often (1-2 times per minute) as needed between an idle / active state
4. Observe crash

### Debug Logs.

```plain
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40015788  PS      : 0x00060d30  A0      : 0x80049eba  A1      : 0x3ffdc7c0  
A2      : 0x3ffdc7f0  A3      : 0x3ffdc808  A4      : 0x000000ff  A5      : 0x00000001  
A6      : 0x00000003  A7      : 0x00000001  A8      : 0x0014f00f  A9      : 0x00000000  
A10     : 0x000000a6  A11     : 0x000000a6  A12     : 0x00000000  A13     : 0x000000c0  
A14     : 0x3ffd9f04  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0014f013  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace:0x40015785:0x3ffdc7c0 0x40049eb7:0x3ffdc7f0 0x40164cf6:0x3ffdc840 0x40042386:0x3ffdc870 0x40089766:0x3ffdc890 0x40019d11:0x3ffdc8c0 0x40055b4d:0x3ffdc8e0 0x4015b547:0x3ffdc900 0x4015bbca:0x3ffdc920
0x40164cf6: hci_le_rem_con_param_req_reply_cmd_handler at llc_hci.c:?

0x40089766: ke_task_schedule at ke_task.c:?

0x4015b547: r_rw_schedule at ??:?

0x4015bbca: btdm_controller_task at ??:?

ELF file SHA256: 7f571c4f7e7eff18

I (1469) esp_core_dump_flash: Save core dump to flash...
I (1475) esp_core_dump_flash: Erase flash 36864 bytes @ 0x7e0000
I (1701) esp_core_dump_flash: Write end offset 0x86a4, check sum length 4
I (1701) esp_core_dump_flash: Core dump has been saved to flash.
```

### More Information.

Forum topic: https://esp32.com/viewtopic.php?f=28&t=37739

zhp0406, 2024-04-16:
@expresspotato 
thanks for reporting the bug,we will check it.

expresspotato, 2024-04-28:
> @expresspotato thanks for reporting the bug,we will check it.

Hi @zhp0406,

Were you able to determine the root cause and a fix for this issue?

zhp0406, 2024-04-29:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7192
load:0x40078000,len:15564
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3904
entry 0x40080640
[0;32mI (31) boot: ESP-IDF v5.4-dev-120-g1b95af2096-dirty 2nd stage bootloader[0m
[0;32mI (31) boot: compile time Apr 29 2024 15:59:36[0m
[0;32mI (33) boot: Multicore bootloader[0m
[0;32mI (37) boot: chip revision: v3.1[0m
[0;32mI (41) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (46) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (50) boot.esp32: SPI Flash Size : 2MB[0m
[0;32mI (55) boot: Enabling RNG early entropy source...[0m
[0;32mI (60) boot: Partition Table:[0m
[0;32mI (64) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (86) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (94) boot: End of partition table[0m
[0;32mI (98) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2f310h (193296) map[0m
[0;32mI (173) esp_image: segment 1: paddr=0003f338 vaddr=3ffbdb60 size=00ce0h (  3296) load[0m
[0;32mI (174) esp_image: segment 2: paddr=00040020 vaddr=400d0020 size=a9564h (693604) map[0m
[0;32mI (416) esp_image: segment 3: paddr=000e958c vaddr=3ffbe840 size=04274h ( 17012) load[0m
[0;32mI (423) esp_image: segment 4: paddr=000ed808 vaddr=40080000 size=1aa94h (109204) load[0m
[0;32mI (479) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (479) boot: Disabling RNG early entropy source...[0m
[0;32mI (491) cpu_start: Multicore app[0m
[0;32mI (500) cpu_start: Pro cpu start user code[0m
[0;32mI (500) cpu_start: cpu freq: 160000000 Hz[0m
[0;32mI (500) app_init: Application information:[0m
[0;32mI (503) app_init: Project name:     bt_ble_coex[0m
[0;32mI (508) app_init: App version:      v5.4-dev-120-g1b95af2096-dirty[0m
[0;32mI (515) app_init: Compile time:     Apr 29 2024 15:59:32[0m
[0;32mI (521) app_init: ELF file SHA256:  7c4f2dacf...[0m
[0;32mI (526) app_init: ESP-IDF:          v5.4-dev-120-g1b95af2096-dirty[0m
[0;32mI (533) efuse_init: Min chip rev:     v0.0[0m
[0;32mI (538) efuse_init: Max chip rev:     v3.99 [0m
[0;32mI (542) efuse_init: Chip rev:         v3.1[0m
[0;32mI (548) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (555) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (561) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM[0m
[0;32mI (567) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM[0m
[0;32mI (573) heap_init: At 3FFCC258 len 00013DA8 (79 KiB): DRAM[0m
[0;32mI (579) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (585) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (592) heap_init: At 4009AA94 len 0000556C (21 KiB): IRAM[0m
[0;32mI (600) spi_flash: detected chip: generic[0m
[0;32mI (603) spi_flash: flash io: dio[0m
[0;33mW (606) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.[0m
[0;32mI (621) coexist: coex firmware version: 7fff68fa5[0m
[0;32mI (626) main_task: Started on CPU0[0m
[0;32mI (636) main_task: Calling app_main()[0m
[0;32mI (646) BTDM_INIT: BT controller compile version [4012cfb][0m
[0;32mI (646) BTDM_INIT: Bluetooth MAC: 08:3a:8d:15:0b:5e[0m
[0;32mI (656) phy_init: phy_version 4791,2c4672b,Dec 20 2023,16:06:06[0m
[0;33mW (1346) BT_BTC: A2DP Enable with AVRC[0m
[0;32mI (1356) BT_AV: A2DP PROF STATE: Init Complete[0m
[0;32mI (1366) BT_BLE_COEX: event: 10[0m
[0;32mI (1366) BT_BLE_COEX: event: 10[0m
[0;32mI (1366) BT_BLE_COEX: event: 10[0m
[0;32mI (1366) BT_AV: Get delay report value: delay_value: 1200 * 1/10 ms[0m
[0;32mI (1376) BT_BLE_COEX: REGISTER_APP_EVT, status 0, app_id 0[0m
[0;32mI (1386) BT_AV: Set delay report value: success, delay_value: 1250 * 1/10 ms[0m
[0;32mI (1386) BT_BLE_COEX: CREATE_SERVICE_EVT, status 0,  service_handle 40[0m
[... 26 lines cut ...]
[0;32mI (35346) BT_BLE_COEX: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 35,latency = 0, timeout = 400[0m
[0;32mI (36446) BT_BLE_COEX: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 36,latency = 0, timeout = 400[0m
[0;32mI (64226) BT_BLE_COEX: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 37,latency = 0, timeout = 400[0m
[0;32mI (65246) BT_BLE_COEX: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 38,latency = 0, timeout = 400[0m
[0;32mI (66256) BT_BLE_COEX: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 39,latency = 0, timeout = 400[0m
[0;32mI (67336) BT_BLE_COEX: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 40,latency = 0, timeout = 400[0m
[0;32mI (68246) BT_BLE_COEX: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 41,latency = 0, timeout = 400[0m
[0;32mI (73376) BT_BLE_COEX: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 42,latency = 0, timeout = 400[0m
[... cut ...]
[0;32mI (212896) BT_BLE_COEX: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 55,latency = 0, timeout = 400[0m
[0;32mI (213926) BT_BLE_COEX: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 56,latency = 0, timeout = 400[0m
[0;32mI (214916) BT_BLE_COEX: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 57,latency = 0, timeout = 400[0m
[0;32mI (235446) BT_BLE_COEX: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 58,latency = 0, timeout = 400[0m
```

expresspotato, 2024-04-30:
Well there are other issues that are unresolved in 4.4.x, namely https://github.com/espressif/esp-idf/issues/11315, and a few others... 

Our project is very large, so it's not easy to switch between 4.3.7 and 4.4.x (nor 5.x anything for that matter) without extensive testing. To my knowledge v.4.3.5 didn't have this issue.

I see in your test that A2DP, HFP and AVRC aren't connected yet. Can you try and connect those first?

zhp0406, 2024-04-30:
Do you mean this issue only occurs when A2DP, HFP, and AVRC are connected? I don't have A2DP, HFP, and AVRC connected here, only BLE connections. 
Can you test whether the issue can be reproduced with only BLE connections?

zhp0406, 2024-04-30:
If you have a demo that reproduces the issue , it would be helpful for me to resolve the problem.”

expresspotato, 2024-05-08:
> Do you mean this issue only occurs when A2DP, HFP, and AVRC are connected? I don't have A2DP, HFP, and AVRC connected here, only BLE connections. Can you test whether the issue can be reproduced with only BLE connections?

Upon further observation, the crash occurs during an indicate. The parameters are not yet changed, and an indicate is sent from the ESP32 (GATTS) -> to an Android phone (GATTC) during the update process.

One instance:

I (74413) LIBERTY: 8 Bit --> 4077935
I (74418) LIBERTY: Internal --> 82228
I (74423) LIBERTY: SPI RAM --> 3999095
I (74428) LIBERTY: Largest free block in Internal --> 69620
I (74433) LIBERTY: Largest free block in SPI RAM --> 3932148
W (74438) LIBERTY: --> Memory used (After POST connection tasks): 0 / 20
W (74448) LIBERTY: --> BLE connection params set: Idle
E (74453) LIBERTY: --> Finished, sending back item... is threaded (1)
E (74463) LIBERTY: --> Dispatch callback!
E (74468) LIBERTY: --> CBBP Command Response Sent (1)
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40015788  PS      : 0x00060830  A0      : 0x80049eba  A1      : 0x3ffdc720  
A2      : 0x3ffdc750  A3      : 0x3ffdc768  A4      : 0x000000ff  A5      : 0x00000001  
A6      : 0x00000003  A7      : 0x00000001  A8      : 0x0014f00f  A9      : 0x00000000  
A10     : 0x0000008f  A11     : 0x0000008f  A12     : 0x00000000  A13     : 0x000000c0  
A14     : 0x3ffd9e18  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0014f013  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace:0x40015785:0x3ffdc720 0x40049eb7:0x3ffdc750 0x40162492:0x3ffdc7a0 0x40042386:0x3ffdc7d0 0x40087ece:0x3ffdc7f0 0x40019d11:0x3ffdc820 0x40055b4d:0x3ffdc840 0x40158ce7:0x3ffdc860 0x4015936a:0x3ffdc880
0x40162492: hci_le_rem_con_param_req_reply_cmd_handler at llc_hci.c:?

0x40087ece: ke_task_schedule at ke_task.c:?

0x40158ce7: r_rw_schedule at ??:?

0x4015936a: btdm_controller_task at ??:?

ELF file SHA256: 83803e802dc23644

I (2981) esp_core_dump_flash: Save core dump to flash...
I (2987) esp_core_dump_flash: Erase flash 28672 bytes @ 0x7e0000
I (3186) esp_core_dump_flash: Write end offset 0x6b44, check sum length 4
I (3186) esp_core_dump_flash: Core dump has been saved to flash.
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7592
load:0x40078000,len:15860
load:0x40080400,len:4416
0x40080400: _init at ??:?

Another:

E (307599) LIBERTY: --> BLE request active
W (307599) LIBERTY: --> BLE connection params set: Active
E (307649) LIBERTY: --> CBBP Request over BT (87684 bytes)
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40015788  PS      : 0x00060a30  A0      : 0x80049eba  A1      : 0x3ffdc720  
A2      : 0x3ffdc750  A3      : 0x3ffdc768  A4      : 0x000000ff  A5      : 0x00000001  
A6      : 0x00000003  A7      : 0x00000001  A8      : 0x0014f00f  A9      : 0x00000000  
A10     : 0x0000000f  A11     : 0x0000000f  A12     : 0x00000000  A13     : 0x00000018  
A14     : 0x3ffd9e18  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0014f013  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace:0x40015785:0x3ffdc720 0x40049eb7:0x3ffdc750 0x40162492:0x3ffdc7a0 0x40042386:0x3ffdc7d0 0x40087ece:0x3ffdc7f0 0x40019d11:0x3ffdc820 0x40055b4d:0x3ffdc840 0x40158ce7:0x3ffdc860 0x4015936a:0x3ffdc880
0x40162492: hci_le_rem_con_param_req_reply_cmd_handler at llc_hci.c:?

0x40087ece: ke_task_schedule at ke_task.c:?

0x40158ce7: r_rw_schedule at ??:?

0x4015936a: btdm_controller_task at ??:?

ELF file SHA256: 83803e802dc23644

I (4085) esp_core_dump_flash: Save core dump to flash...
I (4091) esp_core_dump_flash: Erase flash 28672 bytes @ 0x7e0000
I (4284) esp_core_dump_flash: Write end offset 0x6ec4, check sum length 4
I (4284) esp_core_dump_flash: Core dump has been saved to flash.
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7592
load:0x40078000,len:15860
load:0x40080400,len:4416
0x40080400: _init at ??:?

entry 0x400806a0

expresspotato, 2024-09-30:
Hey @esp-zhp , so this is not fixed on v4.4.8... We checked and we really don't do anything bizzare in the code leading up to, or after the connection parameter update. 

We simply make the request and even wait for the callback with the updated parameters before sending another GATT indicate. 

Poking around, L2CA_UpdateBleConnParams does return true from the call stack, so it seems like the bug is on Espressif's side. 

Since the source code is closed, I can't debug it further... What happens in hci_le_rem_con_param_req_reply_cmd_handler?

Without HFP connected, I'm unable to reproduce this bug, but I gave up after 50 attempts. With HFP connected, it will crash in in 15 - 20 attempts. 

Can you try with HFP connected and send data over a ble characteristic through writes?

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40015788  PS      : 0x00060430  A0      : 0x80049eba  A1      : 0x3ffdf690  
A2      : 0x3ffdf6c0  A3      : 0x3ffdf6d8  A4      : 0x000000ff  A5      : 0x00000001  
A6      : 0x00000003  A7      : 0x00000001  A8      : 0xfcabba12  A9      : 0x00000000  
A10     : 0x00000064  A11     : 0x00000064  A12     : 0x00000000  A13     : 0x000000c0  
A14     : 0x3ffd1f80  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xfcabba16  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace: 0x40015785:0x3ffdf690 0x40049eb7:0x3ffdf6c0 0x4021ed8e:0x3ffdf710 0x40042386:0x3ffdf740 0x40090256:0x3ffdf760 0x40019d11:0x3ffdf790 0x40055b4d:0x3ffdf7b0 0x40214103:0x3ffdf7d0 0x40214761:0x3ffdf7f0
0x4021ed8e: hci_le_rem_con_param_req_reply_cmd_handler at llc_hci.c:?

0x40090256: ke_task_schedule at ke_task.c:?

0x40214103: r_rw_schedule at ??:?

0x40214761: btdm_controller_task at ??:?

```

esp-zhp, 2024-10-09:
@expresspotato 
could you give me a demo to reproduces the issue?

expresspotato, 2024-11-20:
Interesting. So I've updated to v5.1.5 and the issue persists, in a slightly different manner. The stack still refers to `hci_le_rem_con_param_req_reply_cmd_handler`

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40015788  PS      : 0x00060430  A0      : 0x802286c5  A1      : 0x3ffe2cd0  
--- 0x40015788: r_ea_offset_req in ROM

A2      : 0x3ffe2d20  A3      : 0x3ffe2d38  A4      : 0x000000ff  A5      : 0x00000001  
A6      : 0x00000003  A7      : 0x00000001  A8      : 0x0014f00f  A9      : 0x00000000  
A10     : 0x00000017  A11     : 0x00000017  A12     : 0x00000000  A13     : 0x00000018  
A14     : 0x3ffdfc30  A15     : 0x3ffdff80  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0014f013  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  
--- 0x4000c46c: memset in ROM
0x4000c477: memset in ROM

Backtrace: 0x40015785:0x3ffe2cd0 0x402286c2:0x3ffe2d00 0x40049eb7:0x3ffe2d20 0x40231972:0x3ffe2d70 0x40042386:0x3ffe2da0 0x400919ca:0x3ffe2dc0 0x40019d11:0x3ffe2df0 0x40055b4d:0x3ffe2e10 0x402266bf:0x3ffe2e30 0x40226d21:0x3ffe2e50 0x4009ea05:0x3ffe2e80
--- 0x40015785: r_ea_offset_req in ROM
0x402286c2: r_ea_offset_req_hack at ??:?
0x40049eb7: r_lld_con_param_rsp in ROM
0x40231972: hci_le_rem_con_param_req_reply_cmd_handler at llc_hci.c:?
0x40042386: llc_hci_command_handler in ROM
0x400919ca: ke_task_schedule at ke_task.c:?
0x40019d11: r_ke_event_schedule in ROM
0x40055b4d: r_rwip_schedule in ROM
0x402266bf: r_rw_schedule at ??:?
0x40226d21: btdm_controller_task at ??:?
0x4009ea05: vPortTaskWrapper at /Users/kevin/esp/esp-idf-v5.1.5/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

esp-zhp, 2024-11-21:
Yes, we haven’t resolved this issue yet. I urgently need a demo from you to reproduce the problem.

expresspotato, 2024-12-26:
So I've finally had the time to produce an example for you. The code does crash **infrequently**, so you may have to reboot the device and make several attempts.

Overall, there isn't anything special going on. 

```
kevin@Kevin-Parkers-MacBook-Pro a2dp_gatts_coex % idf.py build && idf.py flash -p /dev/tty.usbserial-1120 && idf.py monitor -p /dev/tty.usbserial-1120
Executing action: all (aliases: build)
Running ninja in directory /Users/kevin/esp/esp-idf-v5.1.5/examples/bluetooth/bluedroid/coex/a2dp_gatts_coex/build
Executing "ninja all"...
[1/1] cd /Users/kevin/esp/esp-idf-v5.1.5/examples/bluetooth/bluedroid/coex/a2dp_gatts_coex/build/bootloader/esp-idf/e...000 /Users/kevin/esp/esp-idf-v5.1.5/examples/bluetooth/bluedroid/coex/a2dp_gatts_coex/build/bootloader/bootloader.binBootloader binary size 0x6860 bytes. 0x7a0 bytes (7%) free.
[8/9] Generating binary image from built executableesptool.py v4.7.0
Creating esp32 image...
Merged 27 ELF sections
Successfully created esp32 image.
Generated /Users/kevin/esp/esp-idf-v5.1.5/examples/bluetooth/bluedroid/coex/a2dp_gatts_coex/build/bt_ble_coex.bin
[9/9] cd /Users/kevin/esp/esp-idf-v5.1.5/examples/bluetooth/bluedroid/coex/a2dp_gatts_coex/build/esp-idf/esptool_py &...ion-table.bin /Users/kevin/esp/esp-idf-v5.1.5/examples/bluetooth/bluedroid/coex/a2dp_gatts_coex/build/bt_ble_coex.binbt_ble_coex.bin binary size 0xf42c0 bytes. Smallest app partition is 0x100000 bytes. 0xbd40 bytes (5%) free.
Warning: The smallest app partition is nearly full (5% free space left)!

Project build complete. To flash, run this command:
/Users/kevin/.espressif/python_env/idf5.1_py3.10_env/bin/python ../../../../../../esp-idf/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size 2MB --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/bt_ble_coex.bin
or run 'idf.py -p (PORT) flash'
Executing action: flash
Running ninja in directory /Users/kevin/esp/esp-idf-v5.1.5/examples/bluetooth/bluedroid/coex/a2dp_gatts_coex/build
Executing "ninja flash"...
[1/5] cd /Users/kevin/esp/esp-idf-v5.1.5/examples/bluetooth/bluedroid/coex/a2dp_gatts_coex/build/esp-idf/esptool_py && /Users/kevin/.espressif/python_env/idf5.1_py3.10_env/bin/python /Users/kevin/esp/esp-idf-v5.1.5/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /Users/kevin/esp/esp-idf-v5.1.5/examples/bluetooth/bluedroid/coex/a2dp_gatts_coex/build/partition_table/partition-table.bin /Users/kevin/esp/esp-idf-v5.1.5/examples/bluetooth/bluedroid/coex/a2dp_gatts_coex/build/bt_ble_coex.bin
bt_ble_coex.bin binary size 0xf42c0 bytes. Smallest app partition is 0x100000 bytes. 0xbd40 bytes (5%) free.
Warning: The smallest app partition is nearly full (5% free space left)!
[2/5] Performing build step for 'bootloader'
[1/1] cd /Users/kevin/esp/esp-idf-v5.1.5/examples/bluetooth/bluedroid/coex/a2dp_gatts_coex/build/bootloader/esp-idf/esptool_py && /Users/kevin/.espressif/python_env/idf5.1_py3.10_env/bin/python /Users/kevin/esp/esp-idf-v5.1.5/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /Users/kevin/esp/esp-idf-v5.1.5/examples/bluetooth/bluedroid/coex/a2dp_gatts_coex/build/bootloader/bootloader.bin
Bootloader binary size 0x6860 bytes. 0x7a0 bytes (7%) free.
[3/5] No install step for 'bootloader'
[4/5] Completed 'bootloader'
[4/5] cd /Users/kevin/esp/esp-idf-v5.1.5/components/esptool_py && /opt/homebrew/bin/cmake -D IDF_PATH=/Users/kevin/esp/esp-idf-v5.1.5 -D "SERIAL_TOOL=/Users/kevin/.espressif/python_env/idf5.1_py3.10_env/bin/python;;/Users/kevin/esp/esp-idf-v5.1.5/components/esptool_py/esptool/esptool.py;--chip;esp32" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/Users/kevin/esp/esp-idf-v5.1.5/examples/bluetooth/bluedroid/coex/a2dp_gatts_coex/build -P /Users/kevin/esp/esp-idf-v5.1.5/components/esptool_py/run_serial_tool.cmake
esptool.py --chip esp32 -p /dev/tty.usbserial-1120 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x10000 bt_ble_coex.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.7.0
Serial port /dev/tty.usbserial-1120
Connecting......
Chip is ESP32-PICO-V3-02 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, Embedded PSRAM, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 10:06:1c:17:b3:90
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00010000 to 0x00104fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 26720 bytes to 16732...
Writing at 0x00001000... (50 %)
Writing at 0x000076a1... (100 %)
Wrote 26720 bytes (16732 compressed) at 0x00001000 in 0.9 seconds (effective 226.7 kbit/s)...
Hash of data verified.
Compressed 1000128 bytes to 593057...
Writing at 0x00010000... (2 %)
Writing at 0x0001d747... (5 %)
Writing at 0x0002cb57... (8 %)
Writing at 0x0003936a... (10 %)
Writing at 0x0004357f... (13 %)
Writing at 0x00048dd0... (16 %)
Writing at 0x0004eccb... (18 %)
Writing at 0x00054c3c... (21 %)
Writing at 0x0005a5b5... (24 %)
Writing at 0x0005feea... (27 %)
Writing at 0x000653ef... (29 %)
Writing at 0x0006b130... (32 %)
[... cut ...]
0x40042386: llc_hci_command_handler in ROM
0x4008a03e: ke_task_schedule at ke_task.c:?
0x40019d11: r_ke_event_schedule in ROM
0x40055b4d: r_rwip_schedule in ROM
0x4014b0af: r_rw_schedule at ??:?
0x4014b711: btdm_controller_task at ??:?
0x4009511d: vPortTaskWrapper at /Users/kevin/esp/esp-idf-v5.1.5/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: 0f74c65b09e1c3ba

Rebooting...
```

[a2dp_gatts_coex.zip](https://github.com/user-attachments/files/18250471/a2dp_gatts_coex.zip)

expresspotato, 2025-01-04:
Hi @esp-zhp, any progress on fixing this?

esp-zhp, 2025-01-10:
@expresspotato nogoing...will fix

esp-zhp, 2025-01-12:
[libbtdm_app.zip](https://github.com/user-attachments/files/18389326/libbtdm_app.zip)
[IDFGH_12607_01_12_5.log](https://github.com/user-attachments/files/18389328/IDFGH_12607_01_12_5.log)

expresspotato, 2025-01-13:
@esp-zhp thank you. I'll integrate this and test over the next day or so.

expresspotato, 2025-01-14:
Hey @esp-zhp, the bug appears to be fixed. There have been no occurrences over the past 24 hours on my two test devices.

expresspotato, 2025-01-15:
Hey @esp-zhp,

So looks like the bug is not fixed, but appears at this less frequent stack:

```
E (6587561) LIBERTY: --> Debug run count: 556
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40235969  PS      : 0x00060230  A0      : 0x80238e0a  A1      : 0x3ffe13e0  
--- 0x40235969: r_ea_interval_duration_req at ??:?

A2      : 0x3ffe1400  A3      : 0x3ffe1418  A4      : 0x3ffb8360  A5      : 0x07280000  
A6      : 0x00000000  A7      : 0x00000002  A8      : 0x0000001e  A9      : 0x00360001  
A10     : 0x00000036  A11     : 0x0000001e  A12     : 0x00000002  A13     : 0x0000001e  
A14     : 0x00000004  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0036000b  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  
--- 0x4000c46c: memset in ROM
0x4000c477: memset in ROM

Backtrace: 0x40235966:0x3ffe13e0 0x40238e07:0x3ffe1400 0x40028d42:0x3ffe1460 0x40092a16:0x3ffe14a0 0x40019d11:0x3ffe14d0 0x40055b4d:0x3ffe14f0 0x402339fb:0x3ffe1510 0x4023405d:0x3ffe1530 0x4009f8f9:0x3ffe1560
--- 0x40235966: r_ea_interval_duration_req at ??:?
0x40238e07: r_LM_AddSniff at ??:?
0x40028d42: lc_op_loc_sniff_req_handler in ROM
0x40092a16: ke_task_schedule at ke_task.c:?
0x40019d11: r_ke_event_schedule in ROM
0x40055b4d: r_rwip_schedule in ROM
0x402339fb: r_rw_schedule at ??:?
0x4023405d: btdm_controller_enable at ??:?
0x4009f8f9: vTaskStartScheduler at /Users/kevin/esp/esp-idf-v5.1.5/components/freertos/FreeRTOS-Kernel/tasks.c:2617
```

BetterJincheng, 2025-01-15:
@expresspotato please attch the elf file to help to analyze on the crash.

expresspotato, 2025-01-18:
Hi @BetterJincheng ,

So I can't upload an elf of our project, but should be able to get you one of the crash within the coming days.

In addition to the one above (after the fix) this crash appeared today:

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4022384d  PS      : 0x00060530  A0      : 0x80049e4c  A1      : 0x3ffd37a0  
--- 0x4022384d: r_ea_interval_duration_req at ??:?

A2      : 0x3ffd37c0  A3      : 0x3ffd37d8  A4      : 0x00060520  A5      : 0x400990e4  
--- 0x400990e4: r_global_int_restore at ??:?

A6      : 0x00000000  A7      : 0x00000002  A8      : 0x00009310  A9      : 0x0000000c  
A10     : 0x000001e0  A11     : 0x00009310  A12     : 0x00000002  A13     : 0x00000180  
A14     : 0x0000000c  A15     : 0x3ffcf764  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000016  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
--- 0x4000c2e0: memcpy in ROM
0x4000c2f6: memcpy in ROM

Backtrace: 0x4022384a:0x3ffd37a0 0x40049e49:0x3ffd37c0 0x4022ccf6:0x3ffd3810 0x40042386:0x3ffd3840 0x4009281e:0x3ffd3860 0x40019d11:0x3ffd3890 0x40055b4d:0x3ffd38b0 0x402218df:0x3ffd38d0 0x40221f41:0x3ffd38f0 0x4009d09d:0x3ffd3920
--- 0x4022384a: r_ea_interval_duration_req at ??:?
0x40049e49: r_lld_con_param_rsp in ROM
0x4022ccf6: hci_le_rem_con_param_req_reply_cmd_handler at llc_hci.c:?
0x40042386: llc_hci_command_handler in ROM
0x4009281e: ke_task_schedule at ke_task.c:?
0x40019d11: r_ke_event_schedule in ROM
0x40055b4d: r_rwip_schedule in ROM
0x402218df: r_rw_schedule at ??:?
0x40221f41: btdm_controller_task at ??:?
0x4009d09d: vPortTaskWrapper at /Users/kevin/esp/esp-idf-v5.1.5/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: aba8e6a6f18b2dea
```

expresspotato, 2025-02-07:
Hey @BetterJincheng ,

Attached is an example that continues to crash...

**Installing**

1) Use esptool to erase the flash completely
2) Install the binary, or compile from the included source onto a LytaT or equivalent (ESP32)
3) Observe the log, it should stay booted and say `Finished main!`
4) At this point the board is advertising BLE and Bluetooth Classic

1) Install the Android app, either by adb install test-crash.apk or by compiling and running it
2) Allow the permission for nearby devices
3) Enter the mac address in the format of 00:00:00:00:00:00. See the log for it as `I (1838) BTDM_INIT: Bluetooth MAC: 08:d1:f9:5b:65:e6`. Note the WiFi and BT mac's differ.
4) Tap connect
5) You should see two pairing requests
6) Then the screen should show "Connecting..."
7) Wait until the bluetooth stack crashes

**What this test does**
1) Connects to the ESP32 (GATTS) from the Android phone (GATTC)
2) ESP32 sends indications constantly
3) Android phone sends indications constantly
4) ESP32 disconnects and reconnects HFP / A2D every 30 seconds or so, the bug appears sooner this way
5) One of **two** distinct crashes will appear

[crash-test-esp32.zip](https://github.com/user-attachments/files/18712013/crash-test-esp32.zip)

**Crash Stacks**

There appear to be two distinct stacks ending in the same function.

_Stack One_
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40157608  PS      : 0x00060c30  A0      : 0x8015770e  A1      : 0x3ffcde80  
--- 0x40157608: r_ea_offset_req at ??:?

A2      : 0x3ffcded0  A3      : 0x3ffcdee8  A4      : 0x000000ff  A5      : 0x00000001  
A6      : 0x3ffcb3e0  A7      : 0x00000001  A8      : 0x0000001b  A9      : 0x00000017  
A10     : 0x00000018  A11     : 0x00010441  A12     : 0x00060820  A13     : 0x0000000b  
A14     : 0x00000018  A15     : 0x00000003  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00010445  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  
--- 0x4000c46c: memset in ROM
0x4000c477: memset in ROM

Backtrace: 0x40157605:0x3ffcde80 0x4015770b:0x3ffcdeb0 0x40049eb7:0x3ffcded0 0x401609be:0x3ffcdf20 0x40042386:0x3ffcdf50 0x4008d58e:0x3ffcdf70 0x40019d11:0x3ffcdfa0 0x40055b4d:0x3ffcdfc0 0x40155617:0x3ffcdfe0 0x40155c75:0x3ffce000 0x40096bdd:0x3ffce030
--- 0x40157605: r_ea_offset_req at ??:?
0x4015770b: r_ea_offset_req_hack at ??:?
0x40049eb7: r_lld_con_param_rsp in ROM
0x401609be: hci_le_rem_con_param_req_reply_cmd_handler at llc_hci.c:?
0x40042386: llc_hci_command_handler in ROM
0x4008d58e: ke_task_schedule at ke_task.c:?
0x40019d11: r_ke_event_schedule in ROM
0x40055b4d: r_rwip_schedule in ROM
0x40155617: r_rw_schedule at ??:?
0x40155c75: btdm_controller_task at ??:?
0x40096bdd: vPortTaskWrapper at /Users/kevin/esp/esp-idf-v5.1.5/components/freertos/FreeRTOS
-Kernel/portable/xtensa/port.c:162
```

_Stack Two_
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4022dae9  PS      : 0x00060d30  A0      : 0x80230f92  A1      : 0x3ffe13e0  
--- 0x4022dae9: r_ea_interval_duration_req at ??:?

A2      : 0x3ffe1400  A3      : 0x3ffe1418  A4      : 0x3ffb8360  A5      : 0x05f80000  
A6      : 0x00000000  A7      : 0x00000002  A8      : 0x00000018  A9      : 0x000000bf  
A10     : 0x00000320  A11     : 0x00000018  A12     : 0x00000002  A13     : 0x00000190  
A14     : 0x0000000c  A15     : 0x00000001  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000000c9  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  
--- 0x4000c46c: memset in ROM
0x4000c477: memset in ROM

Backtrace: 0x4022dae6:0x3ffe13e0 0x40230f8f:0x3ffe1400 0x40028d42:0x3ffe1460 0x4009307a:0x3ffe14a0 0x40019d11:0x3ffe14d0 0x40055b4d:0x3ffe14f0 0x4022bb7b:0x3ffe1510 0x4022c1dd:0x3ffe1530 0x4009d9f5:0x3ffe1560
--- 0x4022dae6: r_ea_interval_duration_req at ??:?
0x40230f8f: r_LM_AddSniff at ??:?
0x40028d42: lc_op_loc_sniff_req_handler in ROM
0x4009307a: ke_task_schedule at ke_task.c:?
0x40019d11: r_ke_event_schedule in ROM
0x40055b4d: r_rwip_schedule in ROM
0x4022bb7b: r_rw_schedule at ??:?
0x4022c1dd: btdm_controller_task at ??:?
0x4009d9f5: vPortTaskWrapper at /Users/kevin/esp/esp-idf-v5.1.5/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

**Files**

[crash-test-esp32.zip](https://github.com/user-attachments/files/18712020/crash-test-esp32.zip)
[crash-test-android.zip](https://github.com/user-attachments/files/18712026/crash-test-android.zip)

1) Zipped esp-32 project, crash-test-esp32.zip (includes liberty.bin, liberty.elf) for LytraT
2) Zipped android project, crash-test-android.zip (includes an .apk)

BetterJincheng, 2025-02-08:
@expresspotato Can you provide me with a simple APK installation package? The app debug seems to have failed to install on my phone.

BetterJincheng, 2025-02-08:
@expresspotato Using the attached example，I can only find the following error, no above crashes found.

[image]

BetterJincheng, 2025-02-08:
@expresspotato after a long time, the crash occurs. We are debug on it now.

expresspotato, 2025-02-08:
Hey, @BetterJincheng, thanks for confirming. Usually I'm able to re-produce the crashes within a minutes. 

If you see the CPU watchdog timeout, there is something wrong... I would suggest re-setting the board and app and trying again once paired for both BLE and BT Classic. 

As requested attached is a pre-compiled APK

[app-debug.apk.zip](https://github.com/user-attachments/files/18719423/app-debug.apk.zip)

You can also build the Android App from source, its' been hacked away and much of the functionality removed from a much larger application.

BetterJincheng, 2025-02-11:
@expresspotato We have resolved some issues. Please test with this lib.

[libbtdm_app.zip](https://github.com/user-attachments/files/18747346/libbtdm_app.zip)

expresspotato, 2025-02-11:
Hey @Betterjincheng,

Thank you. There was a problem during link time, as a ROM function was missing. I added the one found in master [here](https://github.com/espressif/esp-idf/blob/master/components/esp_rom/esp32/ld/esp32.rom.ld) for `PROVIDE ( ld_acl_rsw_evt_start_cbk = 0x40031154 );`, but I don't know if this is the correct procedure for adding references to ROM functions, nor if any related code is ever updated.

The board is flashed, and I'll report back within a few days time.

```
[1786/1788] Linking CXX executable liberty.elfFAILED: liberty.elf 
: && /Users/kevin/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -mlongcalls -Wno-frame-address -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32=0 -Wl,--Map=/Users/kevin/workspace/liberty_device/build/liberty.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32.peripherals.ld -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32.rom.newlib-funcs.ld -T esp32.rom.eco3.ld -T memory.ld -T sections.ld -T ulp_main.ld CMakeFiles/liberty.elf.dir/project_elf_src_esp32.c.obj -o liberty.elf -L/Users/kevin/esp/esp-idf-v5.1.5/components/soc/esp32/ld   -L/Users/kevin/esp/esp-idf-v5.1.5/components/esp_rom/esp32/ld   -L/Users/kevin/workspace/liberty_device/build/esp-idf/esp_system/ld   -L/Users/kevin/esp/esp-idf-v5.1.5/components/esp_phy/lib/esp32   -L/Users/kevin/esp/esp-idf-v5.1.5/components/esp_coex/lib/esp32   -L/Users/kevin/esp/esp-idf-v5.1.5/components/esp_wifi/lib/esp32   -L/Users/kevin/esp/esp-idf-v5.1.5/components/bt/controller/lib_esp32/esp32   -L/Users/kevin/workspace/liberty_device/build/esp-idf/main/ulp_main esp-idf/xtensa/libxtensa.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/app_trace/libapp_trace.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/bt/libbt.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/console/libconsole.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_psram/libesp_psram.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/fatfs/libfatfs.a  esp-idf/json/libjson.a  esp-idf/mqtt/libmqtt.a  esp-idf/perfmon/libperfmon.a  esp-idf/spiffs/libspiffs.a  esp-idf/ulp/libulp.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/main/libmain.a  esp-idf/audio_sal/libaudio_sal.a  esp-idf/adf_utils/libadf_utils.a  esp-idf/display_service/libdisplay_service.a  esp-idf/esp_dispatcher/libesp_dispatcher.a  esp-idf/audio_hal/libaudio_hal.a  esp-idf/espressif__nghttp/libespressif__nghttp.a  esp-idf/esp-adf-libs/libesp-adf-libs.a  esp-idf/audio_pipeline/libaudio_pipeline.a  esp-idf/esp_peripherals/libesp_peripherals.a  esp-idf/audio_mixer/libaudio_mixer.a  esp-idf/espressif__esp-dsp/libespressif__esp-dsp.a  esp-idf/esp-sr/libesp-sr.a  esp-idf/audio_recorder/libaudio_recorder.a  esp-idf/wifi_service/libwifi_service.a  esp-idf/dueros_service/libdueros_service.a  esp-idf/esp_actions/libesp_actions.a  esp-idf/tone_partition/libtone_partition.a  esp-idf/audio_stream/libaudio_stream.a  esp-idf/battery_service/libbattery_service.a  esp-idf/bluetooth_service/libbluetooth_service.a  esp-idf/coredump_upload_service/libcoredump_upload_service.a  esp-idf/esp_codec_dev/libesp_codec_dev.a  esp-idf/esp_event_cast/libesp_event_cast.a  esp-idf/input_key_service/libinput_key_service.a  esp-idf/ota_service/libota_service.a  esp-idf/playlist/libplaylist.a  esp-idf/esp-opus/libesp-opus.a  esp-idf/esp32_deflate/libesp32_deflate.a  esp-idf/hulp/libhulp.a  esp-idf/icm426xx/libicm426xx.a  esp-idf/my_board/libmy_board.a  esp-idf/rtp/librtp.a  esp-idf/stwlc38/libstwlc38.a  esp-idf/tli493d/libtli493d.a  esp-idf/app_trace/libapp_trace.a  esp-idf/app_trace/libapp_trace.a  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/mqtt/libmqtt.a  esp-idf/perfmon/libperfmon.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/protocomm/libprotocomm.a  esp-idf/protob [... cut ...] tc  esp-idf/esp_phy/libesp_phy.a  -lphy  -lrtc  -u vfs_include_syscalls_impl  -lbtdm_app && :
/Users/kevin/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/kevin/esp/esp-idf-v5.1.5/components/bt/controller/lib_esp32/esp32/libbtdm_app.a(ea.o):(.iram1.4+0xc): undefined reference to `ld_acl_rsw_evt_start_cbk'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

BetterJincheng, 2025-02-12:
@expresspotato Sorry for the mistake. `PROVIDE ( ld_acl_rsw_evt_start_cbk = 0x40031154 );` is correct.

expresspotato, 2025-02-26:
Hey @BetterJincheng , this appears to be fixed 👍

closed 2025-02-26 by expresspotato