espressif/esp-idf #13606: A2DP_sink example randomly changes device(smartphone) volume

marosd, 2024-04-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.1.2

### Espressif SoC revision.

ESP32-WROOM-32

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

My own board

### Power Supply used.

External 3.3V

### What is the expected behavior?

I built example from the template. I just only change BT name and pins for I2S interface. Expected behavior is to connect from phone to A2DP_sink and listen music. Just simple transfer from BT to I2S.

### What is the actual behavior?

Connect works, playback works, but every 2 or 3 seconds is randomly changed volume on the device (phone). It could be from 0 to 100%. Phone everytime shows volume changed bar on the screen. It was tested on iPhon and Android too. Same behavior.

### Steps to reproduce.

No triggers. It happen just after playback.

### More Information.

Thank you so much for the help. Maros

shangke1112, 2024-04-15:
This is the behavior of example ,please refer to the function :
**volume_change_simulation(void *arg)**

marosd, 2024-04-15:
> This is the behavior of example ,please refer to the function : **volume_change_simulation()**

Oh. This is shame for me. In the past, I found in the code "volume simulation" function, but I saw inside the function, that volume goes only up. (Mine was varying also down... or maybe it was during that overflow. When I manually added volume to normal, then again -> it set remembered value / or overflow).

Also, I thought its not called. Now, I went deeper in the code and I found it, that its related to dispached function in the main.

Feel so sorry, that I created alarm for no reason. I just new on this platform and when I got a big troubles to set everything to complile, I thought that something is simply wrong.

Thank you so much.

All the best,
Maros

closed 2024-04-15 by marosd