espressif/esp-idf #13607: Bluetooth HFP Audio Gate PCM waveform mode I2S (esp-idf-v5.2)

Ageofenigma, 2024-04-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I want to use PCM mode in the I2S interface for Bluetooth HFP Audio Gate Example to be a slave for SIMCom modem.  Simcom supports only PCM waveform mode ( 256 kHz clock, mono, short 16 bit ). Is it possible to rebuild lib to be able to use this mode? (esp-idf-v5.2)

esp-qing, 2024-04-15:
OK, I will send you the lib file later.

esp-qing, 2024-04-15:
Hi, according to your requirements _[256 kHz clock, mono, short 16 bit],_ this is the corresponding lib file based on idf version v5.2.
[libbtdm_app.zip](https://github.com/espressif/esp-idf/files/14978118/libbtdm_app.zip)
The requirement about slave role can be configured in esp-idf.
If you have any questions, feel free to ask.

esp-qing, 2024-04-16:
This lib file corresponds to esp-idf version 5.2.1. Could you please confirm which version of esp-idf you are using? You can view the log output from the compilation of the previous lib file.

Ageofenigma, 2024-04-16:
**idf.py --version** shows ESP-IDF v5.2.1-dirty

esp-qing, 2024-04-16:
Hi, this is the new lib file corresponding to idf version 5.2.1(commit a322e6bd), please try again. 
[libbtdm_app.zip](https://github.com/espressif/esp-idf/files/14994843/libbtdm_app.zip)

Ageofenigma, 2024-04-16:
It works thanks, but I figured out that the SimCom clock is 2048 KHz and looks like this:
[image]
Could you change it to 2048 KHz clock? Will it work for such a waveform?

Ageofenigma, 2024-04-18:
Is it possible to rebuild it to 2048 KHz clock?

esp-qing, 2024-04-18:
Hi, the default synchronization clock of esp-idf is 8kHZ, and the default data clock is 256kHZ. I guess the clock you want to modify is the data clock.
There has never been a requirement to modify the clock before.  Sorry, after trying, It cannot be directly modified to 2048kHZ at the moment and It will take some time to improve this feature. Maybe you can try modifying the SimCom clock?
If you have any further questions please feel free to ask.

Ageofenigma, 2024-05-06:
Hi, with new firmware switched simcom to a standart I2S mode, but it generates data only for one channel, ESP32 doesn't accept it, what can you advise? [image]

Ageofenigma, 2024-05-06:
Could you enable mono mode so data comes when the SYNC signal is low?

closed 2024-05-07 by Ageofenigma