espressif/esp-idf #13609: MCPWM CAPTURE回调函数没有被触发问题

murarduino, 2024-04-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

测试环境：IDF 5.2.1，模拟环境：Wokwi for VS Code
我想做一个实验：利用MCPWM捕获器捕捉一个PWM的脉冲信号，以测试该信号的幅值。由于担心Wowki里产生的PWM信号可能会存在问题，我模拟使用gpio_set_level方式生成一个类拟的PWM信号。具体代码如下：
`#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_adc/adc_oneshot.h"
#include "driver/mcpwm_cap.h"
#include "esp_log.h"
#include "driver/gpio.h"

#define MCPWM_CAP_PIN  GPIO_NUM_4
#define BLINK_GPIO GPIO_NUM_5

const static char *TAG = "abc_read";

SemaphoreHandle_t pot_semaphore = NULL;

uint16_t cap_analog_value = 0;

// MCPWM 回调函数
static bool abc_reading_callback(mcpwm_cap_channel_handle_t cap_chan, const mcpwm_capture_event_data_t *edata, void *user_data)
{
    ESP_LOGI(TAG, "into abc_callback");
    static int read_analog = 0;
    // TaskHandle_t task_to_notify = (TaskHandle_t)user_data;
    BaseType_t high_task_wakeup = pdFALSE;
    adc_oneshot_unit_handle_t readanalog_handle = NULL;

    // calculate the interval in the ISR,
    // so that the interval will be always correct even when capture_queue is not handled in time and overflow.
    if (edata->cap_edge == MCPWM_CAP_EDGE_POS)
    {
        // store the timestamp when pos edge is detected
        cap_analog_value = adc_oneshot_read(readanalog_handle, ADC_CHANNEL_4, &read_analog);
        xSemaphoreGiveFromISR(pot_semaphore, NULL);
    }

    return high_task_wakeup == pdTRUE;
}
void pulse_make_task(void *Param)
{
    gpio_reset_pin(BLINK_GPIO);
    gpio_set_direction(BLINK_GPIO, GPIO_MODE_OUTPUT);
    static uint32_t ledstatus = 0;

    while (1)
    {
       ledstatus = !ledstatus;
        gpio_set_level(BLINK_GPIO,ledstatus);
        vTaskDelay(500);   
    }
}

void mcpwm_capture_init()
{
    //---mcpwm_capture_timer_init
    ESP_LOGI(TAG, "Install capture timer");
    mcpwm_cap_timer_handle_t cap_timer = NULL;
    mcpwm_capture_timer_config_t cap_conf{
        .group_id = 0,
        .clk_src = MCPWM_CAPTURE_CLK_SRC_DEFAULT,
        .resolution_hz = 0,
    };

    ESP_ERROR_CHECK(mcpwm_new_capture_timer(&cap_conf, &cap_timer));
    // ESP_LOGI(TAG,"mcpwm time resolution %d",mcpwm_capture_timer_get_resolution(cap_timer,&out_resl));

    //---mcpwm_capture_channel_init
    ESP_LOGI(TAG, "Install capture channel");
    mcpwm_cap_channel_handle_t cap_chan = NULL;
    mcpwm_capture_channel_config_t cap_ch_conf{
        .gpio_num = MCPWM_CAP_PIN,
        .intr_priority = 0,
        .prescale = 1,
        .flags = {
            .pos_edge = 1,            // Whether to capture on positive edge
            .neg_edge = 0,            // Whether to capture on negative edge
            .pull_up = 1,             // Whether to pull up internally
            .pull_down = 0,           // Whether to pull down internally
            .invert_cap_signal = 0,   // Invert the input capture signal
            .io_loop_back = 0,        // For debug/test
            .keep_io_conf_at_exit = 1 // For debug/test
        }

    };

    cap_ch_conf.gpio_num = MCPWM_CAP_PIN;
    ESP_ERROR_CHECK(mcpwm_new_capture_channel(cap_timer, &cap_ch_conf, &cap_chan));

    //---mcpwm_capture-callback functions register
    ESP_LOGI(TAG, "Register capture callback");
    // TaskHandle_t cur_task = xTaskGetCurrentTaskHandle();
    mcpwm_capture_event_callbacks_t abc_cbs;
    abc_cbs.on_cap = abc_reading_callback;

    // ESP_ERROR_CHECK(mcpwm_capture_channel_register_event_callbacks(cap_chan, &abc_cbs, cur_task));
    ESP_ERROR_CHECK(mcpwm_capture_channel_register_event_callbacks(cap_chan, &abc_cbs, NULL));

    //---mcpwm_capture Enable
    ESP_LOGI(TAG, "Enable capture channel");
    ESP_ERROR_CHECK(mcpwm_capture_channel_enable(cap_chan));

    //--MCPWM-Capture Timer start
    ESP_LOGI(TAG, "Enable and start capture timer");
    ESP_ERROR_CHECK(mcpwm_capture_timer_enable(cap_timer));
    ESP_ERROR_CHECK(mcpwm_capture_timer_start(cap_timer));
}

extern "C" void app_main(void)
{
    // 创建二进制信号量
    pot_semaphore = xSemaphoreCreateBinary();
    // 初始化 MCPWM 模块
    mcpwm_capture_init();
    xTaskCreate(pulse_make_task, "Make_PULSE", 1024 * 4, NULL, 3, NULL);
    while (1)
    {
        if (xSemaphoreTake(pot_semaphore, portMAX_DELAY) == pdTRUE)
        {
            printf("value is : %u\n", cap_analog_value);
        }
    }
}`
Wokwi的diagram.json文件如下：
`{
  "version": 1,
  "author": "xxxx",
  "editor": "wokwi",
  "parts": [
    {
      "type": "board-esp32-devkit-c-v4",
      "id": "esp",
      "top": -20.86,
      "left": 99.7,
      "rotate": 90,
      "attrs": {}
    },
    { "type": "wokwi-led", "id": "led1", "top": -61.2, "left": -25, "attrs": { "color": "red" } },
    {
      "type": "wokwi-resistor",
      "id": "r1",
      "top": 32.2,
      "left": -48.85,
      "rotate": 270,
      "attrs": { "value": "1000" }
    }
  ],
  "connections": [
    [ "esp:TX", "$serialMonitor:RX", "", [] ],
    [ "esp:RX", "$serialMonitor:TX", "", [] ],
    [ "led1:A", "esp:5", "red", [ "v172.8", "h144" ] ],
    [ "r1:2", "led1:C", "green", [ "h0" ] ],
    [ "esp:GND.2", "r1:1", "black", [ "v38.4", "h-249.6" ] ],
    [ "esp:25", "r1:2", "green", [ "v0" ] ],
    [ "esp:4", "led1:A", "blue", [ "v0" ] ]
  ],
  "dependencies": {}
}`

在Wokwi模拟测试中，日志如下：
`ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7172
load:0x40078000,len:15540
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3904
entry 0x40080640
I (54) boot: ESP-IDF v5.2.1-dirty 2nd stage bootloader
I (55) boot: compile time Apr 12 2024 10:28:48
I (55) boot: Multicore bootloader
I (59) boot: chip revision: v3.0
I (62) boot.esp32: SPI Speed      : 40MHz
I (67) boot.esp32: SPI Mode       : DIO
I (71) boot.esp32: SPI Flash Size : 4MB
I (76) boot: Enabling RNG early entropy source...
I (81) boot: Partition Table:
I (85) boot: ## Label            Usage          Type ST Offset   Length
I (92) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (99) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (107) boot:  2 factory          factory app      00 00 00010000 00100000
I (115) boot: End of partition table
I (119) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0adf4h ( 44532) map
I (183) esp_image: segment 1: paddr=0001ae1c vaddr=3ffb0000 size=02254h (  8788) load
I (197) esp_image: segment 2: paddr=0001d078 vaddr=40080000 size=02fa0h ( 12192) load
I (216) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=15f74h ( 89972) map
I (330) esp_image: segment 4: paddr=00035f9c vaddr=40082fa0 size=097e8h ( 38888) load
I (407) boot: Loaded app from partition at offset 0x10000
I (408) boot: Disabling RNG early entropy source...
I (419) cpu_start: Multicore app
I (427) cpu_start: Pro cpu start user code
I (428) cpu_start: cpu freq: 160000000 Hz
I (428) cpu_start: Application information:
I (431) cpu_start: Project name:     abc_read
I (436) cpu_start: App version:      1
I (440) cpu_start: Compile time:     Apr 12 2024 10:28:22
I (446) cpu_start: ELF file SHA256:  4d0730446...
I (451) cpu_start: ESP-IDF:          v5.2.1-dirty
I (457) cpu_start: Min chip rev:     v0.0
I (462) cpu_start: Max chip rev:     v3.99 
I (466) cpu_start: Chip rev:         v3.0
I (472) heap_init: Initializing. RAM available for dynamic allocation:
I (478) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (485) heap_init: At 3FFB2B40 len 0002D4C0 (181 KiB): DRAM
I (491) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (497) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (504) heap_init: At 4008C788 len 00013878 (78 KiB): IRAM
I (511) spi_flash: detected chip: gd
I (514) spi_flash: flash io: dio
I (520) main_task: Started on CPU0
I (530) main_task: Calling app_main()
I (530) abc_read: Install capture timer
I (530) abc_read: Install capture channel
I (530) gpio: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (540) abc_read: Register capture callback
I (550) abc_read: Enable capture channel
I (550) abc_read: Enable and start capture timer
I (560) gpio: GPIO[5]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 `

我在回调函数的第一行设置打印进入回调函数的日志，
`   ESP_LOGI(TAG, "into abc_callback");`
但在日志中没有显示。请问，这是哪里的问题？

suda-morris, 2024-04-15:
IDF also have a test case for MCPWM capture, where we also use the GPIO as a signal simulator: https://github.com/espressif/esp-idf/blob/master/components/esp_driver_mcpwm/test_apps/mcpwm/main/test_mcpwm_cap.c#L72

This test case works very well, can you pick that case to your wokwi environment? If that still can't work, can you run it in a real hardware? I'm not sure if the wokwi has any issue in mocking the MCPWM.

murarduino, 2024-04-15:
> IDF also have a test case for MCPWM capture, where we also use the GPIO as a signal simulator: [...]

请问，如何使用TEST_CASE。我新建了一个项目用于你的代码测试，将test_mcpwm_cap.c和test_mcpwm_utils.h文件放到main目录中。但VSCode提示如下图：
[image]

suda-morris, 2024-04-15:
you can run the test_app locally: https://github.com/espressif/esp-idf/tree/master/components/esp_driver_mcpwm/test_apps/mcpwm

It's nothing special but an idf application project.

murarduino, 2024-04-15:
> you can run the test_app locally: https://github.com/espressif/esp-idf/tree/master/components/esp_driver_mcpwm/test_apps/mcpwm
> 
> It's nothing special but an idf application project.

[image]

murarduino, 2024-04-15:
> you can run the test_app locally: https://github.com/espressif/esp-idf/tree/master/components/esp_driver_mcpwm/test_apps/mcpwm
> 
> It's nothing special but an idf application project.

`D:\Esp-Doc\5>cd esp_driver_mcpwm

D:\Esp-Doc\5\esp_driver_mcpwm>dir
 驱动器 D 中的卷是 software
 卷的序列号是 4E30-2DA5

 D:\Esp-Doc\5\esp_driver_mcpwm 的目录

2024/04/15  17:12    <DIR>          .
2024/04/15  17:13    <DIR>          ..
2024/04/15  17:12               718 CMakeLists.txt
2024/04/15  17:12    <DIR>          include
2024/04/15  17:12             1,052 Kconfig
2024/04/15  17:12               218 linker.lf
2024/04/15  17:12    <DIR>          src
2024/04/15  17:12    <DIR>          test_apps
               3 个文件          1,988 字节
               5 个目录 88,369,979,392 可用字节

D:\Esp-Doc\5\esp_driver_mcpwm>idf.py build
Executing action: all (aliases: build)
Running cmake in directory D:\Esp-Doc\5\esp_driver_mcpwm\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=D:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 D:\Esp-Doc\5\esp_driver_mcpwm"...
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error in CMakeLists.txt:
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error in CMakeLists.txt:
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.24)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
See also "D:/Esp-Doc/5/esp_driver_mcpwm/build/CMakeFiles/CMakeOutput.log".
See also "D:/Esp-Doc/5/esp_driver_mcpwm/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1, output of the command is in the D:\Esp-Doc\5\esp_driver_mcpwm\build\log\idf_py_stderr_output_25380 and D:\Esp-Doc\5\esp_driver_mcpwm\build\log\idf_py_stdout_output_25380

D:\Esp-Doc\5\esp_driver_mcpwm>`

closed 2025-04-03 by Alvin1Zhang