espressif/esp-idf #13610: Failed to allocate RSA interrupt 261

lekshith, 2024-04-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm trying to establish MQTT broker connection from my ESP32 to publish some data. The client initialization was working perfectly fine with the esp_crt_bundle certificates until I included the data packets to be sent over the connection. It is now failing with the following error "
[image]
" which looks like a certificate validation error. There is enough memory available, I've made use external SPIRAM as well. Nothing else was changed in the code expect for adding some json data packets to be sent over. Please suggest any workaround for this issue.

lekshith, 2024-04-15:
[image]

Harshal5, 2024-04-17:
Hi @lekshith,

Could you share the ESP-IDF version that you are using for your project?
Just by guessing the issue using the log screenshot that you have provided, I think this might be because no available free interrupts are remaining. We have fixed this issue in the commit https://github.com/espressif/esp-idf/commit/d3be7bda056dfcb7edbac4cf8757ec1ac7d0a1ea, and it has also been backported until ESP-IDF v4.4.

Thanks!

Harshal5, 2024-04-17:
Oh okay, in the log screenshot, I can see `v5.1` in the PATH of the file. The above commit has been backported to the `release/v5.1` branch and is present in the tag `v5.1.3`. Could you confirm if the issue persists even when using `v5.1.3`? Alternatively, you could also cherry the commit https://github.com/espressif/esp-idf/commit/4c30f2a4a0229979b4700c6acb34d49705fb6670 for your `ESP-IDF` version to check the same.

Harshal5, 2024-04-22:
This issue looks similar to https://github.com/espressif/esp-idf/issues/12289, the above fix would fix RSA-related issues but you might come across interrupt registration failure for some other peripherals as there are a limited amount of interrupts per core on the device. As mentioned in https://github.com/espressif/esp-idf/issues/12289#issuecomment-1778956939, you would need to move some peripheral interrupts to the other core.

Closing the issue as of now due to a similar issue. Please feel free to re-open for any follow-up.

lekshith, 2024-04-24:
[image]

Harshal5, 2024-04-25:
Hello @lekshith,

> fatal: bad revision '4c30f2a'
> my git version is up to date and I am in the main branch when I'm trying to commit.

Looks like your git commit-tree does not contain the above commit. You might need to fetch the commit.
Could you share with me the output of the command `idf.py --version`? This would be helpful for me to know the exact ESP-IDF version you are using so that I would be able to give you exact instructions to proceed.

Thanks!

Harshal5, 2024-04-25:
In case you are using v5.1.3, the commit https://github.com/espressif/esp-idf/commit/4c30f2a4a0229979b4700c6acb34d49705fb6670 should be already present in your ESP-IDF version.
The basic issue behind this error is that the core on which you are allocating the RSA interrupt has exhausted all the free interrupts and thus is not able to allocate a new one. You might either need to reduce some interrupts or move some interrupts to the other core. A quick fix could be disabling the RSA interrupt (`CONFIG_MBEDTLS_MPI_USE_INTERRUPT`) ([reference](https://github.com/espressif/esp-idf/issues/12289#issuecomment-1829377564)).

lekshith, 2024-04-25:
I did try disabling the RSA interrupt
# CONFIG_MBEDTLS_HARDWARE_MPI is not set.
This does stop the RSA interrupt while initializing the client, but now I'm facing the following error,
[image]
As you can see, the error suggests "SSL - A fatal alert message was received from our peer", where the connection is refused from the server after verifying the certificate.

lekshith, 2024-04-25:
Please refer to the more detailed logs below:
I (733475) MQTT: ########## ########### ###########
I (733515) mbedtls: ssl_tls.c:3919 => handshake

I (733515) mbedtls: ssl_msg.c:2358 => flush output

I (733515) mbedtls: ssl_msg.c:2367 <= flush output

I (733525) mbedtls: ssl_tls.c:3838 client state: MBEDTLS_SSL_HELLO_REQUEST

I (733525) mbedtls: ssl_msg.c:2358 => flush output

I (733535) mbedtls: ssl_msg.c:2367 <= flush output

I (733535) mbedtls: ssl_tls.c:3838 client state: MBEDTLS_SSL_CLIENT_HELLO

I (733545) mbedtls: ssl_client.c:924 => write client hello

I (733555) mbedtls: ssl_msg.c:2788 => write handshake message

I (733565) mbedtls: ssl_msg.c:2948 => write record

I (733565) mbedtls: ssl_msg.c:3085 <= write record

I (733575) mbedtls: ssl_msg.c:2909 <= write handshake message

I (733575) mbedtls: ssl_client.c:1012 <= write client hello

I (733585) mbedtls: ssl_msg.c:2358 => flush output

I (733585) mbedtls: ssl_msg.c:2372 message length: 141, out_left: 141

I (733595) mbedtls: ssl_msg.c:2379 ssl->f_send() returned 141 (-0xffffff73)

I (733605) mbedtls: ssl_msg.c:2406 <= flush output

I (733615) mbedtls: ssl_tls.c:3838 client state: MBEDTLS_SSL_SERVER_HELLO

I (733615) mbedtls: ssl_tls12_client.c:1193 => parse server hello

I (733625) mbedtls: ssl_msg.c:4122 => read record

I (733635) mbedtls: ssl_msg.c:2160 => fetch input

I (733635) mbedtls: ssl_msg.c:2300 in_left: 0, nb_want: 5

I (733645) mbedtls: ssl_msg.c:2320 in_left: 0, nb_want: 5

I (733645) mbedtls: ssl_msg.c:2323 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)

I (733655) mbedtls: ssl_msg.c:2345 <= fetch input

I (733665) mbedtls: ssl_msg.c:2160 => fetch input

I (733665) mbedtls: ssl_msg.c:2300 in_left: 5, nb_want: 7

I (733675) mbedtls: ssl_msg.c:2320 in_left: 5, nb_want: 7

I (733685) mbedtls: ssl_msg.c:2323 ssl->f_recv(_timeout)() returned 2 (-0xfffffffe)

I (733685) mbedtls: ssl_msg.c:2345 <= fetch input

I (733695) mbedtls: ssl_msg.c:5027 got an alert message, type: [2:40]

W (733705) mbedtls: ssl_msg.c:5034 is a fatal alert message (msg 40)

W (733705) mbedtls: ssl_msg.c:4177 mbedtls_ssl_handle_message_type() returned -30592 (-0x7780)

W (733715) mbedtls: ssl_tls12_client.c:1197 mbedtls_ssl_read_record() returned -30592 (-0x7780)

I (733725) mbedtls: ssl_tls.c:3930 <= handshake

E (733735) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780
I (733745) esp-tls-mbedtls: Certificate verified.
E (733745) esp-tls: Failed to open new connection
E (733755) transport_base: Failed to open a new connection
E (733765) HTTP_CLIENT: Connection failed, sock < 0
I (733765) MQTT: HTTP_EVENT_ERROR
E (733765) MQTT: HTTP POST request failed: ESP_ERR_HTTP_CONNECT
I (733775) MQTT: Last esp error code: 0x801a
I (733775) MQTT: Last mbedtls failure: 0x7780
No response from server after POST request!

lekshith, 2024-04-25:
Here are the supported cipher suites from the server end for reference
# TLS 1.3 (suites in server-preferred order)
TLS_AES_256_GCM_SHA384 (0x1302)   ECDH x25519 (eq. 3072 bits RSA)   FS	256
TLS_CHACHA20_POLY1305_SHA256 (0x1303)   ECDH x25519 (eq. 3072 bits RSA)   FS	256
TLS_AES_128_GCM_SHA256 (0x1301)   ECDH x25519 (eq. 3072 bits RSA)   FS	128
# TLS 1.2 (suites in server-preferred order)
TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256 (0xc02f)   ECDH x25519 (eq. 3072 bits RSA)   FS	128
TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384 (0xc030)   ECDH x25519 (eq. 3072 bits RSA)   FS	256
TLS_ECDHE_RSA_WITH_CHACHA20_POLY1305_SHA256 (0xcca8)   ECDH x25519 (eq. 3072 bits RSA)   FS	256

lekshith, 2024-04-26:
If anyone is facing the same issue, here are some insights to overcome this.
1. Go to SSL Labs Server Test online and check all the cipher-suites that is supported by the server.
2. you can either disable the hardware MPI (bignum) acceleration in the menuconfig or move some ISR to a different core, either option works as expected.
3. depending on the cipher-suites of the server, enable the same in the menuconfig under TLS key exchange methods.
Hope this helps.

closed 2024-04-26 by lekshith

cmorganBE, 2024-06-11:
For anyone else with this issue I debugged it by calling:

```
esp_intr_dump();
```

both at startup and before the tls connection was started (which I presume is why the RSA interrupt was being registered, to perform the crypto operations).

From there it was clear CPU0 in my case had no free Level 1 interrupts but CPU1 had three or four.

Switched from xTaskCreateStatic() to xTaskCreateStaticPinnedToCore() for a task that was registering some gpio interrupts and also making use of the mcpwm module and its interrupt, and pinned to core '1'.

esp_intr_dump() showed one fewer Level 1 interrupt on CPU1 and the RSA registration issues went away (as I think that task was running/registering from CPU0).