espressif/esp-idf #13611: Decode coredump problem

hitecSmartHome, 2024-04-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have a function which saves the `coredump` partition data to a file so i can download it.

```cpp
void Sys::saveCoreDump(){
    size_t size = 0;
    size_t address = 0;
    char* coreDump;
    if( !alloc(coreDump, 640) ){ return; } // Check if allocation was successfull
    if (esp_core_dump_image_get(&address, &size) != ESP_OK) { free(coreDump); return; } // Try to get the bin image pointers
    const esp_partition_t* pt = NULL;
    pt = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_COREDUMP, "coredump");
    if (pt == NULL) { free(coreDump); return; } // Check if we found the partition or not.
    uint8_t bf[256];
    int16_t toRead;
    db.remove(CORE_DUMP_PATH); // Remove the previous file if any
    for (int16_t i = 0; i < (size / 256) + 1; i++) {
        strcpy(coreDump, "");
        toRead = (size - i * 256) > 256 ? 256 : (size - i * 256);
        esp_err_t er = esp_partition_read(pt, i * 256, bf, toRead);
        if (er != ESP_OK) {
            break;
        }
        for (int16_t j = 0; j < 256; j++) {
            char str_tmp[3];
            sprintf(str_tmp, "%02x", bf[j]);
            strcat(coreDump, str_tmp);
        }
        db.append(CORE_DUMP_PATH, coreDump); // Append the bin content
    }
    esp_core_dump_image_erase(); // Erase partition
    free(coreDump);
}
```
```cpp
esp_err_t Sys::esp_core_dump_image_erase() {
    /* Find the partition that could potentially contain a (previous) core dump. */
    const esp_partition_t* core_part = esp_partition_find_first(ESP_PARTITION_TYPE_DATA,
        ESP_PARTITION_SUBTYPE_DATA_COREDUMP,
    "coredump");
    if (!core_part) {
        logger.write("No core dump partition found!|n");
        return ESP_ERR_NOT_FOUND;
    }
    if (core_part->size < sizeof(uint32_t)) {
        logger.write("Too small core dump partition!\n");
        return ESP_ERR_INVALID_SIZE;
    }

    esp_err_t err = ESP_OK;
    err = esp_partition_erase_range(core_part, 0, core_part->size);
    if (err != ESP_OK) {
        Serial.printf("Failed to erase core dump partition (%d)!\n", err);
        return err;
    }

    // on encrypted flash esp_partition_erase_range will leave encrypted
    // garbage instead of 0xFFFFFFFF so overwriting again to safely signalize
    // deleted coredumps
    const uint32_t invalid_size = 0xFFFFFFFF;
    err = esp_partition_write(core_part, 0, &invalid_size, sizeof(invalid_size));
    if (err != ESP_OK) {
        logger.write("Failed to write core dump partition size (%d)!\n", err);
    }

    return err;
}
```

Where db.append is just a wrapper for LittleFs append.
I got the folllowing downloadable file:

```bin
646c0000000100000000000000000000000000007f454c4601010100000000000000000004005e00010000000000000034000000000000000000000034002000260028000000000004000000f40400000000000000000000c02a0000c02a0000060000000000000001000000b42f000058fbfd3f58fbfd3f640100006401000006000000000000000100000018310000a0f6fd3fa0f6fd3fb0040000b0040000060000000000000001000000c8350000d4ecfb3fd4ecfb3f64010000640100000600000000000000010000002c37000040ebfb3f40ebfb3f8001000080010000060000000000000001000000ac38000040f4fb3f40f4fb3f6401000064010000060000000000000001000000103a0000b0f2fb3fb0f2fb3f8001000080010000060000000000000001000000903b000018dffb3f18dffb3f6401000064010000060000000000000001000000f43c000070dbfc3f70dbfc3f8001000080010000060000000000000001000000743e000098a6fd3f98a6fd3f6401000064010000060000000000000001000000d83f000090a4fd3f90a4fd3f0002000000020000060000000000000001000000d84100006cb1fd3f6cb1fd3f64010000640100000600000000000000010000003c430000a0d4fe3fa0d4fe3f8001000080010000060000000000000001000000bc44000058acfc3f58acfc3f640100006401000006000000000000000100000020460000b0aafc3fb0aafc3fa0010000a0010000060000000000000001000000c0470000fcdcfc3ffcdcfc3f64010000640100000600000000000000010000002449000030effc3f30effc3fe0010000e0010000060000000000000001000000044b0000acfdfb3facfdfb3f6401000064010000060000000000000001000000684c000020fcfb3f20fcfb3f8001000080010000060000000000000001000000e84d0000b464fd3fb464fd3f64010000640100000600000000000000010000004c4f0000a061fd3fa061fd3f00030000000300000600000000000000010000004c52000064defc3f64defc3f6401000064010000060000000000000001000000b053000020f8fc3f20f8fc3ff0010000f0010000060000000000000001000000a0550000ccdffc3fccdffc3f6401000064010000060000000000000001000000045700007008fd3f7008fd3fb0010000b0010000060000000000000001000000b4580000f4fdfd3ff4fdfd3f6401000064010000060000000000000001000000185a0000003bfc3f003bfc3fd0010000d0010000060000000000000001000000e85b000054fafa3f54fafa3f64010000640100000600000000000000010000004c5d0000c0a5fb3fc0a5fb3fa0010000a0010000060000000000000001000000ec5e0000ccf7fa3fccf7fa3f64010000640100000600000000000000010000005060000020f6fa3f20f6fa3fa0010000a0010000060000000000000001000000f061000048d4fb3f48d4fb3f640100006401000006000000000000000100000054630000a0d2fb3fa0d2fb3fa0010000a0010000060000000000000001000000f4640000c012fd3fc012fd3f6401000064010000060000000000000001000000586600001011fd3f1011fd3fa0010000a0010000060000000000000001000000f867000004b0fd3f04b0fd3f64010000640100000600000000000000010000005c69000030aefd3f30aefd3fc0010000c00100000600000000000000040000001c6b0000000000000000000014010000140100000600000000000000080000004c02000001000000434f52450045535000000000000000000000000000000000000000000000000058fbfd3f00000000000000000000000000000000000000000000000000000000000000000000000000000000000000007c3108402303060049c300406bc30040000000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000c09a088060f7fd3fa0f7fd3fb1f7fd3f1000000090dffd3f09000000070000000000000001000000b1f7fd3fb1f7fd3f03000000f857fc3f98e87a00ffff3f0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000080000004c02000001000000434f524500455350000000000000000000000000000000000000000000000000d4ecfb3f0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000fe742640200d0600000000000000000000000000160000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000064b11c8000ecfb3f000000000100008025a2088040d2fb3f0300000023000600deac1c80d0ebfb3f00000000ffff3f007e1b3fe878f0f53f7cf0f53fffff3f0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000080000004c02000001000000434f52450045535000000000000000000000000000000000000000000000000040f4fb3f0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000fe74264020050600000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 [... cut ...] e464f002a58fbfd3fe80000001d000000ee00000000000000c200000000000000c300000000000000c400000000000000c500000000000000c600000000000000b100000058d31c40b200000000000000b3000000
```

I have tried to decode it with the following command:

`idf.py coredump-info -c C:\Users\Pc\Desktop\coreDump\coreDump.bin`

But i got the following error:

```cmd
Executing action: coredump-info
Failed to load core dump: Core dump version "0x344d" is not supported!
```

Is there a way to decode this to a meaningfull output?

igrr, 2024-04-15:
Probably you need to convert the file from hexadecimal representation back to binary first, before passing it into the coredump-info command. Or looking at it the other way around, is there any specific reason you are encoding the binary data to hexadecimal before writing it into the file?

hitecSmartHome, 2024-04-15:
Oh, I tought the decoder will eat it anyway.
So you say I should do something like this instead?

```cpp
void Sys::saveCoreDumpBin() {
    size_t size = 0;
    size_t address = 0;
    uint8_t* coreDump; // Change char* to uint8_t*
    if (!alloc(coreDump, 640)) { return; }
    if (esp_core_dump_image_get(&address, &size) != ESP_OK) { free(coreDump); return; }
    const esp_partition_t* pt = NULL;
    pt = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_COREDUMP, "coredump");
    if (pt == NULL) { free(coreDump); return; }
    uint8_t bf[256];
    int16_t toRead;
    db.remove(CORE_DUMP_PATH);
    for (int16_t i = 0; i < (size / 256) + 1; i++) {
        toRead = (size - i * 256) > 256 ? 256 : (size - i * 256);
        esp_err_t er = esp_partition_read(pt, i * 256, bf, toRead);
        if (er != ESP_OK) {
            break;
        }
        memcpy(coreDump + i * 256, bf, toRead); // Copy directly into coreDump
    }
    db.appendBinary(CORE_DUMP_PATH, coreDump, size); // Append binary data
    esp_core_dump_image_erase();
    free(coreDump);
}
```

hitecSmartHome, 2024-04-15:
I have converted it back to binary with this tool: https://www.rapidtables.com/convert/number/hex-to-binary.html

But I still get the following error: 

```cmd
Executing action: coredump-info
Failed to load core dump: Core dump version "0x345d" is not supported!
```

hitecSmartHome, 2024-04-15:
[coreDump.txt](https://github.com/espressif/esp-idf/files/14980214/coreDump.txt)
[coreDumpBin.txt](https://github.com/espressif/esp-idf/files/14980219/coreDumpBin.txt)

hitecSmartHome, 2024-04-15:
> is there any specific reason you are encoding the binary data to hexadecimal before writing it into the file?

Actually no.

igrr, 2024-04-15:
Unfortunately I don't know what the functions `alloc(coreDump, 640)` and `db.appendBinary(CORE_DUMP_PATH, coreDump, size)` do, so I can't say if the code is correct or not.

If `coreDump` array is so large that it can contain all the coredump binary, then I think you don't need the `for` loop, you can simply call `esp_partition_read` for the entire `size`.

> Failed to load core dump: Core dump version "0x345d" is not supported!

Your coreDumpBin.txt contains literal "0" and "1" characters instead of data. I think the tool means "binary" in the mathematical sense ("0" and "1" symbols) while coredump expects "binary" in the sense of "raw data". So the first byte of your binary file has to have value 0x34. Right now this byte instead gets encoded as 32 bytes, each a "0" or "1" character...

igrr, 2024-04-15:
By the way, I you have difficulty implementing coredump analysis for your application, you might consider using our managed service, ESP Insights: https://insights.espressif.com/.

hitecSmartHome, 2024-04-16:
I see. So it should be that easy?

```cpp
void Sys::readCoreDumpAndSave(){
    size_t size = 0;
    size_t address = 0;
    // Get the pointer to the image
    if (esp_core_dump_image_get(&address, &size) != ESP_OK) {
        printf("[System] - Failed to get core dump info\n");
        return;
    }
    // Allocate an array for the image buffer with ps_malloc()
    uint8_t* coreDumpBuffer;
    if( !alloc(coreDumpBuffer, size) ) {
        printf("[System] - Failed to allocate core dump buffer\n");
        return;
    }
    const esp_partition_t* pt = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_COREDUMP, "coredump");
    if(!pt){
        printf("[System] - Failed to find coredump partition\n");
        free(coreDumpBuffer);
        return;
    }
    // Read the whole image at once
    esp_err_t er = esp_partition_read(pt, address, coreDumpBuffer, size);
    if (er != ESP_OK) {
        printf("[System] - Failed to read core dump\n");
        free(coreDumpBuffer);
        return;
    }
}
```

> Unfortunately I don't know what the functions alloc(coreDump, 640) and db.appendBinary(CORE_DUMP_PATH, coreDump, size) do, so I can't say if the code is correct or not.

As i said the alloc is just a wrapper for ps_malloc() and the db method call is just a wrapper for LittleFS write operation.

```cpp
boolean Sys::alloc(uint8_t *&ptr, int size) {
    ptr = (uint8_t *)ps_malloc(size * sizeof(uint8_t));
    if (!ptr || ptr == NULL) {
        return false;
    }
    return true;
}
```

hitecSmartHome, 2024-04-16:
This could be my final function where

#define CORE_DUMP_PATH "/system/coreDump.bin"

```cpp
void Sys::readCoreDumpAndSave(){
    size_t size = 0;
    size_t address = 0;
    // Get the pointer to the image
    if (esp_core_dump_image_get(&address, &size) != ESP_OK) {
        printf("[System] - Failed to get core dump info\n");
        return;
    }
    // Allocate an array for the image buffer with ps_malloc()
    uint8_t* coreDumpBuffer;
    if( !alloc(coreDumpBuffer, size) ) {
        printf("[System] - Failed to allocate core dump buffer\n");
        return;
    }
    const esp_partition_t* pt = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_COREDUMP, "coredump");
    if(!pt){
        printf("[System] - Failed to find coredump partition\n");
        free(coreDumpBuffer);
        return;
    }
    // Read the whole image at once
    esp_err_t er = esp_partition_read(pt, address, coreDumpBuffer, size);
    if (er != ESP_OK) {
        printf("[System] - Failed to read core dump\n");
        free(coreDumpBuffer);
        return;
    }
    db.remove(CORE_DUMP_PATH);
    File coreDumpFile = db.getFileObject(CORE_DUMP_PATH,FILE_WRITE);
    size_t written = coreDumpFile.write(coreDumpBuffer, size);
    free(coreDumpBuffer);
    if (written != size) {
        printf("[System] - Failed to write core dump\n");
        return;
    }
    printf("[System] - Successfully wrote core dump\n");
}
```

hitecSmartHome, 2024-04-16:
With this modification I'm still unable to decode it

`Failed to load core dump: Invalid application image for coredump: coredump SHA256(b7fbaab6626d2226) != app SHA256()`

I have got a 27kb bin file now.

[coreDump.txt](https://github.com/espressif/esp-idf/files/14994097/coreDump.txt)

hitecSmartHome, 2024-04-16:
> By the way, if you have difficulty implementing coredump analysis for your application, you might consider using our managed service, ESP Insights: https://insights.espressif.com/.

Unfortunatelly I'm using Arduino as a component and can't move to more recent IDF releases so the Insight is unavailable for me.

erhankur, 2024-04-16:
@hitecSmartHome Does you application elf file modified after coredump has been saved to flash?

hitecSmartHome, 2024-04-16:
Nop. I'm doing a crash now intentionally and download the results after restart immidiatelly.

```cpp
void crashMCU(){
    printf("Crashing...\n");
    while(true){
        int i = 0;
        int crash = 1/i;
        printf("Crash: %d\n", crash);
    }
}
```

erhankur, 2024-04-16:
I can see from the saved coredump file (after trimming the 20bytes header), your application elf sha is `b7fbaab6626d2226`
 ```
xtensa-esp32-elf-readelf -n trimmed.dmp

Displaying notes found at file offset 0x000067fc with length 0x00000114:
  Owner                Data size 	Description
  ESP_CORE_DUMP_INFO   0x00000048	Unknown note type: (0x0000204a)
   description data: 00 01 00 00 62 37 66 62 61 61 62 36 36 32 36 64 32 32 32 36 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

You can verify it using a tool like `shasum -a 256 build/blink.elf`

Results must be the same. And you can debug `elf_write_core_dump_info` function to see what is going on.

hitecSmartHome, 2024-04-16:
So why does `idf.py coredump-info -t C:\Users\Pc\Desktop\coreDump\coreDump.txt` fails?

hitecSmartHome, 2024-04-16:
Should i trim the 20 bytes header before i save it to file?

erhankur, 2024-04-16:
No, you don't need to trim. I did it to see the note section in the elf file. `idf.py coredump-info` will take care of the headers for you.

Can you calculate and compare the elf sha256 outputs as I described above?  Or you can ensure you save the coredump from scratch in every panic.  If `ESP_COREDUMP_FLASH_NO_OVERWRITE` is selected maybe you have an old coredump file but a new application built. Please send us the console logs until the coredump finishes so that we can have a better understanding.

hitecSmartHome, 2024-04-16:
I see. Will check it. Will crash it again and compare the new log to the file sha.

Here is my core dump sdkconfig

```
#
# Core dump
#
CONFIG_ESP_COREDUMP_ENABLE_TO_FLASH=y
# CONFIG_ESP_COREDUMP_ENABLE_TO_UART is not set
# CONFIG_ESP_COREDUMP_ENABLE_TO_NONE is not set
# CONFIG_ESP_COREDUMP_DATA_FORMAT_BIN is not set
CONFIG_ESP_COREDUMP_DATA_FORMAT_ELF=y
CONFIG_ESP_COREDUMP_CHECKSUM_CRC32=y
# CONFIG_ESP_COREDUMP_CHECKSUM_SHA256 is not set
CONFIG_ESP_COREDUMP_CHECK_BOOT=y
CONFIG_ESP_COREDUMP_ENABLE=y
CONFIG_ESP_COREDUMP_LOGS=y
CONFIG_ESP_COREDUMP_MAX_TASKS_NUM=64
CONFIG_ESP_COREDUMP_STACK_SIZE=0
# end of Core dump
```

I can't find `ESP_COREDUMP_FLASH_NO_OVERWRITE` in the menuconfig

hitecSmartHome, 2024-04-16:
I do save it in every panic. When my esp boots up I have the following logic

```cpp
esp_reset_reason_t resetReason = esp_reset_reason();
if( resetReason != ESP_RST_SW && resetReason != ESP_RST_POWERON ){
    readCoreDumpAndSave();
}
```

```cpp
void Sys::readCoreDumpAndSave(){
    size_t size = 0;
    size_t address = 0;
    // Get the pointer to the image
    if (esp_core_dump_image_get(&address, &size) != ESP_OK) {
        printf("[System] - Failed to get core dump info\n");
        return;
    }
    // Allocate an array for the image buffer with ps_malloc()
    uint8_t* coreDumpBuffer;
    if( !alloc(coreDumpBuffer, size) ) {
        printf("[System] - Failed to allocate core dump buffer\n");
        return;
    }
    const esp_partition_t* pt = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_COREDUMP, "coredump");
    if(!pt){
        printf("[System] - Failed to find coredump partition\n");
        free(coreDumpBuffer);
        return;
    }
    // Read the whole image at once
    esp_err_t er = esp_partition_read(pt, 0, coreDumpBuffer, size);
    if (er != ESP_OK) {
        printf(
            "[System] - Failed to read core dump. Error: %s\n",
            esp_err_to_name(er)
        );
        free(coreDumpBuffer);
        return;
    }
    File coreDumpFile = db.getFileObject(CORE_DUMP_PATH,FILE_WRITE);
    size_t written = coreDumpFile.write(coreDumpBuffer, size);
    free(coreDumpBuffer);
    if (written != size) {
        printf("[System] - Failed to write core dump\n");
        return;
    }
    printf("[System] - Successfully wrote core dump\n");
    esp_core_dump_image_erase();
}

esp_err_t Sys::esp_core_dump_image_erase() {
    /* Find the partition that could potentially contain a (previous) core dump. */
    const esp_partition_t* core_part = esp_partition_find_first(ESP_PARTITION_TYPE_DATA,
        ESP_PARTITION_SUBTYPE_DATA_COREDUMP,
    "coredump");
    if (!core_part) {
        logger.write("No core dump partition found!|n");
        return ESP_ERR_NOT_FOUND;
    }
    if (core_part->size < sizeof(uint32_t)) {
        logger.write("Too small core dump partition!\n");
        return ESP_ERR_INVALID_SIZE;
    }

    esp_err_t err = ESP_OK;
    err = esp_partition_erase_range(core_part, 0, core_part->size);
    if (err != ESP_OK) {
        Serial.printf("Failed to erase core dump partition (%d)!\n", err);
        return err;
    }

    // on encrypted flash esp_partition_erase_range will leave encrypted
    // garbage instead of 0xFFFFFFFF so overwriting again to safely signalize
    // deleted coredumps
    const uint32_t invalid_size = 0xFFFFFFFF;
    err = esp_partition_write(core_part, 0, &invalid_size, sizeof(invalid_size));
    if (err != ESP_OK) {
        logger.write("Failed to write core dump partition size (%d)!\n", err);
    }

    return err;
}
```

hitecSmartHome, 2024-04-16:
```cpp
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.

Core  0 register dump:
PC      : 0x3268e3a2  PS      : 0x00060830  A0      : 0x38b38c81  A1      : 0x3ffd13e0  
A2      : 0x3ffd1420  A3      : 0x3f81c66c  A4      : 0x00ba41c3  A5      : 0x00004240   
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x8026f858  A9      : 0x00000000  
A10     : 0x3ffd1420  A11     : 0x3f81c66c  A12     : 0x1f2638c2  A13     : 0x3ff5f078   
A14     : 0x3ff5f07c  A15     : 0x00000001  SAR     : 0x00000001  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x40083715  LEND    : 0x4008371d  LCOUNT  : 0x00000027   

Backtrace: 0x3268e39f:0x3ffd13e0 |<-CORRUPTED

  #0  0x3268e39f:0x3ffd13e0 in ?? ??:0

ELF file SHA256: 93ed632490e2a93d

I (7513) esp_core_dump_flash: Save core dump to flash...
I (7520) esp_core_dump_flash: Erase flash 28672 bytes @ 0xfec000
[0;32mI (7747) esp_core_dump_flash: Write end offset 0x6c84, check sum length 4
I (7747) esp_core_dump_flash: Core dump has been saved to flash.
Rebooting...
```

This is the output to the console on crash.

Here is the console on boot

```
I (1387) esp_core_dump_flash: Init core dump to flash
I (1393) esp_core_dump_flash: Found partition 'coredump' @ fec000 65536 bytes
[0;32mI (1416) esp_core_dump_flash: Core dump data checksum is correct
I (1416) esp_core_dump_flash: Found core dump 27076 bytes in flash @ 0xfec000
```

My app reads and saves it

```
[System] - Successfully wrote core dump
```

hitecSmartHome, 2024-04-16:
Where should i call `shasum -a 256`?  It is unknown target when i call it with `idf.py shasum -a 256 <path_to_built_elf>`

hitecSmartHome, 2024-04-16:
So i did it on windows CMD `certutil -hashfile C:\Users\Pc\Desktop\coreDump\coreDump.bin` and the sha of the coreDump.bin is `6506a9448fd3d9a30b3d70bc5b3cabfd6b0aa9c9` and the output of the coredump-info is  `Failed to load core dump: Invalid application image for coredump: coredump SHA256(93ed632490e2a93d) != app SHA256().`. I'm sure that this is the latest coreDump.bin and I did not recompile the firmware

erhankur, 2024-04-16:
OK. What do you see when you run the below command? It will read the coredump partition to a file and decode it.

`idf.py coredump-info -s esp-coredump.txt -b 115200` 

Then you can compare `esp-coredump.txt` and `your-coredump.txt`

You should be able to decode your file like this. 

```
espcoredump.py info_corefile -c your-coredump.txt -t raw <your_app_elf_file_path>
```

hitecSmartHome, 2024-04-17:
I can't see a menuconfig option like this

hitecSmartHome, 2024-04-17:
I have tried this command

`espcoredump.py info_corefile -c your-coredump.txt -t raw <your_app_elf_file_path>`

I replaced your-coredump.txt with the path of the coredump and the elf file path with the elf file path

And i got

```
Usage: idf.py info_corefile [OPTIONS]
Try 'idf.py info_corefile --help' for help.

Error: No such option: -c
```

hitecSmartHome, 2024-04-17:
If i do it like this

`idf.py info_corefile your-coredump.txt raw <your_app_elf_file_path>` ( without -t and -c )

```
ninja: error: unknown target 'info_corefile'
[0/1] Re-running CMake...
ninja failed with exit code 1, output of the command is in the
```

hitecSmartHome, 2024-04-17:
I must mention that I'm using Arduino as a component of IDF with PlatformIO.
I have installed IDF separately, launching the ESP-IDF 5.2 PowerShell on windows, cd into my project dir ( which is in PlatformIO directory ) and executing commands there.

erhankur, 2024-04-17:
Sorry, I don't have such a setup to test and see what is going on.  As a last check on my side, if you can send your latest application elf file (here or by email) along with the latest coredump.txt file, I can have a look. Otherwise, I  suggest you to consult PlatformIO community and come back here if something needs to be done on the IDF side.

hitecSmartHome, 2024-04-17:
I see. Thank you for your help. I appreciate it. Will try with the PlatformIO guys.

hitecSmartHome, 2024-05-07:
So the PIO guys said that it isn't their problem. It should be fixed IDF side.
What should i do?

- I have no option in menuconfig to enable the coredump partition read trought serial
- `idf.py coredump-info` and `coredump-debug` both throwing the same error `Failed to load core dump: Invalid application image for coredump: coredump SHA256(02fcd8c6a8f0ea07) != app SHA256()`
- I can not run coredump.py even if the full idf is installed.

**Please verify that my method to retrive and store the coredump data is correct and please make suggestions how to decode it.**

- ESP crashes in a remote location.
- ESP restarts
- ESP reads the partition, writes it to LittleFS filesystem as a coreDump.bin file
- ESP clears the partition memory upon successfull retrival and save
- File is downloadable for further investigation
- Developer can run idf.py of coredump.py to decode the coreDump file somehow??
- Make a fix for the crash???

**Code for this**

```cpp

void Sys::readCoreDumpAndSave(){
    size_t size = 0;
    size_t address = 0;
    // Get the pointer to the image
    if (esp_core_dump_image_get(&address, &size) != ESP_OK) {
        printf("[System] - Failed to get core dump info\n");
        return;
    }
    // Allocate an array for the image buffer with ps_malloc()
    uint8_t* coreDumpBuffer;
    if( !alloc(coreDumpBuffer, size) ) {
        printf("[System] - Failed to allocate core dump buffer\n");
        return;
    }
    const esp_partition_t* pt = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_COREDUMP, "coredump");
    if(!pt){
        printf("[System] - Failed to find coredump partition\n");
        free(coreDumpBuffer);
        return;
    }
    // Read the whole image at once
    esp_err_t er = esp_partition_read(pt, 0, coreDumpBuffer, size);
    if (er != ESP_OK) {
        printf(
            "[System] - Failed to read core dump. Error: %s\n",
            esp_err_to_name(er)
        );
        free(coreDumpBuffer);
        return;
    }
    File coreDumpFile = db.getFileObject(CORE_DUMP_PATH,FILE_WRITE);
    size_t written = coreDumpFile.write(coreDumpBuffer, size);
    free(coreDumpBuffer);
    if (written != size) {
        printf("[System] - Failed to write core dump\n");
        return;
    }
    printf("[System] - Successfully wrote core dump\n");
    esp_core_dump_image_erase();
}

esp_err_t Sys::esp_core_dump_image_erase() {
    /* Find the partition that could potentially contain a (previous) core dump. */
    const esp_partition_t* core_part = esp_partition_find_first(ESP_PARTITION_TYPE_DATA,
        ESP_PARTITION_SUBTYPE_DATA_COREDUMP,
    "coredump");
    if (!core_part) {
        logger.write("No core dump partition found!|n");
        return ESP_ERR_NOT_FOUND;
    }
    if (core_part->size < sizeof(uint32_t)) {
        logger.write("Too small core dump partition!\n");
        return ESP_ERR_INVALID_SIZE;
    }

    esp_err_t err = ESP_OK;
    err = esp_partition_erase_range(core_part, 0, core_part->size);
    if (err != ESP_OK) {
        Serial.printf("Failed to erase core dump partition (%d)!\n", err);
        return err;
    }

    // on encrypted flash esp_partition_erase_range will leave encrypted
    // garbage instead of 0xFFFFFFFF so overwriting again to safely signalize
    // deleted coredumps
    const uint32_t invalid_size = 0xFFFFFFFF;
    err = esp_partition_write(core_part, 0, &invalid_size, sizeof(invalid_size));
    if (err != ESP_OK) {
        logger.write("Failed to write core dump partition size (%d)!\n", err);
    }

    return err;
}

void readRebootReason() {
    esp_reset_reason_t resetReason = esp_reset_reason();
    if( resetReason != ESP_RST_SW && resetReason != ESP_RST_POWERON ){
        readCoreDumpAndSave();
    }
}

void setup(){
    readRebootReason();
}
```

Maybe if we would have an option to save the crash log as a verbose txt I could save that directly to FS and don't have to decode at all. I have plenty of flash space, this would be no problem.

**What I have tried so far**

- Open ESP-IDF 5.2 PowerShell
- cd to project dir
- idf.py coredump-debug -c <path-to-coredump.bin> <path-to-firmware.elf>
- idf.py coredump-debug -c <path-to-coredump.bin> <path-to-firmware.bin>
- idf.py coredump-debug -c <path-to-coredump.bin>
- idf.py coredump-info -c <path-to-coredump.bin> <path-to-firmware.elf>
- idf.py coredump-info -c <path-to-coredump.bin> <path-to-firmware.bin>
- idf.py coredump-info -c <path-to-coredump.bin>
- espcoredump.py is not recognized at all
- coredump.py is not recognized at all
- Tried to copy/paste the raw bin into Arduino IDE expection decoder

erhankur, 2024-05-07:
Now I have Windows environment to test. What I have done so far, 
1 - Run ESP-IDF 5.2 Powershell
2 - pip list to see installed packages.
```
pip list
Package               Version
--------------------- ---------
bitarray              2.8.2
bitstring             4.1.2
CacheControl          0.13.1
certifi               2023.7.22
cffi                  1.16.0
charset-normalizer    3.3.1
click                 8.1.7
colorama              0.4.6
construct             2.10.69
contextlib2           21.6.0
cryptography          41.0.7
ecdsa                 0.18.0
esp-coredump          1.10.0
```
3 - I can see esp-coredump is installed. (If not run  .\install.ps1 and  .\export.ps1)
4 - Run espcoredump.py
```
esp-coredump.exe info_corefile -c coredump.txt -t raw dummy.elf
```
5 - Got error. It is OK since I don't have original elf file.
```
espcoredump.py v1.10.0
Failed to load core dump: Invalid application image for coredump: coredump SHA256(b7fbaab6626d2226) != app SHA256(8bdd9f3ae7de7911)
```

Please try the steps.

hitecSmartHome, 2024-05-07:
Thank you for the suggestion. If i try with an incompatible elf file it produces the same output as yours. But if I crash the ESP like this:

```cpp
printf("Crashing...\n");
while(true){
    int i = 0;
    int crash = 1/i;
    printf("Crash: %d\n", crash);
}
```

It errors out:

```
espcoredump.py v1.11.0
Failed to load core dump: Core dump version "0x3030" is not supported!
```

hitecSmartHome, 2024-05-07:
`esp-coredump          1.11.0`

hitecSmartHome, 2024-05-07:
Here is a link to the core dump files ( I could not upload it to github )

https://drive.google.com/drive/folders/1oWlqmrAvXPMyer-4162RN21DQHfX8lVf?usp=sharing

erhankur, 2024-05-07:
I can decode it.

Your header looks fine. Header version  0x100 meaning CRC - 20 byte header.

```
64 6F 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00
````

```
esp-coredump.exe --gdb-timeout-sec 10 info_corefile -c .\coreDump.bin -t raw .\firmware.elf

hitecSmartHome, 2024-05-07:
Well, I did it with exactly this command.
What did you get as a result?

hitecSmartHome, 2024-05-07:
What did i do wrong? I'm using ESP-IDF 5.2 PowerShell, navigated to the project root dir

hitecSmartHome, 2024-05-07:
Oh okay, I copyed the files to my project dir and run the test again with gdb timeout 10 and i got a result finally!!

In the end i got this for some reason

```
===================== ESP32 CORE DUMP END =====================
===============================================================
Exception in thread Thread-1 (_readerthread):
Traceback (most recent call last):
  File "threading.py", line 1038, in _bootstrap_inner
Exception in thread Thread-2 (_readerthread):
Traceback (most recent call last):
  File "threading.py", line 1038, in _bootstrap_inner
  File "threading.py", line 975, in run
  File "threading.py", line 975, in run
  File "subprocess.py", line 1552, in _readerthread
  File "subprocess.py", line 1552, in _readerthread
OSError: [Errno 22] Invalid argument
OSError: [Errno 22] Invalid argument
Done!
```

hitecSmartHome, 2024-05-07:
Anyway, this gave me a lot of usefull info! Thank you very much!

closed 2024-05-07 by erhankur

hitecSmartHome, 2024-05-07:
Thank you very much.

saiteja0990, 2025-01-21:
hi,  @erhankur 
i have tried to get coerdump data from espressif ide coredump partition but, i am getting error like.....
error[ espcoredump.py v1.12.0
Failed to load core dump: Core dump version "0xffff" is not supported! ]
i used command "esp-coredump -p com3 info_corefile build/app-template.elf"

and i also tried saving coredump data in txt file and run this command "espcoredump.py --gdb-timeout-sec 10 info_corefile -c .\crashLogs3.bin -t raw .\delta.ino.elf" 
error [Executing action: coredump-debug
Failed to load core dump: Invalid core dump CRC 5b12b31, should be ffffffff]

i used esp32 4mb board 

[crashLogs.txt](https://github.com/user-attachments/files/18488790/crashLogs.txt)

erhankur, 2025-01-21:
First, make sure the coredump has actually been saved to flash.
A core dump version of "0xffff" indicates there is no written coredump data. Could it have been erased before you read it?

Also, please share the exact commands you used, step by step, with the same application elf file.
I see references to app-template.elf, delta.ino.elf, crashLogs3.bin, and crashLogs3.txt. 
Hard to follow what is going on there.

saiteja0990, 2025-01-22:
Thank you for response, issue resolved based on u r previous explanation

KlaasjanN, 2025-07-18:
@erhankur I noticed you referenced esp-coredump.exe, where can I find that file? (In .exe format, not .py). When was it dropped? I thought I was going crazy as I can't find it anymore.

erhankur, 2025-07-18:
On Windows, pip generates a launcher executable in the Scripts directory of the Python environment. 
Check it here. `c:\Espressif\python_env\idf5.4_py3.11_env\Scripts\`