espressif/esp-idf #13619: ESP32C6 BLE crash when advertising + PM + NimBLE

malachib, 2024-04-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

While attempting BLE Advertise while using NimBLE, PM + Modem Sleep I run into problems.  I'm following the lead of examples/bluetooth/nimble/power_save (which flashes and runs without incident)

When specifying advertising intervals other than 0 and CONFIG_ESP_SLEEP_POWER_DOWN_FLASH is set, I get a reset right after begin advertise:

```
rst:0x8 (TG1_WDT_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
```

If I turn off all the watchdogs, it changes to:

```
rst:0x10 (LP_WDT_SYS),boot:0xc (SPI_FAST_FLASH_BOOT)
```

zhaoweiliang2021, 2024-04-19:
@malachib 
Hi, can you provide an example that reproduces this issue?  And contains the corresponding sdkconfig file. Thanks

malachib, 2024-04-21:
Yes definitely - I am on the road will be back in a couple days and set it up

malachib, 2024-04-23:
Here's a capture.  Be sure to set CONFIG_ESP_SLEEP_POWER_DOWN_FLASH on:
[23APR24.PGESP-49-2.tar.gz](https://github.com/espressif/esp-idf/files/15079855/23APR24.PGESP-49-2.tar.gz)

Let me know if you need the example cut down more.  I think it's succinct enough to clearly understand the issue.

zhaoweiliang2021, 2024-04-24:
@malachib [bugfix_patch.zip](https://github.com/espressif/esp-idf/files/15087456/bugfix_patch.zip)
After decompression, you will get the bugfix.patch file. Put the patch file into the idf (v5.1.3) root directory.
run "git apply bugfix.patch"
My local test has passed.
The final fix awaits the release of the new version 5.1. Thanks.

malachib, 2024-04-24:
>  My local test has passed. The final fix code awaits the release of the new version 5.1. Thanks.

Thank you for this!  Will try it shortly

AxelLin, 2024-05-21:
> The final fix code awaits the release of the new version 5.1. Thanks.

@zhaoweiliang2021 
Is this fixed in esp-idf ? If yes, please share the commit of the fix.

zhaoweiliang2021, 2024-05-22:
@AxelLin bugfix commits for this issue on version v5.1.4 : commit (12dca8d0866bc3276d69c9280841522a321b65fa) and commit (54a15b81f98692bafc062c729dd5b73fa98e51c2)

AxelLin, 2024-07-15:
> @AxelLin bugfix commits for this issue on version v5.1.4 : commit ([12dca8d](https://github.com/espressif/esp-idf/commit/12dca8d0866bc3276d69c9280841522a321b65fa)) and commit ([54a15b8](https://github.com/espressif/esp-idf/commit/54a15b81f98692bafc062c729dd5b73fa98e51c2))

@zhaoweiliang2021 Please close the issue if all impacted branches are fixed.

closed 2024-07-24 by Indastri