espressif/esp-idf #13620: Setting an ETH manufacturer

Flamabalistic, 2024-04-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Is it possible to set the manufacturer of an ESP32-S3 connected via ETH, such that it shows up when doing an `arp-scan`?

david-cermak, 2024-05-02:
You mean the vendor prefix of the MAC address?
Please check this example (S3 with SPI Ethernet)

https://github.com/espressif/esp-idf/blob/d4cd437ede613fffacc06ac6d6c93a083829022f/examples/ethernet/basic/components/ethernet_init/ethernet_init.c#L263-L270

where we use the locally administered OUI range, you can supply your own manufacturer's reserved address.

Flamabalistic, 2024-05-29:
I did some more research as I didn't fully understand what I was asking!

For anybody else's sake who finds this (feel free to add/correct anything):
- MAC addresses are typically allocated by IEEE
- When performing an ARP scan - the name returned is the name allocated by IEEE, ESP does NOT have control over this
- Read here for more info https://standards.ieee.org/products-programs/regauth/

closed 2024-05-29 by Flamabalistic