espressif/esp-idf #13621: ESP-BLE-MESH- directed_forwarding- 'Ran out of retransmit attempts'

besuree, 2024-04-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm trying to gain some experience with the BLE network in its new version V1.1 with examples from ESP32.  I'm using as client the project in [examples/bluetooth/esp_ble_mesh/directed_forwarding/df_client](url) as server the project in [examples/bluetooth/esp_ble_mesh/directed_forwarding/df_server](url) 

- No. 5 ESP32-C6-DevKitC1 V1.2 devices running the df_server project.
- No. 1 ESP32-C6 DevKitC1 V1.2 device running the df_client project.

[image]

After Provisioner (detected by the lighting of the white colored LEDs on both nand client), pressing the 'boot' button the client starts sending messages but goes into timeout. 

This is what I get from the client serial at boot time:
```
F file SHA256:ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x15 (USB_UART_HPSYS),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001fed8
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x1804
load:0x4086c410,len:0xe58
load:0x4086e610,len:0x2e24
entry 0x4086c41a
[1B][0;32mI (23) boot: ESP-IDF v5.2.1 2nd stage bootloader[1B][0m
[1B][0;32mI (23) boot: compile time Apr 16 2024 10:39:43[1B][0m
[1B][0;32mI (24) boot: chip revision: v0.0[1B][0m
[1B][0;32mI (26) boot.esp32c6: SPI Speed      : 80MHz[1B][0m
[1B][0;32mI (30) boot.esp32c6: SPI Mode       : DIO[1B][0m
[1B][0;32mI (35) boot.esp32c6: SPI Flash Size : 2MB[1B][0m
[1B][0;32mI (40) boot: Enabling RNG early entropy source...[1B][0m
[1B][0;32mI (45) boot: Partition Table:[1B][0m
[1B][0;32mI (49) boot: ## Label            Usage          Type ST Offset   Length[1B][0m
[1B][0;32mI (56) boot:  0 nvs              WiFi data        01 02 00009000 00006000[1B][0m
[1B][0;32mI (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000[1B][0m
[1B][0;32mI (71) boot:  2 factory          factory app      00 00 00010000 00177000[1B][0m
[1B][0;32mI (79) boot: End of partition table[1B][0m
[1B][0;32mI (83) esp_image: segment 0: paddr=00010020 vaddr=420f0020 size=3e3f0h (254960) map[1B][0m
[1B][0;32mI (144) esp_image: segment 1: paddr=0004e418 vaddr=40800000 size=01c00h (  7168) load[1B][0m
[1B][0;32mI (147) esp_image: segment 2: paddr=00050020 vaddr=42000020 size=ee8f0h (977136) map[1B][0m
[1B][0;32mI (351) esp_image: segment 3: paddr=0013e918 vaddr=40801c00 size=0ff1ch ( 65308) load[1B][0m
[1B][0;32mI (367) esp_image: segment 4: paddr=0014e83c vaddr=40811b20 size=02b6ch ( 11116) load[1B][0m
[1B][0;32mI (375) boot: Loaded app from partition at offset 0x10000[1B][0m
[1B][0;32mI (376) boot: Disabling RNG early entropy source...[1B][0m
[1B][0;32mI (387) cpu_start: Unicore app[1B][0m
[1B][0;33mW (397) clk: esp_perip_clk_init() has not been implemented yet[1B][0m
[1B][0;32mI (403) cpu_start: Pro cpu start user code[1B][0m
[1B][0;32mI (403) cpu_start: cpu freq: 160000000 Hz[1B][0m
[1B][0;32mI (404) cpu_start: Application information:[1B][0m
[1B][0;32mI (406) cpu_start: Project name:     onoff_client[1B][0m
[1B][0;32mI (412) cpu_start: App version:      v5.2.1[1B][0m
[1B][0;32mI (416) cpu_start: Compile time:     Apr 16 2024 10:36:31[1B][0m
[1B][0;32mI (422) cpu_start: ELF file SHA256:  4dd2bfe73...[1B][0m
[1B][0;32mI (428) cpu_start: ESP-IDF:          v5.2.1[1B][0m
[1B][0;32mI (433) cpu_start: Min chip rev:     v0.0[1B][0m
[1B][0;32mI (437) cpu_start: Max chip rev:     v0.99 [1B][0m
[1B][0;32mI (442) cpu_start: Chip rev:         v0.0[1B][0m
[1B][0;32mI (447) heap_init: Initializing. RAM available for dynamic allocation:[1B][0m
[1B][0;32mI (454) heap_init: At 4081C820 len 0005FDF0 (383 KiB): RAM[1B][0m
[1B][0;32mI (460) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM[1B][0m
[1B][0;32mI (466) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM[1B][0m
[1B][0;32mI (474) spi_flash: detected chip: generic[1B][0m
[1B][0;32mI (477) spi_flash: flash io: dio[1B][0m
[1B][0;33mW (481) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.[1B][0m
[1B][0;32mI (495) sleep: Configure to isolate all GPIO pins in sleep state[1B][0m
[... 397 lines cut ...]
[1B][0;32mI (531) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 [1B][0m
[1B][0;32mI (551) BLE_INIT: Using main XTAL as clock source[1B][0m
[1B][0;32mI (551) BLE_INIT: ble controller commit:[217f1bf][1B][0m
[1B][0;32mI (551) phy_init: phy_version 250,e14681b,Jan 24 2024,17:43:11[1B][0m
[1B][0;32mI (601) phy: libbtbb version: 939f79c, Jan 24 2024, 17:43:26[1B][0m
[1B][0;32mI (611) EXAMPLE:Provisioner bd address: : 40 4c ca 44 ae 7e [1B][0m
Bluetooth Mesh v1.1 commit: [154b4fcc98]
[1B][0;32mI (671) EXAMPLE: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0[1B][0m
[1B][0;32mI (671) EXAMPLE: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code 0[1B][0m
[1B][0;32mI (671) EXAMPLE: BLE Mesh Node initialized[1B][0m
[1B][0;32mI (681) main_task: Returned from app_main()[1B][0m
[1B][0;32mI (20631) EXAMPLE: ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer PB-GATT[1B][0m
[1B][0;32mI (21031) EXAMPLE: ESP_BLE_MESH_NODE_PROV_LINK_CLOSE_EVT, bearer PB-GATT[1B][0m
[1B][0;32mI (21041) EXAMPLE: ESP_BLE_MESH_NODE_PROV_COMPLETE_EVT[1B][0m
[1B][0;32mI (21041) EXAMPLE: net_idx: 0x0000, addr: 0x0002[1B][0m
[1B][0;32mI (21041) EXAMPLE: flags: 0x00, iv_index: 0x00000000[1B][0m
[1B][0;32mI (21051) EXAMPLE: ESP_BLE_MESH_PROVISIONER_ADD_LOCAL_APP_KEY_COMP_EVT, err_code 0[1B][0m
Mismatch proxy client type 0x00
Mismatch proxy client type 0x00
Mismatch proxy client type 0x00
[1B][0;33mW (21111) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x13[1B][0m

``` 
On the server I get "Mismatch proxy client type 0x00" 
Any ideas?

besuree, 2024-04-18:
Has anyone experienced the same problem ?

forx157, 2024-04-19:
Hi, @besuree 
Sorry no one got back to you earlier.
I need some more information in order to help you debug this, so could you enable the `INFO` log level (path: `(Top) → Component config → ESP BLE Mesh Support → BLE Mesh STACK DEBUG LOG LEVEL → BLE_MESH_STACK`) and share the log with me?

besuree, 2024-04-19:
Hi, @forx157 
Thanks for your response.

- This time I used 4 devices with 'df_server' project and 1 device with 'df_client'. 
- I changed the debug level to INFO for all the device (server and client)
 
**Test description**
With the relevant projects loaded on the devices, I first turn on the device with 'df_client' project. Then I turn on the devides one at a time with 'df_server' project.

**Serial output**
node-0 serial output:
```
[1B][0;32mI (345) esp_image: segment 3: paddr=001321b8 vaddr=408042esp_timerESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x15 (USB_UART_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x400294a6
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x1804
load:0x4086c410,len:0xe58
load:0x4086e610,len:0x2e24
entry 0x4086c41a
[1B][0;32mI (23) boot: ESP-IDF v5.2.1-dirty 2nd stage bootloader[1B][0m
[1B][0;32mI (23) boot: compile time Apr 19 2024 08:28:38[1B][0m
[1B][0;32mI (24) boot: chip revision: v0.0[1B][0m
[1B][0;32mI (26) boot.esp32c6: SPI Speed      : 80MHz[1B][0m
[1B][0;32mI (31) boot.esp32c6: SPI Mode       : DIO[1B][0m
[1B][0;32mI (36) boot.esp32c6: SPI Flash Size : 2MB[1B][0m
[1B][0;32mI (40) boot: Enabling RNG early entropy source...[1B][0m
[1B][0;32mI (46) boot: Partition Table:[1B][0m
[1B][0;32mI (49) boot: ## Label            Usage          Type ST Offset   Length[1B][0m
[1B][0;32mI (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000[1B][0m
[1B][0;32mI (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000[1B][0m
[1B][0;32mI (72) boot:  2 factory          factory app      00 00 00010000 00177000[1B][0m
[1B][0;32mI (79) boot: End of partition table[1B][0m
[1B][0;32mI (83) esp_image: segment 0: paddr=00010020 vaddr=420e8020 size=3bd58h (245080) map[1B][0m
[1B][0;32mI (142) esp_image: segment 1: paddr=0004bd80 vaddr=40800000 size=04298h ( 17048) load[1B][0m
[1B][0;32mI (147) esp_image: segment 2: paddr=00050020 vaddr=42000020 size=e2190h (926096) map[1B][0m
[1B][0;32mI (339) esp_image: segment 3: paddr=001321b8 vaddr=40804298 size=104dch ( 66780) load[1B][0m
[1B][0;32mI (360) boot: Loaded app from partition at offset 0x10000[1B][0m
[1B][0;32mI (361) boot: Disabling RNG early entropy source...[1B][0m
[1B][0;32mI (372) cpu_start: Unicore app[1B][0m
[1B][0;33mW (381) clk: esp_perip_clk_init() has not been implemented yet[1B][0m
[1B][0;32mI (387) cpu_start: Pro cpu start user code[1B][0m
[1B][0;32mI (387) cpu_start: cpu freq: 160000000 Hz[1B][0m
[1B][0;32mI (388) cpu_start: Application information:[1B][0m
[1B][0;32mI (390) cpu_start: Project name:     onoff_server[1B][0m
[1B][0;32mI (396) cpu_start: App version:      v5.2.1-dirty[1B][0m
[1B][0;32mI (401) cpu_start: Compile time:     Apr 19 2024 08:28:03[1B][0m
[1B][0;32mI (407) cpu_start: ELF file SHA256:  df5979065...[1B][0m
[1B][0;32mI (412) cpu_start: ESP-IDF:          v5.2.1-dirty[1B][0m
[1B][0;32mI (418) cpu_start: Min chip rev:     v0.0[1B][0m
[1B][0;32mI (422) cpu_start: Max chip rev:     v0.99 [1B][0m
[1B][0;32mI (427) cpu_start: Chip rev:         v0.0[1B][0m
[1B][0;32mI (432) heap_init: Initializing. RAM available for dynamic allocation:[1B][0m
[1B][0;32mI (439) heap_init: At 4081C1B0 len 00060460 (385 KiB): RAM[1B][0m
[1B][0;32mI (445) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM[1B][0m
[1B][0;32mI (451) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM[1B][0m
[1B][0;32mI (459) spi_flash: detected chip: generic[1B][0m
[1B][0;32mI (462) spi_flash: flash io: dio[1B][0m
[1B][0;33mW (466) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.[1B][0m
[1B][0;32mI (480) sleep: Configure to isolate all GPIO pins in sleep state[1B][0m
[1B][0;32mI (486) sleep: Enable automatic switching of GPIO sleep configuration[1B][0m
[1B][0;32mI (493) coexist: coex firmware version: 77cd7f8[1B][0m
[1B][0;32mI (499) coexist: coexist rom version 5b8dcfa[1B][0m
[1B][0;32mI (504) main_task: Started on CPU0[1B][0m
[1B][0;32mI (504) main_task: Calling app_main()[1B][0m
[1B][0;32mI (504) EXAMPLE: Initializing...[1B][0m
[1B][0;32mI (514) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (524) led_encoder: Install led strip encoder[1B][0m
[... 1904 lines cut ...]
[1B][0;32mI (439295) BLE_MESH: Attempts: 3[1B][0m
[1B][0;32mI (439295) BLE_MESH: Resending 0/1, cred 0x00[1B][0m
[1B][0;32mI (439295) BLE_MESH: Resending 1/1, cred 0x00[1B][0m
[1B][0;32mI (440025) BLE_MESH: Attempts: 2[1B][0m
[1B][0;32mI (440025) BLE_MESH: Resending 0/1, cred 0x00[1B][0m
[1B][0;32mI (440025) BLE_MESH: Resending 1/1, cred 0x00[1B][0m
[1B][0;32mI (440755) BLE_MESH: Attempts: 1[1B][0m
[1B][0;32mI (440755) BLE_MESH: Resending 0/1, cred 0x00[1B][0m
[1B][0;32mI (440755) BLE_MESH: Resending 1/1, cred 0x00[1B][0m
[1B][0;32mI (441485) BLE_MESH: Attempts: 0[1B][0m
[1B][0;32mI (441485) BLE_MESH: Resending 0/1, cred 0x00[1B][0m
[1B][0;32mI (441485) BLE_MESH: Resending 1/1, cred 0x00[1B][0m
``` 

Before I press the button ' boot' from some server devices I get 
```
Directed Forwarding Timeout, opcode 0x807c, to 0x0003
```  
but after a short time, direct forwoding is enabled. When all the devices have direct forwading enabled, I press the 'boot' button.  After pressing the button, on node 0 I see  ```BLE_MESH: No matching TX context for Seg Ack``` .
[... cut ...] oner's LED is white, it indicates that it is provisioning a specific Node. When the provisioning process is complete, the LED of the provisioned Node will turn to white, and this process continues until all nodes join the network and the LEDs are white.' I do not notice this. I notice that the led is white during provisoning but when I get 
```PB-ADV link close, reason 0x00 ``` it turns off.  So provisioning is not finished? this for the "Mismatch proxy client type" error?

Thanks for your help.

besuree, 2024-04-19:
Hi, @forx157 
No, I did not make any changes. I downloaded the IDF-5.2.1 version from [this link](https://dl.espressif.com/dl/esp-idf/) and compiled the example through IDF:
once the IDF was launched I changed the directory to  "C:\Espressif\frameworks\esp-idf v5.2.1\examples\bluetooth\esp_ble_mesh\directed_forwarding\df_server" I ran
```idf.py set-target esp32c6``` 
```idf.py build``` 
```idf.py -p (COM) flash``` 
the same for "df_client" project

besuree, 2024-04-22:
Hi, @forx157 
I downloaded the IDF 5.2 version from git.
I compiled both projects "df_client" and "df_server" with log "VERBOSE". 

these are the results:

**CLIENT**
```
t: App verESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x15 (USB_UART_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x400294a6
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x1804
load:0x4086c410,len:0xde0
load:0x4086e610,len:0x2e04
entry 0x4086c41a
[1B][0;32mI (23) boot: ESP-IDF v5.2 2nd stage bootloader[1B][0m
[1B][0;32mI (23) boot: compile time Apr 22 2024 10:29:25[1B][0m
[1B][0;32mI (24) boot: chip revision: v0.0[1B][0m
[1B][0;32mI (25) boot.esp32c6: SPI Speed      : 80MHz[1B][0m
[1B][0;32mI (30) boot.esp32c6: SPI Mode       : DIO[1B][0m
[1B][0;32mI (35) boot.esp32c6: SPI Flash Size : 2MB[1B][0m
[1B][0;32mI (40) boot: Enabling RNG early entropy source...[1B][0m
[1B][0;32mI (45) boot: Partition Table:[1B][0m
[1B][0;32mI (49) boot: ## Label            Usage          Type ST Offset   Length[1B][0m
[1B][0;32mI (56) boot:  0 nvs              WiFi data        01 02 00009000 00006000[1B][0m
[1B][0;32mI (63) boot:  1 phy_init         RF data          01 01 0000f000 00001000[1B][0m
[1B][0;32mI (71) boot:  2 factory          factory app      00 00 00010000 00177000[1B][0m
[1B][0;32mI (79) boot: End of partition table[1B][0m
[1B][0;32mI (83) esp_image: segment 0: paddr=00010020 vaddr=420f8020 size=41b60h (269152) map[1B][0m
[1B][0;32mI (146) esp_image: segment 1: paddr=00051b88 vaddr=40800000 size=06490h ( 25744) load[1B][0m
[1B][0;32mI (153) esp_image: segment 2: paddr=00058020 vaddr=42000020 size=f367ch (996988) map[1B][0m
[1B][0;32mI (356) esp_image: segment 3: paddr=0014b6a4 vaddr=40806490 size=0b588h ( 46472) load[1B][0m
[1B][0;32mI (367) esp_image: segment 4: paddr=00156c34 vaddr=40811a20 size=02b60h ( 11104) load[1B][0m
[1B][0;32mI (375) boot: Loaded app from partition at offset 0x10000[1B][0m
[1B][0;32mI (376) boot: Disabling RNG early entropy source...[1B][0m
[1B][0;32mI (388) cpu_start: Unicore app[1B][0m
[1B][0;33mW (397) clk: esp_perip_clk_init() has not been implemented yet[1B][0m
[1B][0;32mI (403) cpu_start: Pro cpu start user code[1B][0m
[1B][0;32mI (404) cpu_start: cpu freq: 160000000 Hz[1B][0m
[1B][0;32mI (404) cpu_start: Application information:[1B][0m
[1B][0;32mI (406) cpu_start: Project name:     onoff_client[1B][0m
[1B][0;32mI (412) cpu_start: App version:      v5.2[1B][0m
[1B][0;32mI (417) cpu_start: Compile time:     Apr 22 2024 10:07:14[1B][0m
[1B][0;32mI (423) cpu_start: ELF file SHA256:  ef5a953a6...[1B][0m
[1B][0;32mI (428) cpu_start: ESP-IDF:          v5.2[1B][0m
[1B][0;32mI (433) cpu_start: Min chip rev:     v0.0[1B][0m
[1B][0;32mI (437) cpu_start: Max chip rev:     v0.99 [1B][0m
[1B][0;32mI (442) cpu_start: Chip rev:         v0.0[1B][0m
[1B][0;32mI (447) heap_init: Initializing. RAM available for dynamic allocation:[1B][0m
[1B][0;32mI (454) heap_init: At 4081C700 len 0005FF10 (383 KiB): RAM[1B][0m
[1B][0;32mI (460) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM[1B][0m
[1B][0;32mI (466) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM[1B][0m
[1B][0;32mI (474) spi_flash: detected chip: generic[1B][0m
[1B][0;32mI (477) spi_flash: flash io: dio[1B][0m
[1B][0;33mW (481) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.[1B][0m
[1B][0;32mI (495) sleep: Configure to isolate all GPIO pins in sleep state[1B][0m
[1B][0;32mI (501) sleep: Enable automatic switching of GPIO sleep configuration[1B][0m
[1B][0;32mI (508) coexist: coex firmware version: 77cd7f8[1B][0m
[1B][0;32mI (514) coexist: coexist rom version 5b8dcfa[1B][0m
[1B][0;32mI (519) main_task: Started on CPU0[1B][0m
[1B][0;32mI (519) main_task: Calling app_main()[1B][0m
[1B][0;32mI (519) EXAMPLE: Initializing...[1B][0m
[1B][0;32mI (529) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (539) led_encoder: Install led strip encoder[1B][0m
[1B][0;32mI (539) led_encoder: Enable RMT TX channel[1B][0m
use esp timer !!!
[1B][0;31mE (549) esp_timer: Task is already initialized[1B][0m
[... 1223 lines cut ...]
[1B][0;32mI (34957) BLE_MESH: Network PDU, count 2, interval 20[1B][0m
[1B][0;32mI (35557) BLE_MESH: Use security credentials 0x00, proxy 0[1B][0m
[1B][0;32mI (35557) BLE_MESH: Network PDU, count 2, interval 20[1B][0m
[1B][0;32mI (36157) BLE_MESH: Use security credentials 0x00, proxy 0[1B][0m
[1B][0;32mI (36167) BLE_MESH: Network PDU, count 2, interval 20[1B][0m
[1B][0;32mI (36767) BLE_MESH: Use security credentials 0x00, proxy 0[1B][0m
[1B][0;32mI (36767) BLE_MESH: Network PDU, count 2, interval 20[1B][0m
[1B][0;32mI (37367) BLE_MESH: Use security credentials 0x00, proxy 0[1B][0m
[1B][0;32mI (37377) BLE_MESH: Network PDU, count 2, interval 20[1B][0m
[1B][0;32mI (37977) BLE_MESH: Use security credentials 0x00, proxy 0[1B][0m
[1B][0;32mI (37977) BLE_MESH: Network PDU, count 2, interval 20[1B][0m
[1B][0;32mI (38577) BLE_MESH: Use security credentials 0x00, proxy 0[1B][0m
[1B][0;32mI (38587) BLE_MESH: Network PDU, count 2, interval 20[1B][0m
[1B][0;32mI (39187) BLE_MESH: Use security credentials 0x00, proxy 0[1B][0m
[1B][0;32mI (39187) BLE_MESH: Network PDU, count 2, interval 20[1B][0m
[1B][0;32mI (39797) BLE_MESH: Use security credentials 0x00, proxy 0[1B][0m
[1B][0;32mI (39797) BLE_MESH: Network PDU, count 2, interval 20[1B][0m
[1B][0;32mI (40397) BLE_MESH: Use security credentials 0x00, proxy 0[1B][0m
[1B][0;32mI (40397) BLE_MESH: Network PDU, count 2, interval 20[1B][0m
[1B][0;32mI (41007) BLE_MESH: Use security credentials 0x00, proxy 0[1B][0m
[1B][0;32mI (41007) BLE_MESH: Network PDU, count 2, interval 20[1B][0m
[1B][0;32mI (41607) BLE_MESH: Use security credentials 0x00, proxy 0[1B][0m
[1B][0;32mI (41607) BLE_MESH: Network PDU, count 2, interval 20[1B][0m
[1B][0;33mW (42217) BLE_MESH: Incomplete timer expired[1B][0m
```

bhcuong2008, 2025-04-07:
I also see it during my provision with ESP32-C6 chip, esp-idf 5.4.1, NimBLE host stack, in Arduino environment:
```
[   107][I][ble_mesh_init.cpp:133] mesh_host_task(): [MESH_INIT] BLE Host Task Started
I (1532) BLE_MESH(lib): Bluetooth Mesh v1.1 commit: [meshlib:8a5446cb8d]
[  1155][I][onoff_server.cpp:186] BleMeshProvisioningCallback(): [ONOFF_SERVER] ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0
[  1168][I][onoff_server.cpp:189] BleMeshProvisioningCallback(): [ONOFF_SERVER] ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code 0
[  1179][I][onoff_server.cpp:320] InitBleMesh(): [ONOFF_SERVER] BLE Mesh Node initialized
[148106][I][onoff_server.cpp:192] BleMeshProvisioningCallback(): [ONOFF_SERVER] ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer PB-GATT
[148799][I][onoff_server.cpp:196] BleMeshProvisioningCallback(): [ONOFF_SERVER] ESP_BLE_MESH_NODE_PROV_LINK_CLOSE_EVT, bearer PB-GATT
[148811][I][onoff_server.cpp:200] BleMeshProvisioningCallback(): [ONOFF_SERVER] ESP_BLE_MESH_NODE_PROV_COMPLETE_EVT
[148821][I][onoff_server.cpp:107] prov_complete(): [ONOFF_SERVER] net_idx: 0x0001, addr: 0x0002
[148832][I][onoff_server.cpp:108] prov_complete(): [ONOFF_SERVER] flags: 0x00, iv_index: 0x00000000
[148841][W][board.cpp:31] board_led_operation(): [BOARD] led green is already off
W (152361) BLE_MESH(lib): Mismatch proxy client type 0x00
W (152362) BLE_MESH(lib): Mismatch proxy client type 0x00
W (152363) BLE_MESH(lib): Mismatch proxy client type 0x00
D (186179) BLE_MESH(lib): len 3 bytes: 040000
D (186194) BLE_MESH(lib): len 3 bytes: 040000
D (186262) BLE_MESH(lib): NID 0x39 net_idx 0x0001
D (186264) BLE_MESH(lib): NID 0x39 net_idx 0x0001
[184858][I][onoff_server.cpp:268] BleMeshConfigServerCallback(): [ONOFF_SERVER] ESP_BLE_MESH_MODEL_OP_APP_KEY_ADD
D (186277) BLE_MESH(lib): NID 0x39 net_idx 0x0001
[184873][I][onoff_server.cpp:269] BleMeshConfigServerCallback(): [ONOFF_SERVER] net_idx 0x0001, app_idx 0x0001
D (186292) BLE_MESH(lib): NID 0x39 net_idx 0x0001
0xbb, 0x31, 0x74, 0xd6, 0xed, 0xd9, 0xf5, 0xd3, 0x50, 0x45, 0x42, 0x0d, 0x5c, 0xff, 0x31, 0x64, // .1t.....PEB.\.1d
```
I dont understand the meaning of `W (152361) BLE_MESH(lib): Mismatch proxy client type 0x00`

forx157, 2025-04-07:
Hi, @bhcuong2008 
Currently you can ignore the `Mismatch proxy client type 0x00` warning.

bhcuong2008, 2025-04-07:
@forx157 I have updated my sdkconfig so it does not appear that warning again. I could not recreate that warning again. All working fine now. Thank you.

closed 2025-04-08 by Alvin1Zhang