espressif/esp-idf #13623: WS client disconnect crash

hitecSmartHome, 2024-04-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.6 2nd stage bootloader

### Espressif SoC revision.

ESP-32 Wrover-E revision v3.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP-32 Wrover-E

### Power Supply used.

External 3.3V

### What is the expected behavior?

I expected it not to crash when the server goes down.

### What is the actual behavior?

It is crashing.

### Steps to reproduce.

Make a websocket client connect to a server ( node.js in my case ) and shut down the server.

### Debug Logs.

```plain
[0;31mE (1158240) TRANSPORT_WS: Error read data
E (1158241) WEBSOCKET_CLIENT: Error read data
E (1158241) WEBSOCKET_CLIENT: Error receive data
I (1158249) WEBSOCKET_CLIENT: Reconnect after 10000 ms
[HsH_Server] - Disconnected

[HsH_Server] - Connected

assert failed: 0x4008cb8e

Backtrace: 0x400833c1:0x3ffc5d70 0x40089c95:0x3ffc5d90 0x4008eb1d:0x3ffc5db0 0x4008cb8e:0x3ffc5df0 0x4008d43c:0x3ffc5e10 0x4008414b:0x3ffc5e30 0x4008eb4d:0x3ffc5e50 0x401f7d65:0x3ffc5e70 0x401f5b32:0x3ffc5ee0 0x40271997:0x3ffc5f30 0x401f5f17:0x3ffc5f50 0x40271997:0x3ffc5f70 0x401b53b5:0x3ffc5f90

ELF file SHA256: 39bb245d950cb35f
```

### More Information.

Decoded logs:

```log

Decoding stack results
0x400833c1: panic_abort at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\esp_system\panic.c line 408
0x40089c95: esp_system_abort at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\esp_system\esp_system.c line 137
0x4008eb1d: __assert_func at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\newlib\assert.c line 47
0x4008cb8e: tlsf_free at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\heap/heap_tlsf_block_functions.h line 90
0x4008d43c: multi_heap_free_impl at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\heap\multi_heap.c line 212
0x4008414b: heap_caps_free at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\heap\heap_caps.c line 382
0x4008eb4d: free at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\newlib\heap.c line 39
0x401f7d65: esp_vfs_select at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\vfs\vfs.c line 1080
0x401f5b32: base_poll_read at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\tcp_transport\transport_ssl.c line 150
0x40271997: esp_transport_poll_read at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\tcp_transport\transport.c line 156
0x401f5f17: ws_poll_read at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\tcp_transport\transport_ws.c line 528
0x40271997: esp_transport_poll_read at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\tcp_transport\transport.c line 156
0x401b53b5: esp_websocket_client_task at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\esp_websocket_client\esp_websocket_client.c line 689

```

gabsuren, 2024-04-17:
@hitecSmartHome I noticed that you are using ESP-IDF v4.4. Is there a specific reason for choosing this version?
The latest version of `esp_websocket_client` is located in [esp-protocols](https://github.com/espressif/esp-protocols/)
https://github.com/espressif/esp-protocols/tree/master/components/esp_websocket_client

And it's compatible with ESP-IDF version >= 5.0.
Could you please test with latest `esp_websocket_client` stable version, and in case of any issue let us know.

Regards,
Suren

gabsuren, 2024-04-17:
@hitecSmartHome `please note that v4.4 will be end of life in this July, and we will only have one more bugfix release v4.4.8 before its EOL`
To better assist in reproducing the issue, could you please provide additional details? Specifically, we need the source code for the WebSocket connection attempt and the sdkconfig. This information will help us understand and address the problem more effectively

hitecSmartHome, 2024-04-17:
Thank you for your help. The problem might be on my side. Will test it and report back tomorrow!

hitecSmartHome, 2024-04-18:
Yes, I confirmed it was a bad pointer free. Sorry for the issue ticket!