espressif/esp-idf #13624: ESP32-C6 Resets when starting wifi when waking from deep sleep

txf-, 2024-04-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.1

### Espressif SoC revision.

ESPC6-WROOM-1

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

Battery

### What is the expected behavior?

I expect the device to connect to wifi and then continue executing when waking from deep sleep, it should behave the same as when simply powering it on or resetting the device. 

### What is the actual behavior?

The actual behaviour is that when starting WiFi the device resets, whereupon after that reset the device connects normally as expected. 

### Steps to reproduce.

Run the following example code. This has been edited  from the power_save example :

Configure TX power to 10 dBm.

```
/* Power save Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

/*
   this example shows how to use power save mode
   set a router or a AP using the same SSID&PASSWORD as configuration of this example.
   start esp32 and when it connected to AP it will enter power save mode
*/
#include <string.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/event_groups.h"
#include "esp_wifi.h"
#include "esp_log.h"
#include "esp_event.h"
[... 679 lines cut ...]
I (3680) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (3688) wifi:set rx beacon pti, rx_bcn_pti: 10, bcn_timeout: 25000, mt_pti: 10, mt_time: 10000
I (3697) wifi:[ADDBA]TX addba request, tid:0, dialogtoken:1, bufsize:64, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (3707) wifi:[ADDBA]TX addba request, tid:7, dialogtoken:2, bufsize:64, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x20)
I (3718) wifi:[ADDBA]TX addba request, tid:5, dialogtoken:3, bufsize:64, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (3729) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (3735) wifi:[ADDBA]RX addba response, status:0, tid:0/tb:1(0xa1), bufsize:64, batimeout:0, txa_wnd:64
I (3744) wifi:[ADDBA]RX addba response, status:0, tid:7/tb:1(0xa1), bufsize:64, batimeout:0, txa_wnd:64
I (3753) wifi:[ADDBA]RX addba response, status:0, tid:5/tb:1(0xa1), bufsize:64, batimeout:0, txa_wnd:64
W (3771) wifi:<ba-add>idx:0, ifx:0, tid:0, TAHI:0x100999a, TALO:0x5d3247d8, (ssn:0, win:64, cur_ssn:0), CONF:0xc0000005
I (4763) esp_netif_handlers: sta ip: 192.168.0.232, mask: 255.255.255.0, gw: 192.168.0.1
I (4764) power_save: got ip: 192.168.0.232
I (10717) gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (10717) power_save: gpio level 0 before sleep
I (10720) power_save: pulse gpio wakeup source is ready
W (10726) power_save: Entering deep Sleep...
```

### More Information.

I also reduced the TX power to 40 (10dBm) both in code and via SDKCONFIG.  Ticking "the reduce txpower on brownout",  does not have any effect.  

Disabling the Brownout detector does allow the device to correctly identify the reason for reset `reset: Brownout Reset (software or hardware)`, as opposed with the brownout detector active `reset: Power-on Reset`.

Adding a delay (10s) before starting wifi also has no effect.

esp-lis, 2024-05-24:
@txf- Hi, I am try to reproduce it, but I can not reproduce it in my test environment.

In my test environment, I use the default power save example ($IDF_PATH/examples/wifi/power_save/) and simple replace the test code, then build it with default sdkconfig, download it to ESP32-C6-DevKitC-1 V1.2 board and run test, can not reproduce the `reset: Brownout Reset (software or hardware)` or `reset: Power-on Reset`.

Can you provide your sdkconfig, I will use it try to reproduce this problem, thanks.

txf-, 2024-05-24:
@esp-lis  attaching the sdkconfig to this post (adding *.txt extension)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/15431650/sdkconfig.txt)

esp-lis, 2024-05-27:
> @esp-lis attaching the sdkconfig to this post (adding *.txt extension) [sdkconfig.txt](https://github.com/espressif/esp-idf/files/15431650/sdkconfig.txt)

@txf- sorry, I use [sdkconfig.txt](https://github.com/espressif/esp-idf/files/15431650/sdkconfig.txt) to test, but can not reproduce it, I modified some options, such as SSID and partition table related, [this](https://github.com/espressif/esp-idf/files/15454270/sdkconfig_test.txt)
) is the configure file I used to compile.

Addition, for `reset: Brownout Reset (software or hardware)` or `reset: Power-on Reset`. I think it is related to the power supply, can you measure the power supply voltage and current of the board during runtime?

txf-, 2024-05-27:
> Think it is related to the power supply, can you measure the power supply voltage and current of the board during runtime?

@esp-lis , Having just tested that, You are probably correct. I cannot detect any droops below the configured brownout level (there are some), and I don't understand why it fails once then succeeds the second time.

But our regulator an (LDO) has a maximum current supply of 300 mA. I have since noticed in the C6 datasheet, it specifies in the recommended conditions (section 5.2) that the "Cumulative input current" is 0.5A. 

During normal usage the C6 will never pull that much. It appear that even with extensive decoupling caps our LDO is not enough to supply the WiFi radio during initialization.

esp-lis, 2024-05-27:
> > Think it is related to the power supply, can you measure the power supply voltage and current of the board during runtime?
> 
> @esp-lis , Having just tested that, You are probably correct. I cannot detect any droops below the configured brownout level (there are some), and I don't understand why it fails once then succeeds the second time.
> 
> But our regulator an (LDO) has a maximum current supply of 300 mA. I have since noticed in the C6 datasheet, it specifies in the recommended conditions (section 5.2) that the "Cumulative input current" is 0.5A.
> 
> During normal usage the C6 will never pull that much. It appear that even with extensive decoupling caps our LDO is not enough to supply the WiFi radio during initialization. With previous ESP32 devices that wasn't an issue, it may be with this one.

Refer to [esp32_datasheet_en.pdf](https://www.espressif.com.cn/sites/default/files/documentation/esp32_datasheet_en.pdf) for the requirements of supply voltage and current. The datasheet recommends a minimum external power supply current of 0.5 A. If stable and reliable system operation is required, I suggest using a power supply with a higher power rating.

closed 2024-11-04 by Sherry616