espressif/esp-idf #13626: Update BLE appearance values according to bluetooth-SIG

PocketPi, 2024-04-16:
### Is your feature request related to a problem?

I was trying to set the appearance value to 0x0841 (Audio Sink - Standalone Speaker) but got and error.

I dug into the code and saw that only a subset of appearance values are allowed.

### Describe the solution you'd like.

Either keep the list of allowed values up to date. Or make an interface for using all values

zhp0406, 2024-04-17:
[image]
[image]
[image]

zhp0406, 2024-04-17:
@PocketPi 
You're right, we will update it. For your normal usage, you can first add 0x0841.thanks for reporting the issue.

zhp0406, 2024-04-17:
Fix commit will coming soon:
master d1b47cf1ac67d8b6a0eb3c36de86cc226ecd567e
v5.2 ad67e5de59a4f556d272ce83edd3ce01048395fd
v5.1 b0532006c6ed809e3ec7e72d71a4e437d0484d05
v5.0 2af74e428a0f011b6ed211c53005716e2fb38c83
v4.4 3c9ecf56692b7881ac53a9b27e25cb9040e45550

PocketPi, 2024-04-17:
Super thanks,

Indastri, 2024-07-01:
Closing as the issue was solved.

Cheers!

closed 2024-07-01 by Indastri