espressif/esp-idf #13628: Network requests failing occasionally when connected to bridged network

swidget, 2024-04-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-3225-g5a40bb8746

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

External 5V

### What is the expected behavior?

I expected the device to remain connected to WiFi without issue when connected to a bridged network.

### What is the actual behavior?

The device will occasionally fail to perform network requests (without receiving a WiFi Disconnect event), once it begins to fail making requests, this will continue for about 15 minutes until a WiFi disconnect occurs, at which point it will perform a reconnect and begin to operate as normal.
This is occuring using the http_request example.
The issue also results in pinging the default gateway to fail, but UDP broadcast packets are still sent out to the network.
This issue does not occur on all networks, it does occur on one of my networks in bridged mode.

### Steps to reproduce.

Setup the example to connect to the router in bridged mode, then let the http_request to run until issue occurs.

### Debug Logs.

```plain
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xcb8
load:0x403cb700,len:0x2d9c
entry 0x403c8914
I (26) boot: ESP-IDF v5.3-dev-3225-g5a40bb8746-dirty 2nd stage bootloader
I (27) boot: compile time Apr 11 2024 12:44:51
I (28) boot: Multicore bootloader
I (32) boot: chip revision: v0.2
I (36) boot.esp32s3: Boot SPI Speed : 80MHz
I (41) boot.esp32s3: SPI Mode       : DIO
I (45) boot.esp32s3: SPI Flash Size : 2MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000
[... 869 lines cut ...]
connection: close

I (1744540) example: ... done reading from socket. Last read return=0 errno=128.
I (1744550) example: 10...
I (1745550) example: 9...
I (1746550) example: 8...
I (1747550) example: 7...
I (1748550) example: 6...
I (1749550) example: 5...
I (1750550) example: 4...
I (1751550) example: 3...
I (1752550) example: 2...
I (1753550) example: 1...
I (1754550) example: 0...
I (1755550) example: Starting again!
I (1755550) example: DNS lookup succeeded. IP=3.66.72.53
I (1755550) example: ... allocated socket
I (1755680) example: ... connected
I (1755680) example: ... socket send success
I (1755680) example: ... set socket receiving timeout success
HTTP/1.1 301 Moved Permanently
content-length: 0
location: https://dragons.icu:80/
connection: close
```

AxelLin, 2024-06-04:
Is there any progress about this issue?

I have similar issue.
Sometimes, the esp32s3 device joined an AP and got IP, but then the ntp sync fails:

I (8964) wifi:<ba-add>idx:0 (ifx:0, 18:a6:f7:ab:ca:66), tid:0, ssn:415, winSize:64
D (9705) wifi:eb is dhcp or dns sport = 31009, dport = 53
D (9961) wifi:eb is dhcp or dns sport = 31009, dport = 53
D (10961) wifi:eb is dhcp or dns sport = 31009, dport = 53
D (12961) wifi:eb is dhcp or dns sport = 31009, dport = 53
D (15961) wifi:eb is dhcp or dns sport = 21502, dport = 53
D (15963) wifi:eb is dhcp or dns sport = 21502, dport = 53
D (30972) wifi:eb is dhcp or dns sport = 53698, dport = 53
E (38961) Fail to sync NTP time

AxelLin, 2024-07-17:
@mhdong @MaxwellAlan 

Any feedback about this issue?
(The label shows "In Progress" but seems no response for a long period, is there anyone working on this issue?)

AxelLin, 2024-08-07:
I sometimes got below error after device boot and successfully joined an AP:

E (21970) esp-tls: [sock=48] select() timeout
E (21971) esp-tls: Failed to open new connection
E (21971) transport_base: Failed to open a new connection

ESP-YJM, 2024-08-07:
@AxelLin For the failure issue, it seems the device send TCP SYN and not receive SYN +ACK and reach the max retries. It maybe the device send SYN failure in wifi or wifi not receive SYN + ACK. You can simply disabled WIFI AMPDU TX and RX to test. Or you can capture the air packet to check whether the server response the SYN + ACK.

AxelLin, 2024-08-07:
@ESP-YJM 
Now I confirm I can reproduce this issue only when CONFIG_ESP_WIFI_AMPDU_TX_ENABLED=y.
Test with v5.2.2-457-g3883a17f54

ESP-YJM, 2024-08-07:
@AxelLin Could you please try the release/v5.2 branch?

AxelLin, 2024-08-07:
> @AxelLin Could you please try the release/v5.2 branch?

I'm using release/v5.2 branch
$ git describe 
v5.2.2-457-g3883a17f54

ESP-YJM, 2024-08-08:
@AxelLin Ok, I have asked WIFI colleagues to check it.

AxelLin, 2024-08-08:
> @AxelLin Ok, I have asked WIFI colleagues to check it.

FYI, below is the log with verbose build:

lwip_connect(48, addr=125.228.203.199 port=8121)
V (12516) wifi:tx process: alloc static tx buf, pb=0x3c120378 l=58 hdr=0 data=58
lwip_connect(48) failed, err=-5
lwip_select(49, 0x0, 0x3fca8af0, 0x0, tvsec=15 tvusec=0)
V (13170) wifi:tx process: alloc static tx buf, pb=0x3c1203dc l=42 hdr=0 data=42
V (13765) wifi:tx process: alloc static tx buf, pb=0x3c120378 l=58 hdr=0 data=58
V (15265) wifi:tx process: alloc static tx buf, pb=0x3c120378 l=58 hdr=0 data=58
V (16765) wifi:tx process: alloc static tx buf, pb=0x3c120378 l=58 hdr=0 data=58
V (18265) wifi:tx process: alloc static tx buf, pb=0x3c120378 l=58 hdr=0 data=58
V (19765) wifi:tx process: alloc static tx buf, pb=0x3c120378 l=58 hdr=0 data=58
V (21265) wifi:tx process: alloc static tx buf, pb=0x3c120378 l=58 hdr=0 data=58
V (22765) wifi:tx process: alloc static tx buf, pb=0x3c120378 l=58 hdr=0 data=58
V (24265) wifi:tx process: alloc static tx buf, pb=0x3c120378 l=58 hdr=0 data=58
V (25765) wifi:tx process: alloc static tx buf, pb=0x3c120378 l=58 hdr=0 data=58
V (27265) wifi:tx process: alloc static tx buf, pb=0x3c120378 l=58 hdr=0 data=58
lwip_select: timeout expired
E (27554) esp-tls: [sock=48] select() timeout
lwip_close(48)
E (27555) esp-tls: Failed to open new connection
E (27557) transport_base: Failed to open a new connection
E (27563) HTTP_CLIENT: Connection failed, sock < 0

AxelLin, 2024-09-24:
> @AxelLin Ok, I have asked WIFI colleagues to check it.

Any update?

AxelLin, 2024-11-19:
> @AxelLin Ok, I have asked WIFI colleagues to check it.

But there is no update at all even after waiting for 3 Months.

closed 2025-07-14 by Sherry616