espressif/esp-idf #13632: ESP32C3: NVS/SPIFFS not working when flah encryption (developer mode) is enabled

zwieblum, 2024-04-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

2.0.15

### Espressif SoC revision.

ESP32-C3 (QFN32) (revision v0.4)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

XIAO ESP32C3

### Power Supply used.

USB

### What is the expected behavior?

SPIFFS: this should succeed:
SPIFFS.begin(FORMAT_SPIFFS_IF_FAILED); 

NVS Preferences: 
prefs.begin("xxx", false);
prefs.putInt("integer_value", 42);
prefs.getInt("integer_value", -1));
prefs.begin("xxx", false); --- should return 42

### What is the actual behavior?

SPIFFS:
SPIFFS.begin(FORMAT_SPIFFS_IF_FAILED);
W (1022) SPIFFS: mount failed, -10025. formatting...
E (2496) SPIFFS: mount failed, -10025

NVS Preferences:
Each write succeeds, but each read returns 0 length or 0xff bytes.

### Steps to reproduce.

Enable flash encryption (developer mode).
compile and upload SPIFFS and Preferences examples from IDF examples.

### More Information.

If the SPIFFS partition is marked as "encrypted", then LITTLEFS works without a problem on the SPIFFS partition.
NVS/Preferences and SPIFFS work as expected, when flash is not encrypted.

igrr, 2024-04-17:
Regarding SPIFFS, it doesn't support Flash Encryption.

NVS should work with Flash Encryption enabled, though. I see that you are using the Preferences API from https://github.com/espressif/arduino-esp32, could you please describe in more details how exactly are you using Arduino? Are you using it as an IDF component? If yes, could you please attach your sdkconfig file?

zwieblum, 2024-04-17:
Here is the partitiontable and the sdkconfig I use. NVS partition is not encrypted - otherwise I get an error. 

The SPIFFS partition is encrypted for use as LITTLEFS, but unencrypted for use as SPIFFS  - otherwise it would give an error.

I use arduino for easy development for different platforms, it's quite handy. I my setup I have (mis)used the hello_world example to create an a bootloader for devolopment mode and then I upload the arduino program with idf.py like so:

cd hello_encrypted_world
arduino-cli -v --output-dir=/tmp/build compile -b esp32:esp32:XIAO_ESP32C3 ..
cp /tmp/build/*.ino.bin build/hello_world.bin
idf.py -p /dev/ttyACM0 encrypted-app-flash monitor

[partitiontable.csv](https://github.com/espressif/esp-idf/files/15009154/partitiontable.csv)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/15009162/sdkconfig.txt)

igrr, 2024-04-17:
I see, in this case I'm afraid the Flash operations won't work correctly. Flash Encryption needs to be enabled not only in the bootloader, but in the application as well. In this case, the application compiled by Arduino IDE doesn't have Flash Encryption enabled.

I think the only option to have an Arduino application with Flash Encryption enabled is to set up Arduino as an IDF component, and transfer your Arduino code into an IDF-based project.

zwieblum, 2024-04-17:
Ok ... but shouldn't that affect LITTLEFS, too? But LITTLEFS works ...

igrr, 2024-04-17:
Looking at this again, actually `nvs_flash_init` should still work when hardware Flash Encryption is enabled but CONFIG_SECURE_FLASH_ENC_ENABLED is not enabled. The result will be an unencrypted NVS partition. It is possible that the behavior has changed since IDF release v4.4.x (which is used in Arduino-esp32 2.0.x). We'll need to reproduce the issue to find the cause.

zwieblum, 2024-07-09:
Any news on this?

gw8484, 2025-02-27:
Any update?  Looking at the sdkconfig.txt, CONFIG_SECURE_FLASH_ENC_ENABLED is enabled.