espressif/esp-idf #13638: Remove a managed component

anujdeshpande, 2024-04-18:
### Is your feature request related to a problem?

Once I add a managed component, I can't find a method to remove it from my app.

### Describe the solution you'd like.

idf.py remove-dependency <name>

### Describe alternatives you've considered.

Manually deleting - but it comes back when I do a build.

### Additional context.

ESP-IDF v5.1.3

igrr, 2024-04-18:
Hi @anujdeshpande!
Probably you need to check which `idf_component.yml` file contains this component, and remove it from there. If you used `idf.py add-dependency` command without specifying the component where the dependency is added, the dependency would be added to the `main` component. So you need to check the `main/idf_component.yml` file.

anujdeshpande, 2024-04-18:
Yeah I figured that part with grep - but it would be simpler to remove a dependency using the same way I added it. I don't think this is a bug - so marked it as a feature request.