espressif/esp-idf #13640: Unable to use OPENTHREAD_CONFIG_TCP_ENABLE methods like oTcpCommitReceive

mamacker, 2024-04-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1-dirty

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I'm unable to turn on the compiler flag OPENTHREAD_CONFIG_TCP_ENABLE - which prevents me from writting a C program for simple TCP sockets.  Namely - the method otTcpCommitReceive is not available.

### What is the actual behavior?

A way to turn on the flag OPENTHREAD_CONFIG_TCP_ENABLE.

### Steps to reproduce.

Run menu config from a copy of the CLI - the option isn't in the configurations.

### Build or installation Logs.

```plain
*  Executing task: ninja  

[1/9] Performing build step for 'bootloader'
[1/1] cd /home/mamacker/turbineone/Mesh802.15.4/thread/gateway/ot_cli/build/bootloader/esp-idf/esptool_py && /home/mamacker/.espressif/python_env/idf5.2_py3.8_env/bin/python /home/mamacker/esp/v5.2.1/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /home/mamacker/turbineone/Mesh802.15.4/thread/gateway/ot_cli/build/bootloader/bootloader.bin
Bootloader binary size 0x5410 bytes. 0x2bf0 bytes (34%) free.
[2/7] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/tcp_server.c.obj
../main/tcp_server.c: In function 'ot_tcp_loop':
../main/tcp_server.c:141:44: warning: assignment to 'otTcpReceiveAvailable' {aka 'void (*)(otTcpEndpoint *, unsigned int,  _Bool,  unsigned int)'} from incompatible pointer type 'void (*)(otTcpEndpoint *, unsigned int *, _Bool,  unsigned int)' [-Wincompatible-pointer-types]
  141 |     endpointArgs.mReceiveAvailableCallback = HandleTcpReceiveAvailableCallback;
      |                                            ^
../main/tcp_server.c:142:36: warning: assignment to 'otTcpSendDone' {aka 'void (*)(otTcpEndpoint *, otLinkedBuffer *)'} from incompatible pointer type 'void (*)(otTcpEndpoint *, otError)' [-Wincompatible-pointer-types]
  142 |     endpointArgs.mSendDoneCallback = HandleTcpSendDoneCallback;
      |                                    ^
../main/tcp_server.c:156:38: warning: assignment to 'otTcpAcceptDone' {aka 'void (*)(otTcpListener *, otTcpEndpoint *, const otSockAddr *)'} from incompatible pointer type 'void (*)(otTcpListener *, otTcpEndpoint *, otSockAddr *)' [-Wincompatible-pointer-types]
  156 |     listenerArgs.mAcceptDoneCallback = HandleTcpAcceptDoneCallback;
      |                                      ^
../main/tcp_server.c:136:16: warning: unused variable 'listenSockAddr' [-Wunused-variable]
  136 |     otSockAddr listenSockAddr;
      |                ^~~~~~~~~~~~~~
[5/7] Linking CXX executable esp_ot_cli.elf
FAILED: esp_ot_cli.elf 
: && /home/mamacker/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/riscv32-esp-elf-g++  -march=rv32imac_zicsr_zifencei  -nostartfiles -march=rv32imac_zicsr_zifencei --specs=nosys.specs    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32C6=0 -Wl,--Map=/home/mamacker/turbineone/Mesh802.15.4/thread/gateway/ot_cli/build/esp_ot_cli.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32c6.peripherals.ld -T esp32c6.rom.ld -T esp32c6.rom.api.ld -T esp32c6.rom.rvfp.ld -T esp32c6.rom.newlib.ld -T esp32c6.rom.version.ld -T esp32c6.rom.phy.ld -T esp32c6.rom.coexist.ld -T esp32c6.rom.net80211.ld -T esp32c6.rom.pp.ld -T esp32c6.rom.wdt.ld -T esp32c6.rom.newlib-normal.ld -T esp32c6.rom.heap.ld -T memory.ld -T sections.ld CMakeFiles/esp_ot_cli.elf.dir/project_elf_src_esp32c6.c.obj  -o esp_ot_cli.elf -L/home/mamacker/esp/v5.2.1/esp-idf/components/soc/esp32c6/ld   -L/home/mamacker/esp/v5.2.1/esp-idf/components/esp_rom/esp32c6/ld   -L/home/mamacker/turbineone/Mesh802.15.4/thread/gateway/ot_cli/build/esp-idf/esp_system/ld   -L/home/mamacker/esp/v5.2.1/esp-idf/components/esp_phy/lib/esp32c6   -L/home/mamacker/esp/v5.2.1/esp-idf/components/esp_coex/lib/esp32c6   -L/home/mamacker/esp/v5.2.1/esp-idf/components/esp_wifi/lib/esp32c6 esp-idf/riscv/libriscv.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_bootloader_format/libesp_bootloader_format.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/app_trace/libapp_trace.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/console/libconsole.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/fatfs/libfatfs.a  esp-idf/ieee802154/libieee802154.a  esp-idf/json/libjson.a  esp-idf/mqtt/libmqtt.a  esp-idf/nvs_sec_provider/libnvs_sec_provider.a  esp-idf/openthread/libopenthread.a  esp-idf/spiffs/libspiffs.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/iperf/libiperf.a  esp-idf/espressif__esp_ot_cli_extension/libespressif__esp_ot_cli_extension.a  esp-idf/main/libmain.a  esp-idf/app_trace/libapp_trace.a  esp-idf/app_trace/libapp_trace.a  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/fatfs/libfatfs.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/mqtt/libmqtt.a  esp-idf/nvs_sec_provider/libnvs_sec_provider.a  -u nvs_sec_provider_include_impl  esp-idf/spiffs/libspiffs.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/protocomm/libprotocomm.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/json/libjson.a  esp-idf/espressif__esp_ot_cli_extension/libespressif__esp_ot_cli_extension.a  esp-idf/iperf/libiperf.a  esp-idf/riscv/libriscv.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_bootloader_format/libesp_bootloader_format.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libes [... cut ...] ): undefined reference to `otTcpListenerInitialize'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

 *  The terminal process "/usr/bin/bash '-c', 'ninja '" terminated with exit code: 1.
```

### More Information.

Thank you ahead of time - I think I'm missing something basic... but your forums are down, and the SDK documentation doesn't seem to cover TCP much.  The CLI examples online seem to show tcp connection ability - but I can't use the functions myself.

mamacker, 2024-04-18:
Here is the super simple braindead tcp example I'm attempting to build.  

#include "utils.h"
#include <openthread/tcp.h>
#include <openthread/tcp_ext.h>
#include <openthread/nat64.h>

static const char *TAG = "t1_tcp_";

// static TaskHandle_t s_service_task;
#define TCP_PORT 20600

bool mInitialized;
bool mEndpointConnected;
bool mEndpointConnectedFastOpen;
bool mSendBusy;
bool mUseCircularSendBuffer;

otTcpEndpointInitializeArgs endpointArgs;
otTcpEndpoint mEndpoint;

otTcpCircularSendBuffer mSendBuffer;
otLinkedBuffer mSendLink;
uint8_t mSendBufferBytes[2048];
uint8_t mReceiveBufferBytes[2048];

size_t receiveBufferSize = sizeof(mReceiveBufferBytes);

void esp_openthread_tcp_init(void)
{
    // otCliInit(esp_openthread_get_instance(), cli_output_callback, NULL);
}

// Step zero - read to accept.
otTcpIncomingConnectionAction HandleTcpAcceptReadyCallback(otTcpListener *aListener, const otSockAddr *aPeer, otTcpEndpoint **aAcceptInto)
{
    ESP_LOGE(TAG, "TCP connection ready!");
    OT_UNUSED_VARIABLE(aListener);

    otTcpIncomingConnectionAction action;
    if (mEndpointConnected)
    {
        return OT_TCP_INCOMING_CONNECTION_ACTION_DEFER;
    }

    *aAcceptInto = &mEndpoint;
    action = OT_TCP_INCOMING_CONNECTION_ACTION_ACCEPT;
    return action;
}

// Step one - accept has occurred.
void HandleTcpAcceptDoneCallback(otTcpListener *aListener, otTcpEndpoint *aEndpoint, otSockAddr *aPeer)
{
    ESP_LOGE(TAG, "TCP connection!");
    uint16_t aSize = 64;
    char aBuffer[64];

    otIp6AddressToString(&(aPeer->mAddress), aBuffer, aSize);
    ESP_LOGE(TAG, "Accept from: %s", aBuffer);

    mEndpointConnected = true;
}

void HandleTcpEstablishedCallback(otTcpEndpoint *aEndpoint)
{
    ESP_LOGE(TAG, "TCP connection established!");
    // Nothing else to do.
}

void HandleTcpReceiveAvailableCallback(otTcpEndpoint *aEndpoint, unsigned int *aReceiveBuffer, bool aEndOfStream, unsigned int aReceiveBufferLength)
{
    ESP_LOGE(TAG, "TCP connection received!");
    OT_UNUSED_VARIABLE(aReceiveBufferLength);

    /* If we get data before the handshake completes, then this is a TFO connection. */
    if (!mEndpointConnected)
    {
        mEndpointConnected = true;
        mEndpointConnectedFastOpen = true;
    }

    const otLinkedBuffer *data;
    size_t totalReceived = 0;

    data = aEndpoint->mReceiveLinks;

    for (; data != NULL; data = data->mNext)
    {
        ESP_LOGE(TAG, "TCP: Received %u bytes: %.*s", (unsigned int)data->mLength,
                 data->mLength, (char *)data->mData);
        totalReceived += data->mLength;
    }

    otTcpCommitReceive(aEndpoint, totalReceived, 0);

    if (aEndOfStream)
    {
        ESP_LOGE(TAG, "TCP: Reached end of stream");
    }
}

void HandleTcpSendDoneCallback(otTcpEndpoint *aEndpoint, otError aError)
{
    ESP_LOGE(TAG, "TCP connection sent!");
}

void HandleTcpClosedCallback(otTcpEndpoint *aEndpoint, otError aError)
{
    ESP_LOGE(TAG, "TCP connection closed!");
}

void HandleTcpDisconnected(otTcpEndpoint *aEndpoint, otTcpDisconnectedReason aReason)
{
    static const char *const kReasonStrings[] = {
        "Disconnected",            // (0) OT_TCP_DISCONNECTED_REASON_NORMAL
        "Connection refused",      // (1) OT_TCP_DISCONNECTED_REASON_REFUSED
        "Connection reset",        // (2) OT_TCP_DISCONNECTED_REASON_RESET
        "Entered TIME-WAIT state", // (3) OT_TCP_DISCONNECTED_REASON_TIME_WAIT
        "Connection timed out",    // (4) OT_TCP_DISCONNECTED_REASON_TIMED_OUT
    };

    OT_UNUSED_VARIABLE(aEndpoint);

    ESP_LOGE(TAG, "Disconnected: %s", kReasonStrings[aReason]);

    mEndpointConnected = false;
    mEndpointConnectedFastOpen = false;
    mSendBusy = false;
}

static void ot_tcp_loop(void)
{
    ESP_LOGE(TAG, "Setup UDP!");
    otInstance *instance = esp_openthread_get_instance();
    spewIPAddresses(instance);
    // Create the address struct.
    otSockAddr listenSockAddr;

    otTcpEndpointInitializeArgs endpointArgs;
    memset(&endpointArgs, 0, sizeof(endpointArgs));
    endpointArgs.mEstablishedCallback = HandleTcpEstablishedCallback;
    endpointArgs.mReceiveAvailableCallback = HandleTcpReceiveAvailableCallback;
    endpointArgs.mSendDoneCallback = HandleTcpSendDoneCallback;
    endpointArgs.mDisconnectedCallback = HandleTcpDisconnected;
    memset(&mReceiveBufferBytes, 0, sizeof(mReceiveBufferBytes));
    endpointArgs.mReceiveBuffer = mReceiveBufferBytes;
    endpointArgs.mReceiveBufferSize = receiveBufferSize;
    otTcpEndpointInitialize(instance, &mEndpoint, &endpointArgs);

    otTcpListener listener;
    otTcpListenerInitializeArgs listenerArgs;

    memset(&listener, 0, sizeof(listener));
    memset(&listenerArgs, 0, sizeof(listenerArgs));

    listenerArgs.mAcceptReadyCallback = HandleTcpAcceptReadyCallback;
    listenerArgs.mAcceptDoneCallback = HandleTcpAcceptDoneCallback;
    otTcpListenerInitialize(instance, &listener, &listenerArgs);
}

void esp_openthread_tcp_create_task()
{
    // xTaskCreate(ot_service_loop, "ot_service", 4096, xTaskGetCurrentTaskHandle(), 4, &s_service_task);
    ot_tcp_loop();
    return;
}

mamacker, 2024-04-18:
Oh... I may have found a pile of clues: https://github.com/espressif/esp-thread-br/blob/main/components/esp_ot_cli_extension/src/esp_ot_tcp_socket.c

zwx1995esp, 2024-04-19:
Hi, @mamacker based ESP-IDF code, we support the LWIP tcp socket APIs for using TCP transmission upon Thread. Yeah, you can refer to this demo code and feel free to let me know if you have some other questions.

FYI, we also support the tool `iperf`, you can test a lot with this tool. https://github.com/espressif/esp-thread-br/blob/main/components/esp_ot_cli_extension/README.md#iperf

mamacker, 2024-04-20:
Got TCP working.  Closing on my end.

closed 2024-04-20 by mamacker