espressif/esp-idf #13641: esp_http_client_read() blocked posting to default event loop

jonsmirl, 2024-04-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

Chip is ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

In http streaming mode esp_http_client_read() should be able to complete my request without blocking.

### What is the actual behavior?

esp_http_client_read() is blocking after it sends seven events to the default event loop.
https://github.com/espressif/esp-idf/blob/master/components/esp_http_client/esp_http_client.c#L325

### Steps to reproduce.

~~~
static int http_fetch(lua_State *L)
{
  esp_err_t err;
  size_t len;
  
  esp_http_client_config_t config = {
      .url = luaL_checklstring( L, 1, &len ),
      .crt_bundle_attach = esp_crt_bundle_attach,
  };
  esp_http_client_handle_t client = esp_http_client_init(&config);
  esp_http_client_set_header(client, "Connection", "close");
  if ((err = esp_http_client_open(client, 0)) != ESP_OK) {
    ESP_LOGE(TAG, "Failed to open HTTP connection: %s", esp_err_to_name(err));
    return 0;
  }
  int content_length =  esp_http_client_fetch_headers(client);
  char *udata = lua_newuserdata(L, content_length + 1);
  if (udata == NULL) {
    ESP_LOGE(TAG, "Cannot malloc http receive buffer");
    return 0;
  }
  ESP_LOGI(TAG, "Content length %d", content_length);
  int read_len, total = 0;
  do {
    read_len = esp_http_client_read(client, &udata[total], 512);  <=== this is blocking
    total += read_len;
    ESP_LOGI(TAG, "read_len %d", read_len);
    if (read_len <= 0) {
        ESP_LOGE(TAG, "Error read data");
    }
    vTaskDelay( 100 );
  } while (total < content_length);
  udata[total] = 0;
  lua_pushlstring(L, udata, total + 1);
  ESP_LOGD(TAG, "total = %d", total);

  ESP_LOGI(TAG, "HTTP Stream reader Status = %d, content_length = %"PRIu64,
                  esp_http_client_get_status_code(client),
                  esp_http_client_get_content_length(client));
  esp_http_client_close(client);
  esp_http_client_cleanup(client);

  return 1;
}
~~~

### Debug Logs.

```plain
D (49271) HTTP_CLIENT: Write header[4]: GET /data/2.5/forecast?lat=42.321384&lon=-71.283516&APPID=46a84cccf7888660facf7c353f0b96cb&units=metric HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: api.openweathermap.org
Connection: close
Content-Length: 0

D (49421) HTTP_CLIENT: on_message_begin
D (49423) HTTP_CLIENT: HEADER=Server:openresty
D (49424) HTTP_CLIENT: HEADER=Date:Thu, 18 Apr 2024 18:41:38 GMT
D (49424) HTTP_CLIENT: HEADER=Content-Type:application/json; charset=utf-8
D (49426) HTTP_CLIENT: HEADER=Content-Length:15827
D (49426) HTTP_CLIENT: HEADER=Connection:close
D (49427) HTTP_CLIENT: HEADER=X-Cache-Key:/data/2.5/forecast?APPID=46a84cccf7888660facf7c353f0b96cb&lat=42.32&lon=-71.28&units=metric
D (49427) HTTP_CLIENT: HEADER=Access-Control-Allow-Origin:*
D (49430) HTTP_CLIENT: HEADER=Access-Control-Allow-Credentials:true
D (49432) HTTP_CLIENT: HEADER=Access-Control-Allow-Methods:GET, POST
D (49432) HTTP_CLIENT: http_on_headers_complete, status=200, offset=383, nread=383
D (49433) HTTP_CLIENT: http_on_body1 129
D (49433) HTTP_CLIENT: Body received in fetch header state, 0x3fcee803, 129
D (49435) HTTP_CLIENT: http_on_body2 129
D (49435) HTTP_CLIENT: http_on_body3 129
D (49435) HTTP_CLIENT: http_on_body4 129
D (49437) HTTP_CLIENT: http_on_body5 129
D (49438) HTTP_CLIENT: content_length = 15827
I (49447) matterLua: Content length 15827
D (49447) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=15827
D (49451) HTTP_CLIENT: 1 need_read=383, byte_to_read=383, rlen=-1, ridx=129
D (49452) HTTP_CLIENT: 2 need_read=383, byte_to_read=383, rlen=383, ridx=129
D (49452) HTTP_CLIENT: http_on_body1 383
D (49453) HTTP_CLIENT: http_on_body2 383
D (49453) HTTP_CLIENT: http_on_body3 383
D (49453) HTTP_CLIENT: http_on_body4 383
D (49455) HTTP_CLIENT: http_on_body5 383
I (49455) matterLua: read_len 512
D (49555) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=15827
D (49556) HTTP_CLIENT: 1 need_read=512, byte_to_read=512, rlen=-1, ridx=0
D (49556) HTTP_CLIENT: 2 need_read=512, byte_to_read=512, rlen=512, ridx=0
D (49557) HTTP_CLIENT: http_on_body1 512
D (49557) HTTP_CLIENT: http_on_body2 512
D (49559) HTTP_CLIENT: http_on_body3 512
D (49560) HTTP_CLIENT: http_on_body4 512
D (49561) HTTP_CLIENT: http_on_body5 512
I (49561) matterLua: read_len 512
D (49665) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=15827
D (49665) HTTP_CLIENT: 1 need_read=512, byte_to_read=512, rlen=-1, ridx=0
D (49668) HTTP_CLIENT: 2 need_read=512, byte_to_read=512, rlen=512, ridx=0
D (49668) HTTP_CLIENT: http_on_body1 512
D (49668) HTTP_CLIENT: http_on_body2 512
D (49669) HTTP_CLIENT: http_on_body3 512
D (49670) HTTP_CLIENT: http_on_body4 512
D (49672) HTTP_CLIENT: http_on_body5 512
I (49672) matterLua: read_len 512
D (49773) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=15827
D (49774) HTTP_CLIENT: 1 need_read=512, byte_to_read=512, rlen=-1, ridx=0
D (49774) HTTP_CLIENT: 2 need_read=512, byte_to_read=512, rlen=512, ridx=0
D (49776) HTTP_CLIENT: http_on_body1 512
D (49776) HTTP_CLIENT: http_on_body2 512
D (49776) HTTP_CLIENT: http_on_body3 512
D (49777) HTTP_CLIENT: http_on_body4 512
D (49777) HTTP_CLIENT: http_on_body5 512
I (49777) matterLua: read_len 512
D (49880) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=15827
D (49880) HTTP_CLIENT: 1 need_read=512, byte_to_read=512, rlen=-1, ridx=0
D (49881) HTTP_CLIENT: 2 need_read=512, byte_to_read=512, rlen=512, ridx=0
D (49881) HTTP_CLIENT: http_on_body1 512
D (49884) HTTP_CLIENT: http_on_body2 512
D (49884) HTTP_CLIENT: http_on_body3 512
D (49884) HTTP_CLIENT: http_on_body4 512
D (49886) HTTP_CLIENT: http_on_body5 512
I (49886) matterLua: read_len 512
D (49987) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=15827
D (49988) HTTP_CLIENT: 1 need_read=512, byte_to_read=512, rlen=-1, ridx=0
D (49988) HTTP_CLIENT: 2 need_read=512, byte_to_read=512, rlen=512, ridx=0
D (49989) HTTP_CLIENT: http_on_body1 512
D (49989) HTTP_CLIENT: http_on_body2 512
D (49990) HTTP_CLIENT: http_on_body3 512
D (49990) HTTP_CLIENT: http_on_body4 512
D (49991) HTTP_CLIENT: http_on_body5 512
I (49991) matterLua: read_len 512
D (50091) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=15827
D (50091) HTTP_CLIENT: 1 need_read=512, byte_to_read=512, rlen=-1, ridx=0
D (50094) HTTP_CLIENT: 2 need_read=512, byte_to_read=512, rlen=512, ridx=0
D (50094) HTTP_CLIENT: http_on_body1 512
D (50094) HTTP_CLIENT: http_on_body2 512
D (50095) HTTP_CLIENT: http_on_body3 512
D (50096) HTTP_CLIENT: http_on_body4 512  -- inserted one line before
https://github.com/espressif/esp-idf/blob/master/components/esp_http_client/esp_http_client.c#L325
does not print body5 which is inserted one line later. It has blocked in the call.
```

### More Information.

While I prefer to do a single read, I broke it down into a loop doing 512b reads and inserted vTaskDelay(100) to let other tasks run and empty the default event loop queue, but it still blocks.  I am not using the default event queue, but esp_matter uses it.

jonsmirl, 2024-04-19:
If I comment out the event posting to the default event loop
https://github.com/espressif/esp-idf/blob/master/components/esp_http_client/esp_http_client.c#L325
The code works as expected.

hmalpani, 2024-05-08:
Hello @jonsmirl 
I have created a patch to make the timeout configurable in the menuconfig. Can you please check and let me know if it helps in resolving your issue?

[IDFGH-12649.txt](https://github.com/espressif/esp-idf/files/15244644/IDFGH-12649.txt)

AxelLin, 2024-05-08:
@hmalpani 

Do you ever think why esp_event_post() is blocked?
Is the event queue size too small or some process taking too much time in handling the event?

AxelLin, 2024-05-10:
> Hello @jonsmirl I have created a patch to make the timeout configurable in the menuconfig. Can you please check and let me know if it helps in resolving your issue?
> 
> [IDFGH-12649.txt](https://github.com/espressif/esp-idf/files/15244644/IDFGH-12649.txt)

IMHO, this is not a good change especially you **changed the default behavior** from portMAX_DELAY to delay 2000ms.

jonsmirl, 2024-05-10:
The default event loop is being made in a different Espressif component, esp-matter. I am not using events from the http server.
https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_core.cpp#L1043
So maybe esp-matter isn't emptying the queue?

closed 2024-05-16 by espressif-bot

AxelLin, 2024-05-16:
@hmalpani 

You not only adding new config but also changed the default from portMAX_DELAY to 2 sec.
So now you changed the default behavior, I believe this commit will cause more issues.