espressif/esp-idf #13642: Invalid task in heap_caps_get_per_task_info()

jonsmirl, 2024-04-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

Chip is ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I should be able to retrieve heap information from all task.

### What is the actual behavior?

There is an invalid task in the list returned from heap_caps_get_per_task_info()

### Steps to reproduce.

~~~
#define MAX_TASK_NUM 20                         // Max number of per tasks info that it can store
#define MAX_BLOCK_NUM 20                        // Max number of per block info that it can store

static size_t s_prepopulated_num = 0;
static heap_task_totals_t s_totals_arr[MAX_TASK_NUM];
static heap_task_block_t s_block_arr[MAX_BLOCK_NUM];

static void esp_dump_per_task_heap_info(void)
{
    heap_task_info_params_t heap_info = {0};
    heap_info.caps[0] = MALLOC_CAP_INTERNAL;        // Gets heap with CAP_INTERNAL capabilities
    heap_info.mask[0] = MALLOC_CAP_INTERNAL;
    heap_info.caps[1] = MALLOC_CAP_SPIRAM;       // Gets heap info with CAP_SPIRAM capabilities
    heap_info.mask[1] = MALLOC_CAP_SPIRAM;
    heap_info.tasks = NULL;                     // Passing NULL captures heap info for all tasks
    heap_info.num_tasks = 0;
    heap_info.totals = s_totals_arr;            // Gets task wise allocation details
    heap_info.num_totals = &s_prepopulated_num;
    heap_info.max_totals = MAX_TASK_NUM;        // Maximum length of "s_totals_arr"
    heap_info.blocks = s_block_arr;             // Gets block wise allocation details. For each block, gets owner task, address and size
    heap_info.max_blocks = MAX_BLOCK_NUM;       // Maximum length of "s_block_arr"

    heap_caps_get_per_task_info(&heap_info);

    for (int i = 0 ; i < *heap_info.num_totals; i++) {
        if (heap_info.totals[i].task != (TaskHandle_t)0x580e) {
        printf("Task: %s -> MALLOC_CAP_INTERNAL: %d MALLOC_CAP_SPIRAM: %d Highwater %d\n",
                heap_info.totals[i].task ? pcTaskGetName(heap_info.totals[i].task) : "Pre-Scheduler allocs" ,
                heap_info.totals[i].size[0],    // Heap size with CAP_INTERNAL capabilities
                heap_info.totals[i].size[1],   // Heap size with CAP_SPIRAM capabilities
                heap_info.totals[i].task ? uxTaskGetStackHighWaterMark(heap_info.totals[i].task) : 0);
        }
    }

    printf("\n\n");
}
~~~

### Debug Logs.

```plain
When it works I see this:
~~~
Task: Pre-Scheduler allocs -> MALLOC_CAP_INTERNAL: 3560 MALLOC_CAP_SPIRAM: 392 Highwater 0
Task: main -> MALLOC_CAP_INTERNAL: 81096 MALLOC_CAP_SPIRAM: 73760 Highwater 5112
Task: lua -> MALLOC_CAP_INTERNAL: 68760 MALLOC_CAP_SPIRAM: 118036 Highwater 35420
Task: tiT -> MALLOC_CAP_INTERNAL: 448 MALLOC_CAP_SPIRAM: 404 Highwater 252
Task: nimble_host -> MALLOC_CAP_INTERNAL: 36 MALLOC_CAP_SPIRAM: 448 Highwater 3704
Task: wifi -> MALLOC_CAP_INTERNAL: 18260 MALLOC_CAP_SPIRAM: 60212 Highwater 2864
Task: CHIP -> MALLOC_CAP_INTERNAL: 53484 MALLOC_CAP_SPIRAM: 36888 Highwater 3240
Task: sys_evt -> MALLOC_CAP_INTERNAL: 212 MALLOC_CAP_SPIRAM: 104 Highwater 1760
Task: httpd -> MALLOC_CAP_INTERNAL: 112 MALLOC_CAP_SPIRAM: 184 Highwater 524
Task: ipc0 -> MALLOC_CAP_INTERNAL: 21964 MALLOC_CAP_SPIRAM: 0 Highwater 732
Task: ipc1 -> MALLOC_CAP_INTERNAL: 32 MALLOC_CAP_SPIRAM: 0 Highwater 720
Task: btController -> MALLOC_CAP_INTERNAL: 7492 MALLOC_CAP_SPIRAM: 0 Highwater 1820
Task: esp_timer -> MALLOC_CAP_INTERNAL: 112 MALLOC_CAP_SPIRAM: 0 Highwater 2988
~~~
If I remove the " if (heap_info.totals[i].task != (TaskHandle_t)0x580e) {" it will fault on that task.  0x580e is not a valid task handle, why is it in the list?
```

### More Information.

This problem occurred when I switched from IDF 5.1 to 5.2.

jonsmirl, 2024-04-18:
I made it print the invalid task handle, it is in the middle of the list.
~~~
Task: Pre-Scheduler allocs 0x0 -> MALLOC_CAP_INTERNAL: 3560 MALLOC_CAP_SPIRAM: 392 Highwater 0
Task: main 0x3fcb2dac -> MALLOC_CAP_INTERNAL: 81096 MALLOC_CAP_SPIRAM: 73760 Highwater 4872
Task: lua 0x3fca0ad4 -> MALLOC_CAP_INTERNAL: 68780 MALLOC_CAP_SPIRAM: 119712 Highwater 35420
Task: tiT 0x3fcc3cd4 -> MALLOC_CAP_INTERNAL: 448 MALLOC_CAP_SPIRAM: 412 Highwater 368
Task: wifi 0x3fcc6980 -> MALLOC_CAP_INTERNAL: 18260 MALLOC_CAP_SPIRAM: 60140 Highwater 2820
Task: CHIP 0x3fcd3674 -> MALLOC_CAP_INTERNAL: 53380 MALLOC_CAP_SPIRAM: 36888 Highwater 3328
Task: nimble_host 0x3fcdcf14 -> MALLOC_CAP_INTERNAL: 36 MALLOC_CAP_SPIRAM: 448 Highwater 3704
Task: sys_evt 0x3fcc2ab0 -> MALLOC_CAP_INTERNAL: 200 MALLOC_CAP_SPIRAM: 144 Highwater 1764
Task: httpd 0x3fcede60 -> MALLOC_CAP_INTERNAL: 112 MALLOC_CAP_SPIRAM: 176 Highwater 284
Invalid handle 0x580e
Task: ipc0 0x3fcaeeb8 -> MALLOC_CAP_INTERNAL: 21964 MALLOC_CAP_SPIRAM: 0 Highwater 732
Task: ipc1 0x3fcaf634 -> MALLOC_CAP_INTERNAL: 32 MALLOC_CAP_SPIRAM: 0 Highwater 720
Task: btController 0x3fcda70c -> MALLOC_CAP_INTERNAL: 7492 MALLOC_CAP_SPIRAM: 0 Highwater 1944
Task: esp_timer 0x3fcb0838 -> MALLOC_CAP_INTERNAL: 112 MALLOC_CAP_SPIRAM: 0 Highwater 2988
~~~

jonsmirl, 2024-04-18:
This small program hits it:
~~~
/*
   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include <cstring>
#include <esp_err.h>
#include <esp_log.h>
#include <rom/ets_sys.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include "esp_spiffs.h"
#include "esp_heap_task_info.h"

#include "app_screen.h"
#include "app_matter.h"

static const char *TAG = "app_main";

#define MAX_TASK_NUM 20                         // Max number of per tasks info that it can store
#define MAX_BLOCK_NUM 20                        // Max number of per block info that it can store

static size_t s_prepopulated_num = 0;
static heap_task_totals_t s_totals_arr[MAX_TASK_NUM];
static heap_task_block_t s_block_arr[MAX_BLOCK_NUM];

static void esp_dump_per_task_heap_info(void)
{
    heap_task_info_params_t heap_info = {0};
    heap_info.caps[0] = MALLOC_CAP_INTERNAL;        // Gets heap with CAP_INTERNAL capabilities
    heap_info.mask[0] = MALLOC_CAP_INTERNAL;
    heap_info.caps[1] = MALLOC_CAP_SPIRAM;       // Gets heap info with CAP_SPIRAM capabilities
    heap_info.mask[1] = MALLOC_CAP_SPIRAM;
    heap_info.tasks = NULL;                     // Passing NULL captures heap info for all tasks
    heap_info.num_tasks = 0;
    heap_info.totals = s_totals_arr;            // Gets task wise allocation details
    heap_info.num_totals = &s_prepopulated_num;
    heap_info.max_totals = MAX_TASK_NUM;        // Maximum length of "s_totals_arr"
    heap_info.blocks = s_block_arr;             // Gets block wise allocation details. For each block, gets owner task, address and size
    heap_info.max_blocks = MAX_BLOCK_NUM;       // Maximum length of "s_block_arr"

    heap_caps_get_per_task_info(&heap_info);

    for (int i = 0 ; i < *heap_info.num_totals; i++) {
        if (heap_info.totals[i].task == (TaskHandle_t)0x580e) {
            printf("Invalid handle %p\n", heap_info.totals[i].task);
            continue;
        }
        printf("Task: %s %p -> MALLOC_CAP_INTERNAL: %d MALLOC_CAP_SPIRAM: %d Highwater %d\n",
                heap_info.totals[i].task ? pcTaskGetName(heap_info.totals[i].task) : "Pre-Scheduler allocs" ,
                heap_info.totals[i].task,
                heap_info.totals[i].size[0],    // Heap size with CAP_INTERNAL capabilities
                heap_info.totals[i].size[1],   // Heap size with CAP_SPIRAM capabilities
                heap_info.totals[i].task ? uxTaskGetStackHighWaterMark(heap_info.totals[i].task) : 0);
    }

    printf("\n\n");
}

void memTask(void *pvParameter)
{
    int dump = 0;

    while (1) {
        /* Delay 1 tick (assumes FreeRTOS tick is 10ms */
        vTaskDelay(pdMS_TO_TICKS(20000));

        dump++;
        ESP_LOGI(TAG, "Dump #%d Free internal %d biggest block %d total %d PSRAM free %d", dump,
            heap_caps_get_free_size(MALLOC_CAP_INTERNAL), 
            heap_caps_get_largest_free_block(MALLOC_CAP_INTERNAL), 
            xPortGetFreeHeapSize(), heap_caps_get_free_size(MALLOC_CAP_SPIRAM));
        esp_dump_per_task_heap_info();
#ifdef foo
        if (dump % 5 == 0) {
            ESP_LOGI(TAG, "Dump #%d Free %d", dump, xPortGetFreeHeapSize());
            heap_caps_print_heap_info(MALLOC_CAP_INTERNAL);
            ets_printf("----\n");
            heap_caps_print_heap_info(MALLOC_CAP_SPIRAM);
            ets_printf("==========\n");
        }
#endif        
    }

    /* A task should NEVER return */
    vTaskDelete(NULL);
}

extern "C" void app_main()
{
    //lua_task_start();

    StackType_t* mem_task_stack = (StackType_t*)heap_caps_calloc(1, 4096, MALLOC_CAP_SPIRAM);
    static StaticTask_t mem_task;
    xTaskCreateStatic(memTask, "mem", 4096, NULL, 0, mem_task_stack, &mem_task);

    //app_matter();
}
~~~
~~~
I (20500) app_main: Dump #1 Free internal 384664 biggest block 311268 total 384664 PSRAM free 2087288
Task: Pre-Scheduler allocs 0x0 -> MALLOC_CAP_INTERNAL: 3692 MALLOC_CAP_SPIRAM: 392 Highwater 0
Task: main 0x3fc9b0ac -> MALLOC_CAP_INTERNAL: 136 MALLOC_CAP_SPIRAM: 4100 Highwater 6552
Invalid handle 0x580e
Task: ipc0 0x3fc97240 -> MALLOC_CAP_INTERNAL: 12840 MALLOC_CAP_SPIRAM: 0 Highwater 756
Task: ipc1 0x3fc979bc -> MALLOC_CAP_INTERNAL: 32 MALLOC_CAP_SPIRAM: 0 Highwater 760
~~~

jonsmirl, 2024-04-18:
I added a printf for the mem task handle. It does not match what is in the array.
~~~
I (498) main_task: Calling app_main()
Mem task handle 0x3fc94e9c
I (499) main_task: Returned from app_main()
I (20500) app_main: Dump #1 Free internal 384664 biggest block 311268 total 384664 PSRAM free 2087288
Task: Pre-Scheduler allocs 0x0 -> MALLOC_CAP_INTERNAL: 3692 MALLOC_CAP_SPIRAM: 392 Highwater 0
Task: main 0x3fc9b0ac -> MALLOC_CAP_INTERNAL: 136 MALLOC_CAP_SPIRAM: 4100 Highwater 6552
Invalid handle 0x580e
Task: ipc0 0x3fc97240 -> MALLOC_CAP_INTERNAL: 12840 MALLOC_CAP_SPIRAM: 0 Highwater 756
Task: ipc1 0x3fc979bc -> MALLOC_CAP_INTERNAL: 32 MALLOC_CAP_SPIRAM: 0 Highwater 760
~~~

atanisoft, 2024-04-19:
looks to be related to https://github.com/espressif/esp-idf/issues/13467.

jonsmirl, 2024-04-21:
The task structures are getting corrupted over time.  I had a device sitting on my desk for about a day, then it did this crash in uxTaskGetStackHighWaterMark().  I had not touched the device in hours, but it is receiving wifi and ble traffic.  All of my tasks are started at boot and never change.  And why doesn't my little task (in the comment above) show in the task list?

~~~
I (19068078) app_main: Dump #952 Free internal 27560 biggest block 13804 total 27560 PSRAM free 1686880
Task: Pre-Scheduler allocs 0x0 -> MALLOC_CAP_INTERNAL: 3708 MALLOC_CAP_SPIRAM: 392 Highwater 0
Task: main 0x3fcb3788 -> MALLOC_CAP_INTERNAL: 80920 MALLOC_CAP_SPIRAM: 73764 Highwater 5116
Task: lua 0x3fca147c -> MALLOC_CAP_INTERNAL: 69908 MALLOC_CAP_SPIRAM: 151368 Highwater 35408
Task: wifi 0x3fcc729c -> MALLOC_CAP_INTERNAL: 18260 MALLOC_CAP_SPIRAM: 60196 Highwater 2632
Task: CHIP 0x3fcced04 -> MALLOC_CAP_INTERNAL: 53348 MALLOC_CAP_SPIRAM: 36864 Highwater 3328
Task: nimble_host 0x3fcdccc4 -> MALLOC_CAP_INTERNAL: 60 MALLOC_CAP_SPIRAM: 456 Highwater 1800
Task: tiT 0x3fcc45f0 -> MALLOC_CAP_INTERNAL: 448 MALLOC_CAP_SPIRAM: 400 Highwater 148
Task: sys_evt 0x3fcc33cc -> MALLOC_CAP_INTERNAL: 200 MALLOC_CAP_SPIRAM: 104 Highwater 1704
Task: httpd 0x3fcee68c -> MALLOC_CAP_INTERNAL: 112 MALLOC_CAP_SPIRAM: 176 Highwater 396
Task: ipc0 0x3fcaf894 -> MALLOC_CAP_INTERNAL: 21964 MALLOC_CAP_SPIRAM: 0 Highwater 720
Task: ipc1 0x3fcb0010 -> MALLOC_CAP_INTERNAL: 32 MALLOC_CAP_SPIRAM: 0 Highwater 724
Task: btController 0x3fcd5c78 -> MALLOC_CAP_INTERNAL: 7496 MALLOC_CAP_SPIRAM: 0 Highwater 1944

I (19075636) ROUTE_HOOK: Received RIO
I (19075636) ROUTE_HOOK: prefix FD89:EBF4:10D6:: lifetime 1800
I (19088109) app_main: Dump #953 Free internal 26684 biggest block 13804 total 26684 PSRAM free 1686260
Task: Pre-Scheduler allocs 0x0 -> MALLOC_CAP_INTERNAL: 3708 MALLOC_CAP_SPIRAM: 392 Highwater 0
Task: main 0x3fcb3788 -> MALLOC_CAP_INTERNAL: 80920 MALLOC_CAP_SPIRAM: 73764 Highwater 5116
Task: lua 0x3fca147c -> MALLOC_CAP_INTERNAL: 69908 MALLOC_CAP_SPIRAM: 151876 Highwater 35408
Task: wifi 0x3fcc729c -> MALLOC_CAP_INTERNAL: 18844 MALLOC_CAP_SPIRAM: 60196 Highwater 2632
Task: CHIP 0x3fcced04 -> MALLOC_CAP_INTERNAL: 53348 MALLOC_CAP_SPIRAM: 36864 Highwater 3328
Task: nimble_host 0x3fcdccc4 -> MALLOC_CAP_INTERNAL: 60 MALLOC_CAP_SPIRAM: 456 Highwater 1800
Task: tiT 0x3fcc45f0 -> MALLOC_CAP_INTERNAL: 448 MALLOC_CAP_SPIRAM: 400 Highwater 148
Task: sys_evt 0x3fcc33cc -> MALLOC_CAP_INTERNAL: 200 MALLOC_CAP_SPIRAM: 104 Highwater 1704
Task: httpd 0x3fcee68c -> MALLOC_CAP_INTERNAL: 112 MALLOC_CAP_SPIRAM: 176 Highwater 396
Task: ipc0 0x3fcaf894 -> MALLOC_CAP_INTERNAL: 21964 MALLOC_CAP_SPIRAM: 0 Highwater 720
Task: ipc1 0x3fcb0010 -> MALLOC_CAP_INTERNAL: 32 MALLOC_CAP_SPIRAM: 0 Highwater 724
Task: btController 0x3fcd5c78 -> MALLOC_CAP_INTERNAL: 7496 MALLOC_CAP_SPIRAM: 0 Highwater 1944

I (19108140) app_main: Dump #954 Free internal 27560 biggest block 13804 total 27560 PSRAM free 1686260
Task: Pre-Scheduler allocs 0x0 -> MALLOC_CAP_INTERNAL: 3708 MALLOC_CAP_SPIRAM: 392 Highwater 0
Task: main 0x3fcb3788 -> MALLOC_CAP_INTERNAL: 80920 MALLOC_CAP_SPIRAM: 73764 Highwater 5116
Task: lua 0x3fca147c -> MALLOC_CAP_INTERNAL: 69908 MALLOC_CAP_SPIRAM: 151876 Highwater 35408
Task: wifi 0x3fcc729c -> MALLOC_CAP_INTERNAL: 18260 MALLOC_CAP_SPIRAM: 60196 Highwater 2632
Task: CHIP 0x3fcced04 -> MALLOC_CAP_INTERNAL: 53348 MALLOC_CAP_SPIRAM: 36864 Highwater 3328
Task: nimble_host 0x3fcdccc4 -> MALLOC_CAP_INTERNAL: 60 MALLOC_CAP_SPIRAM: 456 Highwater 1800
Task: tiT 0x3fcc45f0 -> MALLOC_CAP_INTERNAL: 448 MALLOC_CAP_SPIRAM: 400 Highwater 148
Task: sys_evt 0x3fcc33cc -> MALLOC_CAP_INTERNAL: 200 MALLOC_CAP_SPIRAM: 104 Highwater 1704
Task: httpd 0x3fcee68c -> MALLOC_CAP_INTERNAL: 112 MALLOC_CAP_SPIRAM: 176 Highwater 396
Task: ipc0 0x3fcaf894 -> MALLOC_CAP_INTERNAL: 21964 MALLOC_CAP_SPIRAM: 0 Highwater 720
Task: ipc1 0x3fcb0010 -> MALLOC_CAP_INTERNAL: 32 MALLOC_CAP_SPIRAM: 0 Highwater 724
Task: btController 0x3fcd5c78 -> MALLOC_CAP_INTERNAL: 7496 MALLOC_CAP_SPIRAM: 0 Highwater 1944

I (19128172) app_main: Dump #955 Free internal 27560 biggest block 13804 total 27560 PSRAM free 1685648
Task: Pre-Scheduler allocs 0x0 -> MALLOC_CAP_INTERNAL: 3708 MALLOC_CAP_SPIRAM: 392 Highwater 0
Task: main 0x3fcb3788 -> MALLOC_CAP_INTERNAL: 80920 MALLOC_CAP_SPIRAM: 73764 Highwater 5116
Task: lua 0x3fca147c -> MALLOC_CAP_INTERNAL: 69908 MALLOC_CAP_SPIRAM: 152452 Highwater 35408
Task: wifi 0x3fcc729c -> MALLOC_CAP_INTERNAL: 18260 MALLOC_CAP_SPIRAM: 60196 Highwater 2632
Task: CHIP 0x3fcced04 -> MALLOC_CAP_INTERNAL: 53348 MALLOC_CAP_SPIRAM: 36864 Highwater 3328
Task: nimble_host 0x3fcdccc4 -> MALLOC_CAP_INTERNAL: 60 MALLOC_CAP_SPIRAM: 456 Highwater 1800
Task: tiT 0x3fcc45f0 -> MALLOC_CAP_INTERNAL: 448 MALLOC_CAP_SPIRAM: 420 Highwater 148
Task: sys_evt 0x3fcc33cc -> MALLOC_CAP_INTERNAL: 200 MALLOC_CAP_SPIRAM: 104 Highwater 1704
Task: httpd 0x3fcee68c -> MALLOC_CAP_INTERNAL: 112 MALLOC_CAP_SPIRAM: 176 Highwater 396
Task: ipc0 0x3fcaf894 -> MALLOC_CAP_INTERNAL: 21964 MALLOC_CAP_SPIRAM: 0 Highwater 720
Task: ipc1 0x3fcb0010 -> MALLOC_CAP_INTERNAL: 32 MALLOC_CAP_SPIRAM: 0 Highwater 724
Task: btController 0x3fcd5c78 -> MALLOC_CAP_INTERNAL: 7496 MALLOC_CAP_SPIRAM: 0 Highwater 19Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42176f16  PS      : 0x00060230  A0      : 0x82009e38  A1      : 0x3c1f6160  
[... cut ...]
The BLE is always listening for advertisements, 17786240 is the last time one of my advertising packets was processed. The crash occurred at 19128172 when printing the btController stack.
~~~
I (17786240) ble_advert: BLE 34:f1:ae:d4:08:00 Count 4266 RSSI 184 Temp 2264 Humid 4207 Battery 3367
I (17786241) esp_matter_attribute: ********** W : Endpoint 0x0009's Cluster 0xFFF3FC01's Attribute 0x00000000 is 0 **********
I (17786373) ble_advert: BLE 34:f1:ae:d4:08:00 Count 4266 RSSI 184 Temp 2264 Humid 4207 Battery 3367
I (17786374) esp_matter_attribute: ********** W : Endpoint 0x0009's Cluster 0xFFF3FC01's Attribute 0x00000000 is 0 **********

~~~

jonsmirl, 2024-04-21:
I added this, derived from https://github.com/espressif/esp-idf/issues/13467 to my IDF and restarted the device overnight.

~~~
diff --git a/components/heap/heap_caps_init.c b/components/heap/heap_caps_init.c
index 063af207a8..47f3c4f5e6 100644
--- a/components/heap/heap_caps_init.c
+++ b/components/heap/heap_caps_init.c
@@ -129,15 +129,16 @@ void heap_caps_init(void)
              * is done by the top level API heap_caps_malloc(). So we need to add it manually
              * after successful allocation. Allocate extra 4 bytes for that purpose. */
             heaps_array = multi_heap_malloc(temp_heaps[i].heap, MULTI_HEAP_ADD_BLOCK_OWNER_SIZE(sizeof(heap_t) * num_heaps));
+            assert(heaps_array != NULL);
+            MULTI_HEAP_SET_BLOCK_OWNER(heaps_array);
+            heaps_array = (heap_t *)MULTI_HEAP_ADD_BLOCK_OWNER_OFFSET(heaps_array);
             if (heaps_array != NULL) {
                 break;
             }
         }
     }
     assert(heaps_array != NULL); /* if NULL, there's not enough free startup heap space */
-    MULTI_HEAP_SET_BLOCK_OWNER(heaps_array);
-    heaps_array = (heap_t *)MULTI_HEAP_ADD_BLOCK_OWNER_OFFSET(heaps_array);
-
+    
     memcpy(heaps_array, temp_heaps, sizeof(heap_t)*num_heaps);
 
     /* Iterate the heaps and set their locks, also add them to the linked list. */
~~~

jonsmirl, 2024-04-21:
I ran with the device sitting idle for 24hrs with that patch without a crash.

So why does this task show in the list?

    StackType_t* mem_task_stack = (StackType_t*)heap_caps_calloc(1, 4096, MALLOC_CAP_SPIRAM);
    static StaticTask_t mem_task;
    xTaskCreateStatic(memTask, "mem", 4096, NULL, 0, mem_task_stack, &mem_task);

SoucheSouche, 2024-04-22:
Hello, @jonsmirl,
Thanks for reporting the issue. We already have been made aware of this problem (https://github.com/espressif/esp-idf/issues/13467) and I worked on a fix for the bug. The changes are already available on the master and release/v5.2 branches for a week.

jonsmirl, 2024-04-22:
I am retesting with a clean 'origin/release/v5.2'. Looks like I merged both
my fix and your fix into my tree when I moved from 5.2.1 to release 5.2.

closed 2024-04-23 by jonsmirl