espressif/esp-idf #13644: esp_wifi_80211_tx is inconsistent at sending frames

jdloft, 2024-04-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-3499-g636ff35b52

### Espressif SoC revision.

ESP32-D0WDQ6 (revision v1.0)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 Adafruit feather and others

### Power Supply used.

USB

### What is the expected behavior?

ESP32 should send a frame every time the command esp_wifi_80211_tx returns `ESP_OK` and/or the CB registered with `esp_wifi_set_tx_done_cb` receives a successful TX done event. These frames are being send in a periodic ESP timer (50 ms apart; see Wireshark screenshot below).

### What is the actual behavior?

Instead, our ESP32 (and actually a number of different ESP32 boards we've tried) randomly doesn't send a number of frames. This can be seen with the following Wireshark screenshot. The test was setup to send the same frame 10 times. Note the skip in sequence numbers between 7 and 9 and the gap in time between those frames.
[image]

This varies between no frames dropped, to 4/10 frames dropped.

### Steps to reproduce.

1. Write a function that sends a frame and send it in a timer (50 ms apart):
2. Start capture in Wireshark
3. Reset chip a number of times to observe issue

### Debug Logs.

```plain
I (2805) main: Starting to send symbol 1
I (2855) send chip: Chip 0: 1.966284
I (2855) send chip: Chip transmit time 0.000794
I (2875) main: Successfully sent 295 bytes
I (2905) send chip: Chip 1: 2.016160
I (2905) send chip: Chip transmit time 0.000273
I (2915) main: Successfully sent 295 bytes
I (2955) send chip: Chip 2: 2.066160
I (2955) send chip: Chip transmit time 0.000254
I (2965) main: Successfully sent 295 bytes
I (3005) send chip: Chip 3: 2.116160
I (3005) send chip: Chip transmit time 0.000254
I (3015) main: Successfully sent 295 bytes
I (3055) send chip: Chip 4: 2.166160
I (3055) send chip: Chip transmit time 0.000253
I (3065) main: Successfully sent 295 bytes
I (3105) send chip: Chip 5: 2.216176
I (3105) send chip: Chip transmit time 0.000381
I (3115) main: Successfully sent 295 bytes
I (3155) send chip: Chip 6: 2.266168
I (3155) send chip: Chip transmit time 0.000266
I (3165) main: Successfully sent 295 bytes
I (3205) send chip: Chip 7: 2.316160
I (3205) send chip: Chip transmit time 0.000258
I (3215) main: Successfully sent 295 bytes
I (3255) send chip: Chip 8: 2.366160
I (3255) send chip: Chip transmit time 0.000257
I (3265) main: Successfully sent 295 bytes
I (3305) send chip: Chip 9: 2.416160
I (3305) send chip: Chip transmit time 0.000257
I (3315) main: Successfully sent 295 bytes
I (3355) send chip: Symbol complete
```

### More Information.

Here are a number of things we tried changing to attempt to narrow down the issue. We were able to reproduce it under the following conditions:
* Associated and disassociated with an AP
* In `WIFI_MODE_AP` and `WIFI_MODE_STA`
* Various timer intervals (up to one second apart and down to 13 ms)
* With and without power save enabled
* In a faraday cage verified by seeing no other frames captured from other devices

Note: this is not similar to #10668 or #13319 since no RTS packets are captured from the ESP32.

This was also reproduced on the following IDF versions:
* `master` or `v5.3-dev-3499-g636ff35b52`
* `v5.2.1`
* `v4.4.7`
* `v4.3.7`
* and `v4.2.1`

The debug logs are logs from our application. The lines starting with "Successfully sent" is printed in the `esp_wifi_set_tx_done_cb` callback, demonstrating that the callback doesn't indicate any problem with transmission. Again, chip 7 (SN=8 in Wireshark) was never sent.

zhangyanjiaoesp, 2024-04-22:
@jdloft 

> Reset chip a number of times to observe issue

Do you mean the issue appears after multiple resets ? Can you provide a demo for us to test?

jdloft, 2024-04-23:
> Can you provide a demo for us to test?

Sure thing. See attached: [gh-13644.tar.gz](https://github.com/espressif/esp-idf/files/15083833/gh-13644.tar.gz)

> Do you mean the issue appears after multiple resets

What I mean is the problem isn't 100% consistent; sometimes all the frames are sent and sometimes they aren't. To reproduce you should reset the device until the problem occurs.

The demo sends ten beacon frames 50 ms apart from the SSID 12:34:56:78:9a:bc on WiFi channel 1.
[image]
In this case the frame with SN 3 was not sent.

zhangyanjiaoesp, 2024-04-24:
@jdloft Did you capture the packets in the shield box? When the SN3 frame is not captured, does the log display fail? Could it be that the packet was sent, but Wireshark did not catch it?

jdloft, 2024-04-24:
Yes this still happens in the faraday cage, so it shouldn't be something to do with clear channel sensing. I also thought it may have been the packet capture so I captured the raw power on the channel in the faraday cage:
[image]
There are definitely gaps where nothing is being transmitted.

zhangyanjiaoesp, 2024-04-25:
@jdloft thanks for the update, I will test it on my side.

zhangyanjiaoesp, 2024-05-11:
@jdloft I have tested locally in the shield box, but I didn't reproduce your issue. I have sent 8000 packets, and the Wireshark capture all the packets.

> [image]
How did you get this graph?

jdloft, 2024-09-29:
I can’t reproduce this with the latest IDF.

closed 2024-09-29 by jdloft