espressif/esp-idf #13646: Magic number (ESP_IMAGE_HEADER_MAGIC) is not visible in the build file

chaabanemalki, 2024-04-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have enabled project app description

```
#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager
```

Q1: When I build I can see all information firmware version, build date project name in the bin file but not the magic number ?
Q2: Is there a python script in esp-idf that can retrieve app description from the binary file ?

mahavirj, 2024-04-22:
@chaabanemalki 

Some pointers that can help in this regard:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/app_image_format.html

https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/basic-commands.html#output-bin-image-details-image-info

chaabanemalki, 2024-04-24:
@mahavirj , Thank you for sharing these links. I have already consulted them before reaching here becauseI couldn't find what I'm looking for in them.

My my understanding is that in order to see `esp_app_desc_t`  in `DROM` segment, I have to do the following:

Include app name and version in sdkconfig 

```
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
```

Set these information in cmakelists.txt

```
set(PROJECT_VER "1.2.3")
project("my_app")
```

When I build  ( idf-4.3.3) I see all the above but not the magic number. See binary view from my build file

[image]

KonstantinKondrashov, 2024-04-24:
Hi @chaabanemalki!
Here is the structure of the app bin file:
0x0000 - [esp_image_header_t](https://github.com/espressif/esp-idf/blob/master/components/bootloader_support/include/esp_app_format.h#L76-L107) ESP_IMAGE_HEADER_MAGIC (0xE9)
0x0018 [esp_image_segment_header_t](https://github.com/espressif/esp-idf/blob/master/components/bootloader_support/include/esp_app_format.h#L114-L120)  ESP_APP_DESC_MAGIC_WORD (0xABCD5432)
0x0020 [esp_app_desc_t](https://github.com/espressif/esp-idf/blob/master/components/esp_app_format/include/esp_app_desc.h#L21-L37)
....

According to your picture everything on the right place. Magic numbers are seen there.

> Q2: Is there a python script in esp-idf that can retrieve app description from the binary file ?

`esptool.py -c esp32 image_info build/*.bin  --version 2`
It is not available on the 4.3.

chaabanemalki, 2024-04-24:
@KonstantinKondrashov , thank you for the clarification, somehow I miss read the file info :)

closed 2024-04-24 by chaabanemalki