espressif/esp-idf #13648: httpd_resp_sendstr vs httpd_resp_send

ceewanna, 2024-04-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I expected that httpd_resp_sendstr would simply send out plain text of which the content is in str. However, it actually delivers a markup html text. When I looked into the code it actually relied on httpd_resp_send which would deliver markup html.

Is there anyway to get plain text response from httpd_resp_sendstr or is there any alternatives to this?

```
static inline esp_err_t httpd_resp_sendstr(httpd_req_t *r, const char *str) {
    return httpd_resp_send(r, str, (str == NULL) ? 0 : HTTPD_RESP_USE_STRLEN);
}
```

hmalpani, 2024-04-22:
Hello @ceewanna 
https_resp_sendstr is expected to send HTTP response. See the documentation [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_http_server.html#_CPPv418httpd_resp_sendstrP11httpd_req_tPKc)

You can use [httpd_send()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_http_server.html#_CPPv410httpd_sendP11httpd_req_tPKc6size_t) to send plaintext response. Please let me know if it helps. Thanks!

ceewanna, 2024-04-22:
Thanks @hmalpani
1. I have never been aware of this httpd_send() but I have just done some quick test on it.
```
    std::string msg = "OK";
    esp_err_t res = httpd_resp_set_status(req, HTTPD_200);
    ESP_LOGI(TAG, "httpd_resp_set_status %d", res); // res = 0
    int ret = httpd_send(req, msg.c_str(), msg.length());
    ESP_LOGI(TAG, "httpd_send return %d", ret); // ret = 2
```

Refreshing the page got it to idle and not returning anything. Any additional stuff I have to add?
Previously, I have been doing this.
```
std::string message;
httpd_resp_sendstr(req, message.c_str());
```
The content of message variable showed in the body of http markup. The page has no other issues except I don't want the markup. I want only the message in text to show on the page. 

2. Down in https_resp_sendstr it reuses https_resp_send with only making auto len and I have no idea why it has to be so.

ceewanna, 2024-04-23:
After a little more struggling checking what esp32 arduino does for httpserver.send(), I found what I need for esp-idf. This will simply send plain text response for me. 
```
    httpd_resp_set_status(req, HTTPD_200);
    httpd_resp_set_type(req, "text/plain");
    httpd_resp_sendstr(req, message.c_str());
```
Thank you for everyone's time. I suppose my case could be closed. 
Anyway I found the namings in esp-idf could be confusing for a newbie.

hmalpani, 2024-04-29:
Hope this issue is resolved. Closing this issue. Please feel free to reopen

closed 2024-04-29 by hmalpani