espressif/esp-idf #13649: esp32 Reconnect issue

fatimatayyab71, 2024-04-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have programmed the esp32-wroom-32 with BLE-ANCS example code, it can connect with apple device but when I want to reconnect it it do not reconnect automatically, rather i have to forget the device and pair it again. but when i programmed the same code to esp32-s3 it reconnects automatically. 
On the hardware description on espressif site it is written that not recommended for new designs, does it means that it is not compatible with apple devices or something else is the reason?

fatimatayyab71, 2024-04-20:
Also BLE 4.2 is disabled in default configuration for esp32 while for esp32s3 it is enabled and added during built, how can we add it into the configuration?

zhp0406, 2024-04-24:
You can use "idf.py menuconfig" to decide whether the 4.2 feature is enabled or not.
(Top) → Component config → Bluetooth → Bluedroid Options
[*] Enable BLE 4.2 features

zhp0406, 2024-04-24:
What version of IDF are you using?

AxelLin, 2024-05-09:
@fatimatayyab71 
Could you share your esp-idf version? (git describe --tags)
Could you share the sdkconfig for both esp32 and esp32s3?

Alvin1Zhang, 2024-12-18:
Thanks for reporting, will close due to inactivity, feel free to reopen with more updates or if the issue still happens.

closed 2024-12-18 by Alvin1Zhang