espressif/esp-idf #13650: esp idf 4.4.6 SDMMC variable bug?

uragan1987, 2024-04-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp idf 4.4.6 

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

want to compile after adding just more debug output inside code, cant compile

target is: ESP32

### What is the actual behavior?

get error: "clk" undeclared

but "clk" is one line later used by code without beeing declared?????
how does it works?

### Steps to reproduce.

this is the line:
https://github.com/espressif/esp-idf/blob/309ffa5565ad88edc262f2d6fb3e45567192bbb0/components/driver/sdmmc_host.c#L442

just add 
printf("clk %d", clk);

### Build or installation Logs.

```plain
[1/9] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Users\ProgPC\Desktop\MMC_SD\HelloWorld\sd_card\sdmmc\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf4.4_py3.11_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v4.4.6/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/Users/ProgPC/Desktop/MMC_SD/HelloWorld/sd_card/sdmmc/build/bootloader/bootloader.bin"
Bootloader binary size 0x6430 bytes. 0xbd0 bytes (11%) free.
[2/7] Building C object esp-idf/driver/CMakeFiles/__idf_driver.dir/sdmmc_host.c.obj
FAILED: esp-idf/driver/CMakeFiles/__idf_driver.dir/sdmmc_host.c.obj 
ccache C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe  -IC:/Users/ProgPC/Desktop/MMC_SD/HelloWorld/sd_card/sdmmc/build/config -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/driver/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/driver/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/driver/include/driver -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/newlib/platform_include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/freertos/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/freertos/include/esp_additions/freertos -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/freertos/port/xtensa/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/freertos/include/esp_additions -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_hw_support/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_hw_support/include/soc -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_hw_support/include/soc/esp32 -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_hw_support/port/esp32/. -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_hw_support/port/esp32/private_include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/heap/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/log/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/lwip/include/apps -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/lwip/include/apps/sntp -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/lwip/lwip/src/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/lwip/port/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/lwip/port/esp32/include/arch -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/soc/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/soc/esp32/. -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/soc/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/hal/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/hal/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/hal/platform_port/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_rom/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_rom/include/esp32 -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_rom/esp32 -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_common/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_system/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_system/port/soc -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_system/port/public_compat -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/xtensa/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/xtensa/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_pm/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_ringbuf/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/efuse/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/efuse/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/vfs/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_wifi/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_event/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_netif/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_eth/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/tcpip_adapter/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_phy/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_phy/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_ipc/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/app_trace/include -IC:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_timer/include -mlongcalls -Wno-frame-address  -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fmacro-prefix-map=C:/Users/ProgPC/Desktop/MMC_SD/HelloWorld/sd_card/sdmmc=. -fmacro-prefix-map=C:/Espressif/frameworks/esp-idf-v4.4.6=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.4.6-dirty\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/driver/CMakeFiles/__idf_driver.dir/sdmmc_host.c.obj -MF esp-idf\driver\CMakeFiles\__idf_driver.dir\sdmmc_host.c.obj.d -o esp-idf/driver/CMakeFiles/__idf_driver.dir/sdmmc_host.c.obj -c C:/Espressif/frameworks/esp-idf-v4.4.6/components/driver/sdmmc_host.c
C:/Espressif/frameworks/esp-idf-v4.4.6/components/driver/sdmmc_host.c: In function 'sdmmc_host_init_slot':
C:/Espressif/frameworks/esp-idf-v4.4.6/components/driver/sdmmc_host.c:442:19: error: 'clk' undeclared (first use in this function)
  printf("clk %d", clk);
                   ^~~
C:/Espressif/frameworks/esp-idf-v4.4.6/components/driver/sdmmc_host.c:442:19: note: each undeclared identifier is reported only once for each function it appears in
ninja: build stopped: subcommand failed.
Executing action: all (aliases: build)
Running ninja in directory C:\Users\ProgPC\Desktop\MMC_SD\HelloWorld\sd_card\sdmmc\build
Executing "ninja all"...
```

### More Information.

win10 20H2

igrr, 2024-04-20:
You probably need to print the value of `sdmmc_slot_gpio_num[slot].clk` — please check the definition of `configure_pin` macro.

https://github.com/espressif/esp-idf/blob/309ffa5565ad88edc262f2d6fb3e45567192bbb0/components/driver/sdmmc_host.c#L64

uragan1987, 2024-04-21:
maybe you can explain me, why "clk" at code not defined, but there is no compiler error?

uragan1987, 2024-04-21:
i have a ESP32S3, using (1 line) connection (only D0) to SDMMC works fine
but same with ESP32 devkitv1 (ESP32-WROOM-32) wont work, cant get connected
cable was checked 100x times, but every time get a 0x107 error, PIN on 14,15,2 

dunno whats wrong here

thats why i currently stucked on this "clk", try to change it

igrr, 2024-04-21:
> thats why i currently stucked on this "clk", try to change it

Sadly it's not possible: on the ESP32, SDMMC pins are fixed. The ability to route SDMMC to arbitrary pins has been added starting from ESP32-S3.

> maybe you can explain me, why "clk" at code not defined, but there is no compiler error?

This is because the `configure_pin` macro I have linked to above replaces the name `clk` with `sdmmc_slot_gpio_num[slot].clk`.

> every time get a 0x107 error, PIN on 14,15,2

We can try to help you with this issue. Please enable verbose log level in menuconfig and attach the output you get when initializing the card. Also please attach the schematic showing how the card is connected.

uragan1987, 2024-04-21:
here are all logs

[cmd_log.txt](https://github.com/espressif/esp-idf/files/15052816/cmd_log.txt)
[image]
[monitor.txt](https://github.com/espressif/esp-idf/files/15052817/monitor.txt)

uragan1987, 2024-05-06:
@igrr i dont have a D0 on this PCB

I need to connect D3 to VDD?
Cause on ESP32-S3 i dont connect D3 to board and 1-line work fine

igrr, 2024-05-06:
> i dont have a D0 on this PCB

That's the orange line shown on your diagram, connected between GPIO2 of the ESP and D0 (data 0) line of the card. It needs a pull-up resistor to VDD (3.3V).

> I need to connect D3 to VDD?

The best option is to connect it to VDD with a pull-up resistor.

Alvin1Zhang, 2025-05-08:
@uragan1987 Thanks for reporting, would you please help share if any updates for the issue? Thanks.

uragan1987, 2025-05-09:
@Alvin1Zhang i dont have success with it, still not working
had freeze there project over a year ago.
maybe there are some updates, havent checked it, but i dont think so

uragan1987, 2025-05-09:
my goal was to readCID and to change CID with samsung SD MMC
there is a  backdoor, but have no success with this way.
https://github.com/beaups/SamsungCID

maybe somebody has a solution here?

omnitex, 2026-11-03:
Hi @uragan1987, is this still relevant for you?

A few notes from your logs, in case you pick it up again.

### clk

It's not a variable. `configure_pin(clk, ...)` is a macro that expands to `sdmmc_slot_gpio_num[slot].clk`. 

So if you want to print it, print `sdmmc_slot_gpio_num[slot].clk`.

On ESP32 the SDMMC pins are fixed anyway (CLK=14, CMD=15, D0=2), so there's nothing to change there.

### 0x107 errors

Those are mostly harmless. They come from the SDIO probe commands timing out. The card does answer CMD8, so CLK and CMD are wired OK.

The init then gets stuck waiting on the bus. That usually means D0 (GPIO2) is missing a pull-up.

From your diagram it looks like you're using an SD reader module. Does it have pull-ups (~10k to 3.3V) on CMD, D0 and D3? Many of those modules don't, and some only work in SPI mode.

Could you share more details about the module?

### SamsungCID

AFAICT that tool only works on Samsung **eMMC**. It relies on a bug in the firmware of Samsung's eMMC controller, and it refuses to run unless the CID manufacturer ID is `0x15` (Samsung eMMC).

Your card answered CMD8 with `0x1AA` and was detected as SDHC -- are you sure you have correct card for the backdoor?

### IDF version

v4.4.6 is out of support now, so please try a recent release. Newer versions return a timeout error here instead of hanging.

uragan1987, 2026-11-03:
Hi, yes Card ist correct, cause can do Change CID in a Linux System . But a SD Card Reader in Motherboard ist needed for this, with a USB Card IT dont work. 

With e.g., /dev/mmcblk0 IT Work
But as /dev/sdX IT wont Work, 

Thats was my Idea to do this with esp32
Maybe there are easy ways?

omnitex, 2026-11-03:
Did you get the card working with esp32 with the `sd_card` example as baseline? I believe the variable bug this issue was opened for is resolved. Did you get it working past that, with the necessary connections and/or pull-ups? 

I can't assist on the Change CID part, but want to confirm if you still have ESP-IDF troubles, tested ideally on the latest version, let me know