espressif/esp-idf #13653: ESP32-S3+DM9051连接以太网错误

Zhicon, 2024-04-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.1.2

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

使用TCP Server例程并在menuconfig里选择DM9051，预期应正常连接网络并创建TCP服务器

### What is the actual behavior?

将示例程序编译并烧录后，DM9051连接网线，控制台打印输出
I (4396) ethernet_connect: Waiting for IP(s).
然后一直没有反应，相关指示灯也不亮，经查询程序发现在执行emac_dm9051_init()函数后GPR寄存器中的PHY电源处于关闭模式，PHY并没有使能，于是在emac_dm9051_init()函数末尾添加上了
`uint8_t tmp = 0;
dm9051_register_read(emac, DM9051_GPR, &tmp);
tmp&= 0xFE;
dm9051_register_write(emac, DM9051_GPR, tmp);`
添加上后网口指示灯会有相应闪烁，但是在log输出中会出现新的错误提示：
I (395) esp_eth.netif.netif_glue: ethernet attached to netif
I (1995) ethernet_connect: Waiting for IP(s).
E (2695) dm9051.mac: dm9051_get_recv_byte_count(703): receive status error: 1H
E (2695) dm9051.mac: dm9051_alloc_recv_buf(728): get rx frame length failed
经查是接收数据包时会会有4字节帧头，帧头第二字节为状态，返回的状态为0x01，数据手册上查询不到此状态的资料。

### Steps to reproduce.

1.直接打开tcp server示例，在menuconfig中选择dm9051，然后编译并运行，log一直停止
I (1995) ethernet_connect: Waiting for IP(s).
2.修改emac_dm9051_init()后，log停止在
E (2695) dm9051.mac: dm9051_get_recv_byte_count(703): receive status error: 1H
E (2695) dm9051.mac: dm9051_alloc_recv_buf(728): get rx frame length failed

### Debug Logs.

```plain
示例代码编译运行log：
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x175c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2e04
entry 0x403c9908
I (26) boot: ESP-IDF HEAD-HASH-NOTFOUND 2nd stage bootloader
I (27) boot: compile time Apr 20 2024 20:06:15
I (27) boot: Multicore bootloader
I (31) boot: chip revision: v0.1
I (35) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (44) boot.esp32s3: SPI Flash Size : 2MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00100000
I (88) boot: End of partition table
I (92) esp_image: segment 0: paddr=00010020 vaddr=3c040020 size=14b80h ( 84864) map
I (115) esp_image: segment 1: paddr=00024ba8 vaddr=3fc93e00 size=029c8h ( 10696) load
I (118) esp_image: segment 2: paddr=00027578 vaddr=40374000 size=08aa0h ( 35488) load
I (129) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=3bec8h (245448) map
I (174) esp_image: segment 4: paddr=0006bef0 vaddr=4037caa0 size=07254h ( 29268) load
I (188) boot: Loaded app from partition at offset 0x10000
I (188) boot: Disabling RNG early entropy source...
I (199) cpu_start: Multicore app
I (199) cpu_start: Pro cpu up.
I (200) cpu_start: Starting app cpu, entry point is 0x403754ac
0x403754ac: call_start_cpu1 at D:/Espressif/frameworks/esp-idf-v5.1.2/components/esp_system/port/cpu_start.c:157

I (0) cpu_start: App cpu up.
I (218) cpu_start: Pro cpu start user code
I (218) cpu_start: cpu freq: 160000000 Hz
I (218) cpu_start: Application information:
I (221) cpu_start: Project name:     tcp_server
I (226) cpu_start: App version:      1
I (230) cpu_start: Compile time:     Apr 20 2024 20:04:41
I (237) cpu_start: ELF file SHA256:  684e668cae8195d0...
I (242) cpu_start: ESP-IDF:          HEAD-HASH-NOTFOUND
I (248) cpu_start: Min chip rev:     v0.0
I (253) cpu_start: Max chip rev:     v0.99 
I (258) cpu_start: Chip rev:         v0.1
I (263) heap_init: Initializing. RAM available for dynamic allocation:
I (270) heap_init: At 3FC97A60 len 00051CB0 (327 KiB): DRAM
I (276) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (283) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (289) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (296) spi_flash: detected chip: generic
I (300) spi_flash: flash io: dio
W (304) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (317) sleep: Configure to isolate all GPIO pins in sleep state
I (324) sleep: Enable automatic switching of GPIO sleep configuration
I (331) app_start: Starting scheduler on CPU0
I (336) app_start: Starting scheduler on CPU1
I (336) main_task: Started on CPU0
I (346) main_task: Calling app_main()
I (396) esp_eth.netif.netif_glue: 02:00:00:12:34:56
I (396) esp_eth.netif.netif_glue: ethernet attached to netif
I (4396) ethernet_connect: Waiting for IP(s).  log输出一直停止在此处，网口指示灯无信号

修改emac_dm9051_init()：
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x175c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2e04
entry 0x403c9908
I (26) boot: ESP-IDF HEAD-HASH-NOTFOUND 2nd stage bootloader
I (27) boot: compile time Apr 20 2024 20:06:15
I (27) boot: Multicore bootloader
I (31) boot: chip revision: v0.1
I (35) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (44) boot.esp32s3: SPI Flash Size : 2MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00100000
I (88) boot: End of partition table
I (92) esp_image: segment 0: paddr=00010020 vaddr=3c040020 size=14b80h ( 84864) map
I (115) esp_image: segment 1: paddr=00024ba8 vaddr=3fc93e00 size=029c8h ( 10696) load
I (118) esp_image: segment 2: paddr=00027578 vaddr=40374000 size=08aa0h ( 35488) load
I (129) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=3bef0h (245488) map
I (174) esp_image: segment 4: paddr=0006bf18 vaddr=4037caa0 size=07254h ( 29268) load
I (187) boot: Loaded app from partition at offset 0x10000
I (188) boot: Disabling RNG early entropy source...
I (199) cpu_start: Multicore app
I (199) cpu_start: Pro cpu up.
I (199) cpu_start: Starting app cpu, entry point is 0x403754ac
0x403754ac: call_start_cpu1 at D:/Espressif/frameworks/esp-idf-v5.1.2/components/esp_system/port/cpu_start.c:157

I (0) cpu_start: App cpu up.
I (217) cpu_start: Pro cpu start user code
I (217) cpu_start: cpu freq: 160000000 Hz
I (218) cpu_start: Application information:
I (220) cpu_start: Project name:     tcp_server
I (226) cpu_start: App version:      1
I (230) cpu_start: Compile time:     Apr 20 2024 20:04:41
I (236) cpu_start: ELF file SHA256:  c695922d2fdf9fa1...
I (242) cpu_start: ESP-IDF:          HEAD-HASH-NOTFOUND
I (248) cpu_start: Min chip rev:     v0.0
I (253) cpu_start: Max chip rev:     v0.99
I (257) cpu_start: Chip rev:         v0.1
I (262) heap_init: Initializing. RAM available for dynamic allocation:
I (269) heap_init: At 3FC97A60 len 00051CB0 (327 KiB): DRAM
I (276) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (282) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (288) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (296) spi_flash: detected chip: generic
I (299) spi_flash: flash io: dio
W (303) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (317) sleep: Configure to isolate all GPIO pins in sleep state
I (323) sleep: Enable automatic switching of GPIO sleep configuration
I (331) app_start: Starting scheduler on CPU0
I (335) app_start: Starting scheduler on CPU1
I (335) main_task: Started on CPU0
I (345) main_task: Calling app_main()
I (395) esp_eth.netif.netif_glue: 02:00:00:12:34:56
I (395) esp_eth.netif.netif_glue: ethernet attached to netif
I (1995) ethernet_connect: Waiting for IP(s).
I (2685) dm9051.mac: flag = 1, status = 1, rx_len = 257
E (2695) dm9051.mac: dm9051_get_recv_byte_count(703): receive status error: 1H
E (2695) dm9051.mac: dm9051_alloc_recv_buf(728): get rx frame length failed  log输出一直停止在此处，网口指示灯有信号
```

kostaond, 2024-04-22:
How is your DM9051 connected to ESP32-S3? Is it custom board or via jump wires? What is configured SPI frequency?

Zhicon, 2024-04-22:
你好，我是通过跳线将DM9051与ESP32-S3连接，SPI频率为36MHz

kostaond, 2024-04-22:
OK, so it might be problem with SPI signal integrity. Please use wires as short as possible and with matching length. Please also decrease SPI frequency to e.g. 8 MHz.

Zhicon, 2024-04-22:
你好，在36MHz的配置中，初始化DM9051过程中读取芯片的VID、PID、以及OUI都是完全正确的，应该不存在SPI信号不完整的可能，同时我也测试了将SPI频率设置到8MHz还是一样的情况。

closed 2024-04-28 by Zhicon