espressif/esp-idf #13656: Core  0 panic'ed (IllegalInstruction) on the bt component

yayugu, 2024-04-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.0)

### Operating System used.

Windows

### How did you build your project?

CLion IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

Keep running without the crash

### What is the actual behavior?

It suddenly crash when running 1 - 12 hours.

### Steps to reproduce.

1. Run with Bluedroid SPP acceptor (classic only, disabled BLE) along with Wifi Station (coexistence).
2. Keep the connection and sending / receiving packets for 1 - 12 hours.
3. Become 
...

### Debug Logs.

```plain
2024-04-21 22:07:46 I (36602792) SPP_ACCEPTOR: ESP_SPP_WRITE_EVT len=42
2024-04-21 22:07:46 I (36602912) UartItf: xPortGetFreeHeapSize=46268 minimum_free_heap_size=38844 stack=1844
2024-04-21 22:07:46 I (36602912) SPP_ACCEPTOR: ESP_SPP_WRITE_EVT len=100
2024-04-21 22:07:46 I (36602912) SPP_ACCEPTOR: ESP_SPP_WRITE_EVT len=280
2024-04-21 22:07:46 I (36602922) SPP_ACCEPTOR: ESP_SPP_WRITE_EVT len=42
2024-04-21 22:07:46 I (36602972) NTC: xPortGetFreeHeapSize=46500 minimum_free_heap_size=38844 stack=1932
2024-04-21 22:07:46 I (36602982) LED_Task: xPortGetFreeHeapSize=46476 minimum_free_heap_size=38844 stack=252
2024-04-21 22:07:46 I (36602982) SPP_ACCEPTOR: ESP_SPP_WRITE_EVT len=25
2024-04-21 22:07:46 I (36603022) SPP_ACCEPTOR: ESP_SPP_WRITE_EVT len=100
2024-04-21 22:07:46 I (36603022) SPP_ACCEPTOR: ESP_SPP_WRITE_EVT len=42
2024-04-21 22:07:46 I (36603132) SPP_ACCEPTOR: ESP_SPP_WRITE_EVT len=100
2024-04-21 22:07:46 I (36603132) SPP_ACCEPTOR: ESP_SPP_WRITE_EVT len=42
2024-04-21 22:07:46 I (36603212) sd_card: esp_get_minimum_free_heap_size=38844 stack=752
2024-04-21 22:07:46 I (36603242) SPP_ACCEPTOR: ESP_SPP_WRITE_EVT len=100
2024-04-21 22:07:46 I (36603242) SPP_ACCEPTOR: ESP_SPP_WRITE_EVT len=42
2024-04-21 22:07:46 I (36603352) SPP_ACCEPTOR: ESP_SPP_WRITE_EVT len=100
2024-04-21 22:07:46 I (36603352) SPP_ACCEPTOR: ESP_SPP_WRITE_EVT len=42
2024-04-21 22:07:47 I (36603432) spp_TASK: esp_get_minimum_free_heap_size=38844 stack=1080
[... 76 lines cut ...]
2024-04-21 22:07:50 A10     : 0x00000000  A11     : 0x3ffc1ed3  A12     : 0x3ffc1e7f  A13     : 0x00000033
2024-04-21 22:07:50 A14     : 0x00000000  A15     : 0x3ffc1e84  SAR     : 0x00000004  EXCCAUSE: 0x00000000
2024-04-21 22:07:50 EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
2024-04-21 22:07:50
2024-04-21 22:07:50
2024-04-21 22:07:50 Backtrace: 0x40092ed0:0x3ffc1f40 0x4008b6cb:0x3ffc1f60 0x400924d9:0x3ffc1f80 0x40092a5f:0x3ffc1fa0 0x40082b7e:0x3ffc1fc0 0x40082af5:0x3ffc1fe0 0x40082ad7:0x00000000 |<-CORRUPTED
2024-04-21 22:07:51 0x40092ed0: r_assert at C:/esp-idf/components/bt/controller/esp32/bt.c:1847
2024-04-21 22:07:51 0x4008b6cb: r_assert_param at ??:?
2024-04-21 22:07:51 0x400924d9: r_rwbt_isr at ??:?
2024-04-21 22:07:51 0x40092a5f: r_rwbtdm_isr_wrapper at intc.c:?
2024-04-21 22:07:51 0x40082b7e: hli_c_handler at C:/esp-idf/components/bt/controller/esp32/hli_api.c:92
2024-04-21 22:07:51 0x40082af5: _highint4_stack_switch at C:/esp-idf/components/bt/controller/esp32/hli_vectors.S:177
2024-04-21 22:07:51 0x40082ad7: xt_highint4 at C:/esp-idf/components/bt/controller/esp32/hli_vectors.S:161
2024-04-21 22:07:51
2024-04-21 22:07:51
2024-04-21 22:07:51
2024-04-21 22:07:51
2024-04-21 22:07:51 ELF file SHA256: ec4329bb8b112ca5
2024-04-21 22:07:51
2024-04-21 22:07:51 Rebooting...
```

### More Information.

This error started happening after I upgraded ESP-IDF from v4.0 to v5.2.1

ESP-YTGerd, 2024-04-25:
Hi, @yayugu 
Thanks for your reporting. It's a known issue and we are currently internally assessing approaches to resolve it.
If there are any developments, they will also be updated here.

yayugu, 2024-04-25:
Hi @ESP-YTGerd,
Thank you for the response. That's great to hear and I'm very looking for the update!

ESP-YTGerd, 2024-04-30:
Hi, @yayugu 
Can try again with the attached new lib? in  which some changes were added and let's see if the issue can reproduce. It's also suggested deleting  the old build via `rm -rf build` in project path before re-building your project with the new lib.
[libbtdm_app.zip](https://github.com/espressif/esp-idf/files/15157701/libbtdm_app.zip)

yayugu, 2024-04-30:
@ESP-YTGerd Thank you working on this issue!
I've tried that, but it failed on `esp_bt_controller_init()`. Here is the code and output:

Code (bluetooth.cpp, based on the [SPP acceptor example](https://github.com/espressif/esp-idf/blob/v5.2.1/examples/bluetooth/bluedroid/classic_bt/bt_spp_acceptor/main/main.c)
```
void initBluetooth()
{
    esp_err_t ret;

    ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_BLE));
    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    if ((ret = esp_bt_controller_init(&bt_cfg)) != ESP_OK)
    {
        LOG_BT_E(SPP_TAG, "%s initialize controller failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }
...
```

Output
```
2024-04-30 13:35:58 I (339) cpu_start: Multicore app
2024-04-30 13:35:58 I (347) cpu_start: Pro cpu start user code
2024-04-30 13:35:58 I (347) cpu_start: cpu freq: 240000000 Hz
2024-04-30 13:35:58 I (348) cpu_start: Application information:
2024-04-30 13:35:58 I (351) cpu_start: Project name:     rws2
2024-04-30 13:35:58 I (355) cpu_start: App version:      3.0.1c
2024-04-30 13:35:58 I (360) cpu_start: Compile time:     Apr 30 2024 13:20:16
2024-04-30 13:35:58 I (366) cpu_start: ELF file SHA256:  a5d7ed994e60e52b...
2024-04-30 13:35:58 I (372) cpu_start: ESP-IDF:          v5.2.1-1-g766a415056-dirty
2024-04-30 13:35:58 I (379) cpu_start: Min chip rev:     v1.0
2024-04-30 13:35:58 I (383) cpu_start: Max chip rev:     v3.99
2024-04-30 13:35:58 I (388) cpu_start: Chip rev:         v3.0
2024-04-30 13:35:58 I (393) heap_init: Initializing. RAM available for dynamic allocation:
2024-04-30 13:35:58 I (400) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
2024-04-30 13:35:58 I (406) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
2024-04-30 13:35:58 I (412) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
2024-04-30 13:35:58 I (419) heap_init: At 3FFD4058 len 0000BFA8 (47 KiB): DRAM
2024-04-30 13:35:58 I (425) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
2024-04-30 13:35:58 I (431) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
2024-04-30 13:35:58 I (437) heap_init: At 40099204 len 00006DFC (27 KiB): IRAM
2024-04-30 13:35:58 I (445) spi_flash: detected chip: generic
2024-04-30 13:35:58 I (448) spi_flash: flash io: dout
2024-04-30 13:35:58 W (453) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
2024-04-30 13:35:58 I (464) coexist: coex firmware version: 77cd7f8
2024-04-30 13:35:58 I (469) main_task: Started on CPU0
2024-04-30 13:35:58 I (479) main_task: Calling app_main()
2024-04-30 13:35:58 W (529) vfs: esp_vfs_register_fd_range is successful for range <54; 64) and VFS ID 2
...
2024-04-30 13:36:03 E (6069) SPP_ACCEPTOR: initBluetooth initialize controller failed: ESP_ERR_INVALID_ARG
2024-04-30 13:36:03
...
2024-04-30 13:36:03 I (6069) main_task: Returned from app_main()
```

yayugu, 2024-04-30:
@ESP-YTGerd Thanks, it worked for the booting up and the initial connection. I'll run the load test for several days and share the result to you!

yayugu, 2024-05-02:
Hi @ESP-YTGerd your new lib is much more stable than before. I haven't seen any errors for 2 days. However, today I observed the error. Here is the dump:

```
2024-05-02 12:07:47 ASSERT_PARAM(-218959118 0), in arch_main.c at line 371
2024-05-02 12:07:47 Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
2024-05-02 12:07:47 Memory dump at 0x40092d48: 00f01d12 00004136 f01d0000
2024-05-02 12:07:47 0x40092d48: btdm_sleep_check_duration at C:/esp-idf/components/bt/controller/esp32/bt.c:915

2024-05-02 12:07:47 Core  0 register dump:
2024-05-02 12:07:47 PC      : 0x40092d4f  PS      : 0x00060d34  A0      : 0x8008b546  A1      : 0x3ffc1dc4
2024-05-02 12:07:47 0x40092d4f: r_assert at C:/esp-idf/components/bt/controller/esp32/bt.c:1848

2024-05-02 12:07:47 A2      : 0x00000000  A3      : 0xf2f2f2f2  A4      : 0x00000000  A5      : 0x3f40d7de
2024-05-02 12:07:47 A6      : 0x00000173  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc1d34
2024-05-02 12:07:47 A10     : 0x00000000  A11     : 0x3ffc1d57  A12     : 0x3ffc1d03  A13     : 0x00000031
2024-05-02 12:07:47 A14     : 0x00000000  A15     : 0x3ffc1d08  SAR     : 0x00000004  EXCCAUSE: 0x00000000
2024-05-02 12:07:47 EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000
2024-05-02 12:07:47
2024-05-02 12:07:47
2024-05-02 12:07:47 Backtrace: 0x40092d4c:0x3ffc1dc4 |<-CORRUPTED
2024-05-02 12:07:48 0x40092d4c: r_assert at C:/esp-idf/components/bt/controller/esp32/bt.c:1847

2024-05-02 12:07:48
2024-05-02 12:07:48
2024-05-02 12:07:48
2024-05-02 12:07:48
2024-05-02 12:07:48 ELF file SHA256: 95087b0a6152646f
2024-05-02 12:07:48
2024-05-02 12:07:48 Rebooting...
```

I hope my experiments help you to solve the issue. Thank you

ESP-YTGerd, 2024-05-06:
Hi, @yayugu 
Thank you for providing the dump information. The issue `ASSERT_PARAM(-218959118 0)` should be unrelated to the problem you previously raised. I added some debug prints, but the changes also involve modifications to both our controller and bluedroid. So could I trouble you to run the test again with the attached lib and patch? Maybe the debug info could help us to solve the newly arising issue. Thanks~
[patch.zip](https://github.com/espressif/esp-idf/files/15217979/patch.zip)

yayugu, 2024-05-06:
Hi @ESP-YTGerd 
I'm keep running the ESP32, but that have no OOMEs since I reported, more than a week. I think the original [patch](https://github.com/espressif/esp-idf/issues/13656#issuecomment-2084488573) solves my issue perfectly and can't wait the new ESP-IDF release with this fix, really appreciate it.

I'll also try with your new patch and report when I found the panic dump again. Thank you so much.

yayugu, 2024-05-06:
Hey, I got the error after running 40min with the new patch. Here is the log:

```
2024-05-06 20:56:07 ets Jul 29 2019 12:21:46
2024-05-06 20:56:07
2024-05-06 20:56:07 rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
2024-05-06 20:56:07 configsip: 0, SPIWP:0xee
2024-05-06 20:56:07 clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
2024-05-06 20:56:07 mode:DOUT, clock div:1
2024-05-06 20:56:07 load:0x3fff0030,len:268
2024-05-06 20:56:07 ho 0 tail 12 room 4
2024-05-06 20:56:07 load:0x40078000,len:12956
2024-05-06 20:56:07 load:0x40080400,len:4
2024-05-06 20:56:07 0x40080400: _init at ??:?

2024-05-06 20:56:07 ho 8 tail 4 room 4
2024-05-06 20:56:07 load:0x40080404,len:2928
2024-05-06 20:56:08 entry 0x4008055c
2024-05-06 20:56:08 I (417) cpu_start: Multicore app
2024-05-06 20:56:08 I (425) cpu_start: Pro cpu start user code
2024-05-06 20:56:08 I (425) cpu_start: cpu freq: 240000000 Hz
2024-05-06 20:56:08 I (425) cpu_start: Application information:
2024-05-06 20:56:08 I (428) cpu_start: Project name:     rws2
2024-05-06 20:56:08 I (433) cpu_start: App version:      3.0.1c
2024-05-06 20:56:08 I (438) cpu_start: Compile time:     May  6 2024 20:54:14
2024-05-06 20:56:08 I (444) cpu_start: ELF file SHA256:  895edf1a2572a036...
2024-05-06 20:56:08 I (450) cpu_start: ESP-IDF:          v5.2.1-1-g766a415056-dirty
2024-05-06 20:56:08 I (456) cpu_start: Min chip rev:     v1.0
2024-05-06 20:56:08 I (461) cpu_start: Max chip rev:     v3.99
2024-05-06 20:56:08 I (466) cpu_start: Chip rev:         v3.0
2024-05-06 20:56:08 I (471) heap_init: Initializing. RAM available for dynamic allocation:
2024-05-06 20:56:08 I (478) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
2024-05-06 20:56:08 I (484) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
2024-05-06 20:56:08 I (490) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
2024-05-06 20:56:08 I (496) heap_init: At 3FFD6D50 len 000092B0 (36 KiB): DRAM
2024-05-06 20:56:08 I (502) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
2024-05-06 20:56:08 I (509) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
2024-05-06 20:56:08 I (515) heap_init: At 4009FDC0 len 00000240 (0 KiB): IRAM
2024-05-06 20:56:08 I (522) spi_flash: detected chip: generic
2024-05-06 20:56:08 I (526) spi_flash: flash io: dout
2024-05-06 20:56:08 W (530) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
2024-05-06 20:56:08 I (541) coexist: coex firmware version: 77cd7f8
2024-05-06 20:56:08 I (546) main_task: Started on CPU0
2024-05-06 20:56:08 I (556) main_task: Calling app_main()
2024-05-06 20:56:08 W (616) vfs: esp_vfs_register_fd_range is successful for range <54; 64) and VFS ID 2
2024-05-06 20:56:08 I (626) wifi:wifi driver task: 3ffd8f84, prio:23, stack:3072, core=1
2024-05-06 20:56:08 I (626) wifi:wifi firmware version: a9f5b59
2024-05-06 20:56:08 I (626) wifi:wifi certification version: v7.0
2024-05-06 20:56:08 I (626) wifi:config NVS flash: disabled
2024-05-06 20:56:08 I (626) wifi:config nano formating: enabled
2024-05-06 20:56:08 I (636) wifi:Init data frame dynamic rx buffer num: 4
2024-05-06 20:56:08 I (636) wifi:Init static rx mgmt buffer num: 5
2024-05-06 20:56:08 I (636) wifi:Init management short buffer num: 6
2024-05-06 20:56:08 I (646) wifi:Init static tx buffer num: 4
2024-05-06 20:56:08 I (646) wifi:Init static rx buffer size: 1600
2024-05-06 20:56:08 I (656) wifi:Init static rx buffer num: 2
2024-05-06 20:56:08 I (656) wifi:Init dynamic rx buffer num: 4
2024-05-06 20:56:08 I (666) wifi_init: rx ba win: 2
2024-05-06 20:56:08 I (666) wifi_init: tcpip mbox: 32
2024-05-06 20:56:08 I (666) wifi_init: udp mbox: 6
2024-05-06 20:56:08 I (676) wifi_init: tcp mbox: 6
2024-05-06 20:56:08 I (676) wifi_init: tcp tx win: 2880
2024-05-06 20:56:08 I (676) wifi_init: tcp rx win: 2880
2024-05-06 20:56:08 I (686) wifi_init: tcp mss: 1440
2024-05-06 20:56:08 I (686) wifi_init: WiFi IRAM OP enabled
2024-05-06 20:56:08 I (696) wifi_init: WiFi RX IRAM OP enabled
...
2024-05-06 20:56:13 I (6256) BTDM_INIT: BT controller compile version [ecf14fe]
2024-05-06 20:56:13 ##C##: mem pool: 524 + 2060 + 1036
2024-05-06 20:56:13 I (6256) BTDM_INIT: Bluetooth MAC: 0c:8b:95:45:32:02
[... 283 lines cut ...]
2024-05-06 22:37:01 Memory dump at 0x4009300c: 00f01d12 00004136 f01d0000
2024-05-06 22:37:01 0x4009300c: btdm_sleep_check_duration at C:/esp-idf/components/bt/controller/esp32/bt.c:922

2024-05-06 22:37:01 Core  0 register dump:
2024-05-06 22:37:01 PC      : 0x40093013  PS      : 0x00060e34  A0      : 0x8008b5da  A1      : 0x3ffc46e4
2024-05-06 22:37:01 0x40093013: r_assert at C:/esp-idf/components/bt/controller/esp32/bt.c:1857

2024-05-06 22:37:01 A2      : 0x00000000  A3      : 0xf2f2f2f2  A4      : 0x00000000  A5      : 0x3f40ab48
2024-05-06 22:37:01 A6      : 0x00000175  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc4654
2024-05-06 22:37:01 A10     : 0x00000000  A11     : 0x3ffc4677  A12     : 0x3ffc4623  A13     : 0x00000033
2024-05-06 22:37:01 A14     : 0x00000000  A15     : 0x3ffc4628  SAR     : 0x00000004  EXCCAUSE: 0x00000000
2024-05-06 22:37:01 EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000
2024-05-06 22:37:01
2024-05-06 22:37:01
2024-05-06 22:37:01 Backtrace: 0x40093010:0x3ffc46e4 |<-CORRUPTED
2024-05-06 22:37:01 0x40093010: r_assert at C:/esp-idf/components/bt/controller/esp32/bt.c:1856

2024-05-06 22:37:01
2024-05-06 22:37:01
2024-05-06 22:37:01
2024-05-06 22:37:01
2024-05-06 22:37:01 ELF file SHA256: 895edf1a2572a036
2024-05-06 22:37:01
2024-05-06 22:37:01 Rebooting...
```

yayugu, 2024-05-07:
Hi @ESP-YTGerd, I tried the newest `libbtdm_app.a` (with the debugging git patch) and got the new error after running 7 hours.

```
(log head is missing)
2024-05-07 16:38:07 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 16:38:08 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 16:38:09 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 16:38:10 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 16:38:11 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 16:38:12 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 16:38:13 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 16:38:15 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 16:38:16 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 16:38:17 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 16:38:18 lc_pwr_incr_ind_handler_hack:lc state 1
(stable for 7 hours)
2024-05-07 23:46:07 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 23:46:08 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 23:46:09 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 23:46:10 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 23:46:10 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 23:46:11 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:12 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:13 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:14 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:15 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:16 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:17 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:18 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:19 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 23:46:20 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:22 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:23 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:24 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 23:46:25 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:26 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:27 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:28 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 23:46:29 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:30 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:31 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:32 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 23:46:33 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:34 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:35 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:36 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:37 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:38 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:39 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:40 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:41 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:42 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:43 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:44 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:45 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 23:46:46 lc_pwr_incr_ind_handler_hack:lc state 1
2024-05-07 23:46:47 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 23:46:48 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 23:46:48 E (25745629) msgReader: tcpReader::native_read recv failed. errno=11 (WiFi/TCP task shows the error since this message, 10mins before crash)
2024-05-07 23:46:49 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 23:46:50 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 23:46:51 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 23:46:52 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 23:46:53 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 23:46:54 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 23:46:55 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 23:46:56 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 23:46:57 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 23:46:58 lc_pwr_max_ind_handler_hack:lc state 1
2024-05-07 23:46:58 E (25755729) msgReader: tcpReader::native_read recv failed. errno=11
[... 678 lines cut ...]
2024-05-07 23:56:32 I (587) wifi:config NVS flash: disabled
2024-05-07 23:56:32 I (587) wifi:config nano formating: enabled
2024-05-07 23:56:32 I (597) wifi:Init data frame dynamic rx buffer num: 4
2024-05-07 23:56:32 I (597) wifi:Init static rx mgmt buffer num: 5
2024-05-07 23:56:32 I (607) wifi:Init management short buffer num: 6
2024-05-07 23:56:32 I (607) wifi:Init static tx buffer num: 4
2024-05-07 23:56:32 I (607) wifi:Init static rx buffer size: 1600
2024-05-07 23:56:32 I (617) wifi:Init static rx buffer num: 2
2024-05-07 23:56:32 I (617) wifi:Init dynamic rx buffer num: 4
2024-05-07 23:56:32 I (627) wifi_init: rx ba win: 2
2024-05-07 23:56:32 I (627) wifi_init: tcpip mbox: 32
2024-05-07 23:56:32 I (627) wifi_init: udp mbox: 6
2024-05-07 23:56:32 I (637) wifi_init: tcp mbox: 6
2024-05-07 23:56:32 I (637) wifi_init: tcp tx win: 2880
2024-05-07 23:56:32 I (647) wifi_init: tcp rx win: 2880
2024-05-07 23:56:32 I (647) wifi_init: tcp mss: 1440
2024-05-07 23:56:32 I (647) wifi_init: WiFi IRAM OP enabled
2024-05-07 23:56:32 I (657) wifi_init: WiFi RX IRAM OP enabled
2024-05-07 23:56:32 I (1077) Main: Bluetooth name RWS.03
2024-05-07 23:56:37 I (6177) BTDM_INIT: BT controller compile version [586d370]
2024-05-07 23:56:37 ##C##: mem pool: 524 + 2060 + 1036
2024-05-07 23:56:37 I (6177) BTDM_INIT: Bluetooth MAC: 0c:8b:95:45:32:02
2024-05-07 23:56:37 I (6187) phy_init: phy_version 4791,2c4672b,Dec 20 2023,16:06:06
2024-05-07 23:56:37 I (6707) main_task: Returned from app_main()
```

ESP-YTGerd, 2024-05-08:
Hi, @yayugu 
What's the peer device of spp acceptor?  It seems that the tx power of peer device is too low and we have been requesting to increase it, while their tx power has remained at a relatively low level.

yayugu, 2024-05-08:
I use [Redmi Note 11](https://www.mi.com/jp/product/redmi-note-11/specs), Android 13 and MIUI Global 14.0.5. 
That's interesting, I will also try to change the phone and ESP32 position (the ESP32 should not crash even the peer device has the issue though😅)

ESP-YTGerd, 2024-05-08:
> I use [Redmi Note 11](https://www.mi.com/jp/product/redmi-note-11/specs), Android 13 and MIUI Global 14.0.5, keep connecting to power charger, not the low battery mode. That's interesting, I will also try to change the phone and ESP32 position (the ESP32 should not crash even the peer device has the issue though😅)
> 
> Update: I put the phone and ESP32 side-by-side and the messages below disappeared
> 
> ```
> lc_pwr_max_ind_handler_hack:lc state 1
> lc_pwr_incr_ind_handler_hack:lc state 1
> ```

hahaha, it's true, we will resolve it and are also looking forward to the results of long-term testing after placing the two devices together.

closed 2024-05-16 by espressif-bot