espressif/esp-idf #13657: How can I uninstall outdated SDK version?

JunOllyLi, 2024-04-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.1

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

So I used to use 5.0.1 to build my project. And recently I upgraded to use 5.2.1
How can I cleanly remove the 5.0.1 SDK from my hard drive? As a matter of fact I still have old 4.x on my hard drive...

### What is the actual behavior?

I did not find a place to uninstall it.

### Steps to reproduce.

1. try  to uninstall certain version by right click the "... cmd" icon
2. the "uninstall" leads me to "installed appa" in Windows
3. but I did not see any IDF SDK there
4.

dobairoland, 2024-04-22:
The guide on uninstalling ESP-IDF is linked from the bottom of this page: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#uninstall-esp-idf

dobairoland, 2024-04-22:
If you go to "Add or remove programs" and search for ESP-IDF then nothing shows up in the list?

JunOllyLi, 2024-04-22:
> If you go to "Add or remove programs" and search for ESP-IDF then nothing shows up in the list?

No it did not.

JunOllyLi, 2024-04-22:
> The guide on uninstalling ESP-IDF is linked from the bottom of this page: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#uninstall-esp-idf

This is not really what I was looking for. I still want to use ESP IDF. But I am migrating from 4.4 to 5.0 then to 5.2. Now I just want to reclaim the gigabytes of storage space from the old 4.4/5.0 frameworks.
But now I guess it does not matter anymore. I tried to use that instruction and uninstalled the ESP-IDF. Now I need to re-install it again.

And the worst thing is that the different versions of the frameworks were still on my HD! And the command shell icon for 4.4/5.0 are still in my start menu!

jakub-kocka, 2024-04-25:
Hello @JunOllyLi,
I have tested your issue on plain Windows installation and you are right that the shortcut icons after the complete uninstallation remain in the start menu, thank you for pointing this out, we will try to fix this.

However, the framework installed with the IDF installer was in my case uninstalled successfully.

As said, IDF is the framework, not the application. So what the IDF installer does is that it automates the process of downloading the framework, including adding all the necessary tools, environmental PATHs, and so on. Uninstallation, as expected should clean all these files and modifications.

Currently, there is no option to uninstall just one or more frameworks with the IDF installer, uninstallation cleans everything. But if you want to get rid of a version of the framework, you can simply delete the directory with the unwanted version in the `frameworks` path (`<YOUR_PATH_OF_IDF_INSTALLATION>/frameworks`).
Also, with this approach desktop and start menu icons will need to be removed (in `All apps` search for the command-lines consoles with the unwanted version of IDF -> _right mouse button_ -> `More` -> `Open file location` and here you can remove the  start menu icons)

JunOllyLi, 2024-04-27:
> However, the framework installed with the IDF installer was in my case uninstalled successfully.
>
> As said, IDF is the framework, not the application. So what the IDF installer does is that it [...]

I think it may has something to do with you have more than one way of installing it? Maybe I installed 4.4 one way, say the online installer; but then installed 5.0.1 with the full downloaded local installer? That may cause Windows to get confused though.

But I have to say this is going to be a problem when the framework evolves. Normally how I would migrate from one framework to another would be I would have them both on the hard drive for a while so I can do comparison between them while I migrate. Then once I am done migrating I would try to delete the old one. So you probably want to make it easier to do.

closed 2024-04-27 by JunOllyLi

dobairoland, 2024-05-17:
I'm just pointing out for future readers that the issue about skipping some of the files during uninstall has been fixed by https://github.com/espressif/idf-installer/pull/241.