espressif/esp-idf #13661: ESP32 Bluetooth fails to Read Remote Extended Features when using btstack

DerEchteJoghurt, 2024-04-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-3225-g5a40bb8746

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

When using [btstack](https://github.com/bluekitchen/btstack), the peripheral examples do not pair with a Google Pixel 6a on Android 14, due to the controller implementation failing to properly respond to the Read Remote Extended Feature command.

See the following issue for further details: https://github.com/bluekitchen/btstack/issues/586

### What is the actual behavior?

The VHCI implementation should properly respond to the Read Remote Extended Feature command when using btstack.

### Steps to reproduce.

1. Integrate the develop branch of BTstack into esp-idf.
1. Enable hci logging and build the hid_mouse_demo.
1. Build, flash and monitor the demo.
1. Pair to the device.
1. After pairing the phone is stuck on "Connecting..." and the logs show not properly responding to the Read Remote Extended Feature command (see https://github.com/bluekitchen/btstack/issues/586#issuecomment-2057401266).

### Debug Logs.

```plain
See https://github.com/bluekitchen/btstack/issues/586#issuecomment-2057401266
```

zhp0406, 2024-04-24:
The ESP32 does not support BLE5.0. The Extended Features refer to BLE50 features.

zhp0406, 2024-04-24:
@DerEchteJoghurt 
If you want to use BLE5.0 features, please consider using other chips in the ESP series, such as ESP32-C2, ESP32-C3, ESP32-C6, ESP32-H2, ESP32-S3, etc.

DerEchteJoghurt, 2024-04-24:
Thank you for the quick response, shouldn't this be represented in the Supported Commands HCI Configuration Parameter?
Octet 2, Bit 6 is currently set here, which would indicate that the command is suppoted.

mringwal, 2024-04-26:
@zhp0406 In the log here, https://github.com/bluekitchen/btstack/issues/586#issuecomment-2057401266, the HCI Read Remote Extended Features for a BR/EDR ACL Connection is pending. If the command would not be supported by the Controller, the Controller should return a HCI Command Complete event with status UNKNOWN HCI COMMAND (0x01) or something similar.

ESP-YTGerd, 2024-04-28:
Hi, @DerEchteJoghurt 
Can try with the attached new lib? in which I have made some changes and added debug info. You can replace the old lib with new lib in path `controller/lib_esp32/esp32/libbtdm_app.a`.  It's also suggested deleting  the old `build` via `rm -rf build` in project path before re-building your project with the new lib.
[libbtdm_app.zip](https://github.com/espressif/esp-idf/files/15140610/libbtdm_app.zip)

DerEchteJoghurt, 2024-04-29:
> Hi, @DerEchteJoghurt Can try with the attached new lib? in which I have made some changes and added debug info. You can replace the old lib with new lib in path [...]

With the new lib, btstack receives a `Read Remote Extended Features Complete (0x23)` event and everything works as expected.
Thanks for the quick patch!

ESP-YTGerd, 2024-04-30:
Hi @DerEchteJoghurt
If the BR/EDR secure connection doesn't need to be opened, it's advisable to disable this feature. In fact, the issue you encountered arises from enabling AES-CCM encryption. We encounter some problems when sending ACL-U packets when AES-CCM is on.

DerEchteJoghurt, 2024-05-02:
> Hi @DerEchteJoghurt If the BR/EDR secure connection doesn't need to be opened, it's advisable to disable this feature. In fact, the issue you encountered arises from enabling AES-CCM encryption. We encounter some problems when sending ACL-U packets when AES-CCM is on.

Is this issue fixed with the build you uploaded here or do you still recommend disabling secure connections?

mringwal, 2024-05-05:
@DerEchteJoghurt Could you upload a PacketLog file with the new libbtdm_app? We can check if Secure Connections are used.

mringwal, 2024-05-05:
> If the BR/EDR secure connection doesn't need to be opened, it's advisable to disable this feature. In fact, the issue you encountered arises from enabling AES-CCM encryption. We encounter some problems when sending ACL-U packets when AES-CCM is on.

Thanks for looking into this. BR/EDR Secure Connection is an optional feature of Bluetooth Controllers. While it's not clear to me that the AES-CCM encryption is better than the one for legacy connection, it replaces a custom crypto algorithm by a modern/standard one, which is a good thing.

If the ESP32 supports BR/EDR Secure Connection, then the implementation should be improved to work in all supported configurations.

mringwal, 2024-05-06:
@ESP-YTGerd Thanks for the clarification!

Indastri, 2024-07-02:
Closing the issue as solved. If you encounter any more problems feel free to reopen this or create a new issue.

Cheers!

closed 2024-07-02 by Indastri

mringwal, 2024-07-02:
Hi @Indastri Has the 'new lib' from this comment https://github.com/espressif/esp-idf/issues/13661#issuecomment-2081302134 been integrated into the latest Release(s)?

ESP-YTGerd, 2024-07-16:
Hi, @mringwal 
Yes, the fix hasn't been merged into the release branch yet because we cannot fully guarantee that it will resolve the issue without introducing new problems. And we are still conducting further internal testing and evaluation.

ESP-YTGerd, 2024-07-29:
Hi, @mringwal 
When Secure Connection is enabled, the packets we send include a 32-bit MIC in the packet payload, making them longer than normal packets. Currently, the ESP32 hardware sometimes has issues scheduling the next transmission or reception when sending DM1 packets with a MIC, which cannot be resolved or have a work around in software. This issue also occurs when sending ACL-C packets, so we can't prohibit sending DM1 packets in our TX packet types. Unfortunately, we cannot resolve this issue in software for now and it's advisable to disable this feature.

ESP-YTGerd, 2025-05-15:
Hi, @mringwal 
We merged a workaround for this issue in commit [6d007b](https://github.com/espressif/esp32-bt-lib/commit/6d007b7167146b31256b29fed31e04dc46989885)

It should be noted that this workaround does not fully resolve all issues encountered when `Secure Connection` is enabled. Therefore, we do not recommend enabling this feature during Bluedroid initialization in our ESP-IDF.

However, if the customer is using a different host stack in combination with the ESP32 controller, this workaround should still address most of the issues — including the one reported in this thread.

mringwal, 2025-05-25:
Hi @ESP-YTGerd Thanks for the update. 

If I understand you correctly, the reported issue should be resolved with Secure Connection enabled, but there are other similar issues that are not fully resolved. In this case, we leave Secure Connections disabled by default. 

The main benefit I can see (besides a theoretically better encryption), is the support for Cross-Transport Key-Derivation (CTKD), although in most cases where mobile phones are involved, the initial pairing generally happens via LE Secure Connections.

ESP-YTGerd, 2025-05-26:
This issue has been resolved through commit https://github.com/espressif/esp32-bt-lib/commit/6d007b7167146b31256b29fed31e04dc46989885 and has been merged into all release branches.