espressif/esp-idf #13665: gdbgui is not starting - unrecognized arguments: --gdb-args

MASE98, 2024-04-22:
**Preview checklist:**
- I have read the documentation ESP-IDF Programming Guide and the issue is not addressed there.

-  I have updated my IDF branch ESP32-C3 (master or release) to the latest version.

**IDF version:**
ESP-IDF v5.3-dev-3225-g5a40bb8746

**Espressif SoC revision**
ESP32-C3 (FN4)  (chip revision v0.4)

**Operating System used**
Linux

**Development Kit**
ESP32-C3-DevKitC-02 v1.1

**Power supply used**
USB

**Pyhton version**
python 3.10.12

**gdbgui version**
gdbgui-0.15.2.0 by idf.py gdbgui

**What is the expected behavior?**
Start gdbgui in the browser.

**What is the actual behavior**
gdbgui is not starting.

```
Executing action: gdbgui
Running:  ['gdbgui', '-g', 'riscv32-esp-elf-gdb', '--gdb-args', '"-x=/home/marco/esp/esp-idf/examples/get-started/hello_world/build/gdbinit/gdbinit"']
gdbgui started as a background task 42895
Executing action: post_debug
usage: gdbgui [-h] [-g GDB_CMD] [-p PORT] [--host HOST] [-r]
              [--auth-file AUTH_FILE] [--user USER] [--password PASSWORD]
              [--key KEY] [--cert CERT] [--remap-sources REMAP_SOURCES]
              [--project PROJECT] [-v] [-n] [-b BROWSER] [--debug]
              [--args ...]
              [debug_program]
gdbgui: error: unrecognized arguments: --gdb-args
"gdbgui" exited with 2
```

dobairoland, 2024-04-29:
@MASE98 Do you have any update on this?

MASE98, 2024-04-29:
I modified the **requirements.gdbgui.txt** file to install gdbgui version 0.13.2.0, then I re-run the ./install.sh --enable-gdbgui.

`gdbgui==0.13.2.0; sys_platform != 'win32'`
`gdbgui==0.13.2.0; sys_platform == 'win32' and python_version < "3.11"`

- But when I ejecute `idf.py gdbgui`, I get the following message: 
- Is that message because I have the latest version of IDF installed and it does not go with version 0.13.2.0? 

```
Executing action: gdbgui
Running:  ['gdbgui', '-g', 'riscv32-esp-elf-gdb', '--gdb-args', '"-x=/home/marco/esp/esp-idf/examples/get-started/hello_world/build/gdbinit/gdbinit"']
gdbgui started as a background task 57268
Executing action: post_debug
Traceback (most recent call last):
  File "/home/marco/.espressif/python_env/idf5.3_py3.10_env/bin/gdbgui", line 5, in <module>
    from gdbgui.backend import main
  File "/home/marco/.espressif/python_env/idf5.3_py3.10_env/lib/python3.10/site-packages/gdbgui/backend.py", line 25, in <module>
    from flask import (
  File "/home/marco/.espressif/python_env/idf5.3_py3.10_env/lib/python3.10/site-packages/flask/__init__.py", line 19, in <module>
    from jinja2 import Markup, escape
ImportError: cannot import name 'Markup' from 'jinja2' (/home/marco/.espressif/python_env/idf5.3_py3.10_env/lib/python3.10/site-packages/jinja2/__init__.py)
"gdbgui" exited with 1
```

dobairoland, 2024-04-29:
> I modified the requirements.gdbgui.txt file to install gdbgui version 0.13.2.0,

No you shouldn't have to do that. If you run the install script, then it will download `~/.espressif/espidf.constraints.v5.3.txt` which sets the version requirements not just for gdbgui but for all of its dependencies. This is important because otherwise it won't work as you can see.

dobairoland, 2024-05-10:
I'm closing this because this seems to caused by hacking into ESP-IDF installation and not using it in the supported way.

If you've got into this situation by mistake then I still can help to get back to the supported way of using ESP-IDF.

closed 2024-05-10 by dobairoland