espressif/esp-idf #13668: Fatfs partition Flash Encryption.

hussainhadi673, 2024-04-22:
I have this partition table:

`# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x603F00,
nvs_key,  data, nvs_keys,0x64F000,0x1000,
storage,   data, fatfs,  0x650000,0x990000,encrypt
coredump, data, coredump,0xFE0000,0x10000,`

Now I have turned on the flash encryption (Development Mode) and upon the first boot my app0 and storage is encrypted. 
Now my question is that at the time of first boot my fatfs is empty, and in my software I store some data in Fatfs on run-time, how will this new data stored on fatfs will get encrypted ?

rrtandler, 2024-05-06:
Hi hussainhadi673,
The fatfs component uses esp_partition component to read / write data to the "storage" parttition. These r/w functions are observing the encrypted flag of the partition table and if ESP32 flash encryption is active, the data is encrypted during the write operation and decrypted during read operations. This happens transparently for the user of fatfs component API.

rrtandler, 2024-06-04:
Hi @hussainhadi673 ,

Have you had a chance to evaluate the answer I have provided or is there still anything more to be clarified ?

hussainhadi673, 2024-06-10:
So that means, Every time I store data on Flash (fatfs parition) after enabling flash encrytion, the data will be encrypted first and then stored.  And similarly when I read the data from fatfs in my software, it will be decrypted first and then I can use it. 
Does SPIFFS follow same mechanism ?

hussainhadi673, 2024-06-11:
Thankyou for the help. It is clear to me now.

closed 2024-07-04 by espressif-bot