espressif/esp-idf #13670: TCP: Connection to server failed. Error code: 118

arvind55555, 2024-04-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi all,
I am currently working on a SPI code that receives data from a transmitter, stores the data in a buffer, and then sends the data to a TCP socket via Wi-Fi. Additionally, I am using dual-core functionality, where one core takes care of SPI transactions and another core takes care of TCP transactions. The problem is that the code without dual-core setup works absolutely fine, but when I implement it with dual-core, I am facing the error "TCP: Connection to server failed. Error code: 118". Below is my code

```
#include <errno.h>
#include <stddef.h>
#include <stdint.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/queue.h"
#include "freertos/semphr.h"
#include "freertos/task.h"

#include "lwip/dns.h"
#include "lwip/igmp.h"
#include "lwip/netdb.h"
#include "lwip/sockets.h"

#include "driver/gpio.h"
#include "driver/ledc.h"
#include "driver/spi_slave.h"
#include "esp_event.h"
#include "esp_log.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "nvs_flash.h"
#include "soc/rtc_periph.h"
#include "spi_flash_mmap.h"

#define core_0 0
#define core_1 1

// WiFi configuration
const char *ssid = "#";
const char *pass = "#";

// Server configuration (replace with your server details)
#define SERVER_IP "#"
#define SERVER_PORT #

#define GPIO_MOSI 23
#define GPIO_MISO 19
#define GPIO_SCLK 18
#define GPIO_CS 5

#define BUFFER_SIZE 16384
#define HALF_BUFFER_SIZE (BUFFER_SIZE / 2)

int sock = -1;

uint32_t buffer1[BUFFER_SIZE];
uint32_t buffer2[BUFFER_SIZE];

volatile uint32_t *currentTxBuffer;
volatile uint32_t *currentRxBuffer;

SemaphoreHandle_t dataReadySemaphore;

// Function to initialize the socket
void initialize_socket() {
  sock = socket(AF_INET, SOCK_STREAM, 0);
  if (sock < 0) {
    ESP_LOGE("TCP", "Unable to create socket");
    return;
  }

  struct sockaddr_in server_address;
  memset(&server_address, 0, sizeof(server_address));
  server_address.sin_family = AF_INET;
  server_address.sin_port = htons(SERVER_PORT);
  inet_pton(AF_INET, SERVER_IP, &(server_address.sin_addr));

  if (connect(sock, (struct sockaddr *)&server_address,
              sizeof(server_address)) < 0) {
    ESP_LOGE("TCP", "Connection to server failed. Error code: %d", errno);
    close(sock);
    sock = -1;
    return;
  }

  ESP_LOGI("TCP", "Socket initialized successfully");
}

// Function to send data to the socket
void send_data_to_socket(const char *data, size_t data_length) {
  if (sock < 0) {
    // Socket not initialized, attempt to initialize
    ESP_LOGI("TCP", "Socket not initialized, attempting to initialize...");
    initialize_socket();
    if (sock < 0) {
      ESP_LOGE("TCP", "Socket initialization failed, cannot send data");
      return;
    }
  }

  // Send data to the server as binary
  int sent_bytes = send(sock, data, data_length, 0);
  if (sent_bytes < 0) {
    ESP_LOGE("TCP", "Failed to send data to server. Error code: %d", errno);
  } else {
    ESP_LOGI("TCP", "Sent %d bytes to server", sent_bytes);
  }
}

// Function to close the socket
void close_socket() {
  if (sock >= 0) {
    close(sock);
    sock = -1;
    ESP_LOGI("TCP", "Socket closed");
  }
}

// WiFi event handler
static void wifi_event_handler(void *event_handler_arg,
                               esp_event_base_t event_base, int32_t event_id,
                               void *event_data) {
  if (event_id == WIFI_EVENT_STA_START) {

  } else if (event_id == WIFI_EVENT_STA_CONNECTED) {

  } else if (event_id == WIFI_EVENT_STA_DISCONNECTED) {

    esp_wifi_connect();
  } else if (event_id == IP_EVENT_STA_GOT_IP) {
  }
}

// WiFi connection setup
void wifi_connection() {
  esp_netif_init();
  esp_event_loop_create_default();
  esp_netif_create_default_wifi_sta();
  wifi_init_config_t wifi_initiation = WIFI_INIT_CONFIG_DEFAULT();
  esp_wifi_init(&wifi_initiation);
  esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, wifi_event_handler,
                             NULL);
  esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, wifi_event_handler,
                             NULL);
  wifi_config_t wifi_configuration = {.sta = {
                                          .ssid = "#",
                                          .password = "#",
                                      }};
  strcpy((char *)wifi_configuration.sta.ssid, ssid);
  strcpy((char *)wifi_configuration.sta.password, pass);
  esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_configuration);
  esp_wifi_start();
  esp_wifi_set_mode(WIFI_MODE_STA);
  esp_wifi_connect();
  printf("wifi_init_softap finished. SSID:%s  password:%s", ssid, pass);
}

void hello_task_core_0(void *pvParameter) {
  spi_bus_config_t buscfg = {
      .mosi_io_num = GPIO_MOSI,
      .miso_io_num = GPIO_MISO,
      .sclk_io_num = GPIO_SCLK,
      .quadwp_io_num = -1,
      .quadhd_io_num = -1,
  };

  spi_slave_interface_config_t slvcfg = {
      .mode = 0,
      .spics_io_num = GPIO_CS,
      .queue_size = 3,
      .flags = 0,
  };

  spi_slave_initialize(HSPI_HOST, &buscfg, &slvcfg, SPI_DMA_CH_AUTO);

  spi_slave_transaction_t t;
  memset(&t, 0, sizeof(t));
  t.length = HALF_BUFFER_SIZE;
  t.rx_buffer = (void *)currentRxBuffer;
  t.tx_buffer = (void *)currentTxBuffer;
  while (1) {
    xSemaphoreTake(dataReadySemaphore, portMAX_DELAY);
    spi_slave_transmit(HSPI_HOST, &t, portMAX_DELAY);
    xSemaphoreGive(dataReadySemaphore); // Signal that data is ready
    vTaskDelay(pdMS_TO_TICKS(1));
  }
}

void hello_task_core_1(void *pvParameter) {
  while (1) {
    xSemaphoreTake(dataReadySemaphore, portMAX_DELAY);
    send_data_to_socket((char *)currentTxBuffer, HALF_BUFFER_SIZE);
    xSemaphoreGive(dataReadySemaphore);

    volatile uint32_t *temp = currentTxBuffer;
    currentTxBuffer = currentRxBuffer;
    currentRxBuffer = temp;
  }
}

void app_main() {
  nvs_flash_init();
  wifi_connection();
  initialize_socket();

  // Create semaphore
  dataReadySemaphore = xSemaphoreCreateBinary();
  if (dataReadySemaphore == NULL) {
    ESP_LOGE("APP_MAIN", "Failed to create semaphore");
    return;
  }

  xTaskCreate(&hello_task_core_0, "core0_task", 8192, NULL, 2, NULL);
  xTaskCreate(&hello_task_core_1, "core1_task", 8192, NULL, 2, NULL);

  close_socket();
}

```
I've encountered an issue with my ESP32 project, and I've created a GitHub issue for it [here](https://github.com/espressif/esp-idf/issues/13658). The problem seems to be related to how I'm using semaphores. Although I tried to address the issue based on the comments, the problem persists. I've also attached a screenshot of my terminal for reference. Can anyone help me out with this? I suspect it might be a synchronization issue, but I'm having trouble pinpointing it. Any assistance would be greatly appreciated
[image]

hansw123, 2024-07-08:
@arvind55555 
hi,from the log ,you create tcp  connect before wifi connected,so you must make sure do tcp connecting after wifi connected

Sherry616, 2024-08-09:
Thanks for reporting, we will close this issue. Feel free to reopen if have more updates. 
Thanks for using our Espressif product!

closed 2024-08-09 by Sherry616

Tanusmitadash, 2024-11-23:
[image]
 how to resolve this error.

hansw123, 2024-11-28:
@Tanusmitadash 
from log , your code created tcp socket when sta ininted, but this should after wifi connected, you can refer the station example
https://github.com/espressif/esp-idf/blob/master/examples/wifi/getting_started/station/main/station_example_main.c and 
https://github.com/espressif/esp-idf/blob/master/examples/protocols/sockets/tcp_client/main/tcp_client_main.c