espressif/esp-idf #13671: classic bluetooth hid device doesn't reconnect

aromprg, 2024-04-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello!
I'm use classic Bluetooth with the [esp-idf\examples\bluetooth\bluedroid\classic_bt\bt_hid_mouse_device](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/classic_bt/bt_hid_mouse_device/main/main.c). In my case after successfully pairing with android-phone, esp_mouse has saved the host BD_ADDR in nvs, and then after break Bluetooth connection (phone disable bt), esp power off/on, it needs to reconnect to the host automatically. As found on https://github.com/espressif/esp-idf/issues/10107 I should call esp_bt_hid_device_connect() for this (I'm create reconnect task with 3000ms delay). This work fine with IDF v5.0.2, but now I upgrade to v5.2.1. [After this commit](https://github.com/espressif/esp-idf/commit/bc1b01a9669b39f7704f59e56969fbb43dd3f2f6) esp_mouse will never be able to connect to the host after the first failed attempt.
```
I (3260) app_main: init end
I (3260) main_task: Returned from app_main()
E (6240) BT_BTC: btc_hd_connect: busy now, status:3, try later!
E (6240) esp_bt_hidd_cb: open failed!
W (8390) BT_HIDD: hidd_l2cif_connect_cfm: connection failed, now disconnect
W (8390) BT_HIDD: hidd_conn_disconnect: already disconnected
W (8390) BT_HCI: hcif conn complete: hdl 0xfff, st 0x4
I (8400) esp_bt_gap_cb: event: 16
E (9240) BT_BTC: btc_hd_connect: busy now, status:3, try later!
E (9240) esp_bt_hidd_cb: open failed!
E (12240) BT_BTC: btc_hd_connect: busy now, status:3, try later!
E (12240) esp_bt_hidd_cb: open failed!
E (15240) BT_BTC: btc_hd_connect: busy now, status:3, try later!
E (15240) esp_bt_hidd_cb: open failed!
```
In [components/bt/host/bluedroid/btc/profile/std/hid/btc_hd.c](https://github.com/espressif/esp-idf/blob/master/components/bt/host/bluedroid/btc/profile/std/hid/btc_hd.c) btc_hd_cb.status remains in the BTC_HD_CONNECTING state forever, without being reset if the connection fails.

boblane1, 2024-04-23:
@aromprg Thanks for your reporting. Please try the following patch to see if the issue has been solved.

[test.zip](https://github.com/espressif/esp-idf/files/15076826/test.zip)

aromprg, 2024-04-23:
This patch fixes the reconnection issue. Everything works as expected. Thank you!

closed 2024-05-16 by espressif-bot