espressif/esp-idf #13673: BLE GATT Service and Characteristics ESP_UUID_LEN_128 issues

rajeshktym, 2024-04-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

1.7.1 Espressif IDF gatt_security_client demo

### Espressif SoC revision.

esp32

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

esp32

### Power Supply used.

External 3.3V

### What is the expected behavior?

before complete I (12337) GATTC_DEMO: ESP_GATTC_SEARCH_CMPL_EVT
expecting esp_ble_gattc_search_service(gattc_if, param->cfg_mtu.conn_id, &remote_filter_service_uuid);
    case ESP_GATTC_SEARCH_RES_EVT: {
        ESP_LOGI(GATTC_TAG, "SEARCH RES: conn_id = %x is primary service %d", p_data->search_res.conn_id, p_data->search_res.is_primary);
        ESP_LOGI(GATTC_TAG, "start handle %d end handle %d current handle value %d", p_data->search_res.start_handle, p_data->search_res.end_handle, p_data->search_res.srvc_id.inst_id);
        

### What is the actual behavior?

I (11547) GATTC_DEMO: 15 9b b3 48 a1 3d 
I (11547) GATTC_DEMO: searched Adv Data Len 31, Scan Response Len 0
I (11557) GATTC_DEMO: searched Device Name Len 0
I (11557) GATTC_DEMO:
 00:18:80:7c:27:2b
I (11617) GATTC_DEMO: 00 18 80 7c 27 2b 
I (11617) GATTC_DEMO: searched Adv Data Len 7, Scan Response Len 21
I (11617) GATTC_DEMO: searched Device Name Len 19
I (11627) GATTC_DEMO: MAXREFDES104/HSP
I (11627) GATTC_DEMO: 3.0
I (11627) GATTC_DEMO:

I (11657) GATTC_DEMO: searched device MAXREFDES104/HSP3.0

I (11667) GATTC_DEMO: connect to the remote device.
I (11677) GATTC_DEMO: stop scan successfully
I (11687) GATTC_DEMO: 15 9b b3 48 a1 3d
I (11687) GATTC_DEMO: searched Adv Data Len 31, Scan Response Len 0
I (11687) GATTC_DEMO: searched Device Name Len 0
I (11697) GATTC_DEMO:

I (11887) GATTC_DEMO: ESP_GATTC_CONNECT_EVT conn_id 0, if 3
I (11887) GATTC_DEMO: REMOTE BDA:
I (11887) GATTC_DEMO: 00 18 80 7c 27 2b
I (11887) GATTC_DEMO: open success
I (12327) GATTC_DEMO: discover service complete conn_id 0
I (12337) GATTC_DEMO: Get service information from remote device
I (12337) GATTC_DEMO: ESP_GATTC_SEARCH_CMPL_EVT
I (12357) GATTC_DEMO: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 241, conn_id 0

### Steps to reproduce.

#define REMOTE_SERVICE_UUID        {0x9E,0xCA,0xDC,0x24,0x0E,0xE5,0xA9,0xED,0x93,0xF3,0xA3,0xB5,0x00,0x00,0x40,0x6E}
#define REMOTE_NOTIFY_CHAR_UUID    {0x9E,0xCA,0xDC,0x24,0x0E,0xE5,0xA9,0xED,0x93,0xF3,0xA3,0xB5,0x01,0x00,0x40,0x6E}

using this UUID basis of below hardware details

BLE GATT Service and Characteristics
This document covers the main service consisting of two characteristics. The notify data characteristics is used during data streaming, where the device is
actively collecting data and transmitting raw data back to the master. The read/write configuration characteristics is used to setup the system and to
perform low level read/write operations.
UUID Type Description
6E400000-B5A3-F393-E0A9-E50E24DCCA9E Service Service Declaration
6E400001-B5A3-F393-E0A9-E50E24DCCA9E Characteristic Notify Data Characteristic
6E400002-B5A3-F393-E0A9-E50E24DCCA9E Characteristic Read/Write Configuration Characteristic

### More Information.

the same module work with manufacture's windows application 
https://www.analog.com/en/resources/reference-designs/maxrefdes104.html

zhp0406, 2024-04-24:
The Bluedroid host supports 128-bit UUIDs. It's important to note that 128-bit UUIDs are divided into base 128-bit UUIDs and non-base 128-bit UUIDs. For base 128-bit UUIDs, Bluedroid converts them into 16-bit UUIDs. The conversion rules can be found in "BLUETOOTH CORE SPECIFICATION Version 5.4 | Vol 3, Part B page 1181."

I can provide you with a patch to demonstrate the use of 128-bit UUIDs. You can test it with the following two examples:

esp-idf/examples/bluetooth/bluedroid/ble/gatt_client
esp-idf/examples/bluetooth/bluedroid/ble/gatt_server
[0001-fix-bt-bluedroid-Debug-for-ble-service-change-charac.zip](https://github.com/espressif/esp-idf/files/15091577/0001-fix-bt-bluedroid-Debug-for-ble-service-change-charac.zip)

zhp0406, 2024-04-24:
If you have any further questions regarding the use of 128-bit UUIDs, please contact us.

rajeshktym, 2024-04-24:
Hello
Thanks for your reply
I am facing another strange problem with your GPIO pins,

when nearest any appliance switch  on/off time trigger my inputs also, what
i can do for this ?

#define IGNITION_SWITCH     4
#define HANDLE_SWITCH       12
#define GPIO_INPUT_PIN_SEL  ((1ULL<<HANDLE_SWITCH) | (1ULL<<IGNITION_SWITCH
))

what i can do for this issue

Thanks
VN Rajesh Kumar
+91 9447772239

zhp0406, 2024-04-25:
@rajeshktym 
This is a new issue(GPIO). Please create another GitHub issue, and our more experienced colleagues will help you with the GPIO problem.

meakashrao, 2024-06-16:
Hello @rajeshktym were you able to get 128 bit UUID's?