espressif/esp-idf #13675: Incorrect vTaskList() output for the affinity

AxelLin, 2024-04-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1-393-gea967cf2fb

### Espressif SoC revision.

ESP32

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM-32E

### Power Supply used.

USB

### What is the expected behavior?

Below shows the vTaskList() output for IDF-5.1 v.s. 5.2.

In IDF-5.1: (v5.1.3-467-gc686e23038, good)
```
Task Name       Status  Prio    HWM     Task#   Affinity
IDLE0           R       0       924     5       0
IDLE1           R       0       916     6       1
tiT             B       18      2216    9       -1
ipc0            S       24      416     1       0
sys_evt         B       20      1784    8       0
ipc1            S       24      440     2       1
esp_timer       S       22      3352    3       0
wifi            B       23      3936    11      0
Tmr Svc         B       1       1580    7       0
```

In IDF-5.2: (v5.2.1-393-gea967cf2fb, bad)
```
Task Name       Status  Prio    HWM     Task#   Affinity
tiT             R       18      2147483647      2200    9
IDLE0           R       0       0       924     5
IDLE1           R       0       1       916     6
sys_evt         B       20      0       1832    8
ipc1            S       24      1       456     2
Tmr Svc         B       1       0       1580    7
esp_timer       S       22      0       3256    3
wifi            B       23      0       3672    11
ipc0            S       1       0       432     1
```

1. Why IDF-5.2 move the Affinity field?
   I would prefer keeping Affinity in the last field.
   i.e. **Don't change the output format (order) for no reason.**
2. In IDF-5.2, it prints **2147483647** rather than -1 if no affinity.

### What is the actual behavior?

Above log shows the problem in IDF-5.2 outut.

### Steps to reproduce.

Just call vTaskList() and verify the output.

Dazza0, 2024-04-24:
@AxelLin Looks like the ordering of the fields went form `Task Name, Status, Prio, HWM, Task#, Affinity` to `Task Name, Status, Affinity, Prio, HWM, Task#`. I'll push a commit to fix this shortly.

closed 2024-04-26 by espressif-bot

AxelLin, 2024-05-13:
Just remind the v5.3 branch still needs fix.

ESP-Marius, 2024-05-13:
@AxelLin I has been merged to our internal 5.3 branch, so it should be available the next time it gets synced to github

nopnop2002, 2024-05-27:
Sorry for the addition.
What does affinity = -1 mean?

Dazza0, 2024-05-28:
> What does affinity = -1 mean?

@nopnop2002 `-1` means no `tskNO_AFFINITY`.