espressif/esp-idf #13678: "esp_pm_dump_locks" always gives 0 for SPI

DemTerritory, 2024-04-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-3499-g636ff35b52

### Espressif SoC revision.

ESP32-S3FH4R2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3 LOLIN Mini

### Power Supply used.

External 5V

### What is the expected behavior?

Receiving data from the sensor via SPI is one of the main actions of my program, which is executed every second. And although the data volume is several tens of bytes, at least “Total_count” should not be equal to zero.

### What is the actual behavior?

[image]

### Steps to reproduce.

esp_pm_dump_locks(stream);

esp-wzh, 2024-09-26:
Sorry, I can't reproduce the issue. Which API in the SPI driver do you use to read data? Can you share a reproduction example? @DemTerritory

DemTerritory, 2024-09-27:
This table is output by the esp_pm_dump_locks() function. I use SPI Master Driver and Polling Transactions to get data from the BME280 sensor, I have not tried other APIs (Slave Drive).
I just thought that maybe this is not a bug. This can happen if pm_lock is created by esp_pm_lock_create(), but is not actually used - esp_pm_lock_acquire()

esp-wzh, 2024-09-29:
Yes, `esp_pm_lock_create` only initializes the lock's properties. The number of acquisitions and holding time of the lock will only be updated when `esp_pm_lock_acquire` is called.

closed 2024-09-29 by DemTerritory