espressif/esp-idf #13679: Wi-Fi AP+STA mode not working on esp32c6 with Wi-Fi 6 AP

alerighi, 2024-04-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

esp32c6

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

When configuring the board to activate both the AP mode and the STA mode (WIFI_MODE_APSTA) the STA connection to a Wi-Fi AP (in my case an Ubiquity Unifi U6-Pro) does initially succeed (the device gets an IP address), but when making the first network request it fails. Note that the connection to the same AP works correctly if only the STA mode is activated (WIFI_MODE_STA). 

### What is the actual behavior?

The AP connect successfully and network requests succeed.

### Steps to reproduce.

I've was able to reproduce the issue by modifying the example "softap_sta" by adding an HTTP request at the end.

```c
/*
 * SPDX-FileCopyrightText: 2023 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Unlicense OR CC0-1.0
 */
/*  WiFi softAP & station Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"

#include "esp_mac.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "esp_netif_net_stack.h"
[... 477 lines cut ...]
I (6602) wifi:<ba-del>idx:0, tid:6
W (6602) wifi:<ba-add>idx:0, ifx:0, tid:6, TAHI:0x100a200, TALO:0xe4322260, (ssn:2, win:64, cur_ssn:2), CONF:0xc0006005
I (9602) wifi:state: run -> init (22c0)
I (9612) wifi:ifidx:0, rssi:-37, nf:-97, phytype(0x3, CBW20-SGI), phymode(0x5, 11ax), max_rate:860, he:1
I (9612) wifi:max ampdu length exponent:3(65535 bytes), mmss:0(no restriction)
I (9612) wifi:pm stop, total sleep time: 0 us / 5594825 us

I (9622) wifi:<ba-del>idx:0, tid:6
I (9622) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,0>, prof:1
I (9632) WiFi Sta: Channel changed from 1 to 1
E (9632) WiFi Sta: !! STA Disconnected! Reason: 34
E (12092) esp-tls: couldn't get hostname for :google.com: getaddrinfo() returns 202, addrinfo=0x0
E (12092) transport_base: Failed to open a new connection: 32769
E (12092) HTTP_CLIENT: Connection failed, sock < 0
E (12102) HTTP: HTTP GET request failed: ESP_ERR_HTTP_CONNECT
```

### More Information.

The disconnect reason seems to be `WIFI_REASON_MISSING_ACKS`. Here a screenshot of the AP settings. As you can see it's using the default settings.

[image]

Thanks,
Alessandro

nishanth-radja, 2024-04-30:
@alerighi can I have the commit ID of the IDF on which you are seeing this issue ?

xuxiao111, 2024-04-30:
Hi, @alerighi Thanks for reporting this issue, and sorry for the late response.
From the debug log, seems like the AP can not receive the ACK frame from the DUT (esp32c6), and then the AP sends the deauthentication to DUT, letting the DUT disconnect with the AP. I have tried to reproduce it in my testbed, but it works fine. Have you used other APs to test this case?

alerighi, 2024-04-30:
Hi, 

the commit I'm using is the one corresponding to the 5.2.1 release:
```
commit a322e6bdad4b6675d4597fb2722eea2851ba88cb (HEAD, tag: v5.2.1)
```

> Have you used other APs to test this case?

I've tried with other AP that are only Wi-Fi 4 2.4GHz and it works. The AP that I'm having issues so far is only the Ubiquity that I've mentioned above. This AP has configured a network that has both the band 2.4 and 5GHz and uses the Wi-Fi 6 standard. 

I can try to reproduce the issue on another Wi-Fi AP that has both the networks, and I will get you back.

nishanth-radja, 2024-04-30:
> Hi,
> 
> the commit I'm using is the one corresponding to the 5.2.1 release:
> 
> ```
> commit a322e6bdad4b6675d4597fb2722eea2851ba88cb (HEAD, tag: v5.2.1)
> ```
> 
> > Have you used other APs to test this case?
> 
> I've tried with other AP that are only Wi-Fi 4 2.4GHz and it works. The AP that I'm having issues so far is only the Ubiquity that I've mentioned above. This AP has configured a network that has both the band 2.4 and 5GHz and uses the Wi-Fi 6 standard.
> 
> I can try to reproduce the issue on another Wi-Fi AP that has Wi-Fi 6 networks, and I will get you back.

@alerighi Thanks for the reply, Also wireless sniffer capture with  the Ubiquity AP will be helpful to pin point the issue.

alerighi, 2024-04-30:
Hi @nishanth-radja,

I've reproduced the issue and captured with Wireshark the packets. 
[wireshark-capture.zip](https://github.com/espressif/esp-idf/files/15168041/wireshark-capture.zip)

To see the traffic of the device you can use the filter: `wlan.addr == 40:4c:ca:41:83:10`

nishanth-radja, 2024-05-02:
@alerighi  Thanks for the captures,How far is the STA from the AP and also is the environment very noisy.I see that STA is not able to reply to most of the packets from AP.Also can you **disable** beam-forming on the AP and disable **RTS** protection.This will reduce the channel noise. Can you try these and send the capture if the problem persist ?

alerighi, 2024-05-02:
Hi,
the AP is near to the STA (approximately 5 meters). 

The environment is noisy since we have a lot of Wi-Fi devices, and also there are a lot of Wi-Fi networks and devices from nearby offices (that we don't have control). 

I will try what you suggested and get you back.

alerighi, 2024-05-08:
Hi,
I've tried to find the options you mentioned in the Ubiquity configuration page, but it seems that I cannot control that configurations. 

Is there something else that I can try?

nishanth-radja, 2024-05-10:
@alerighi If all the device are on channel 1 ,can you move your AP to channel 11 and try. Noise will be lesser in channel 11.Can you provide a sniffer capture on channel 11 during the association.We can see if the problem persist.

AxelLin, 2024-06-06:
> @alerighi Thanks for the captures,How far is the STA from the AP and also is the environment very noisy.I see that STA is not able to reply to most of the packets from AP.

Is it normal that STA is not able to reply to most of the packets from AP **when the AP is near to the STA** (approximately 5 meters)?
Is there any API in esp-idf can tell how noisy the environment is?
(At least, it can help to debug issues because people usually don't know how noisy the environment is.)

AxelLin, 2024-06-06:
@alerighi Any update for https://github.com/espressif/esp-idf/issues/13679#issuecomment-2104187636 ?

alerighi, 2024-06-06:
Hi,

sorry for the late reply. 

I've tested the connection also with another AP (a Microtik hAP ax²). With this sort of AP I can also try out multiple options, being a Microtik is very configurable. 

As it was with the Ubiquity AP that I was using before, on channel 1 the connection is unstable, but on this other AP sometimes has success (after multiple tries). 

By moving the AP to any other channel (I've tried 2, 6, 10) the connection has success at first try. 

Also, by setting the standard to Wi-Fi N (and not AX) the connection also has success on the first try, even on channel 1. 

I attach some logs in the various situations:

Channel 1 Wi-Fi AX (fails)
```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0x1c (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x1804
load:0x4086c410,len:0xe58
load:0x4086e610,len:0x2e24
entry 0x4086c41a
I (23) boot: ESP-IDF v5.2.1-dirty 2nd stage bootloader
I (24) boot: compile time Jun  6 2024 12:24:22
I (24) boot: chip revision: v0.0
I (27) boot.esp32c6: SPI Speed      : 80MHz
I (32) boot.esp32c6: SPI Mode       : DIO
I (36) boot.esp32c6: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=420b0020 size=278a0h (161952) map
I (126) esp_image: segment 1: paddr=000378c8 vaddr=40800000 size=00750h (  1872) load
I (127) esp_image: segment 2: paddr=00038020 vaddr=42000020 size=af7c0h (718784) map
I (280) esp_image: segment 3: paddr=000e77e8 vaddr=40800750 size=15148h ( 86344) load
I (301) esp_image: segment 4: paddr=000fc938 vaddr=408158a0 size=0358ch ( 13708) load
I (310) boot: Loaded app from partition at offset 0x10000
I (311) boot: Disabling RNG early entropy source...
I (322) cpu_start: Unicore app
W (331) clk: esp_perip_clk_init() has not been implemented yet
I (337) cpu_start: Pro cpu start user code
I (338) cpu_start: cpu freq: 160000000 Hz
I (338) cpu_start: Application information:
I (341) cpu_start: Project name:     softap_sta
I (346) cpu_start: App version:      v5.2.1-dirty
I (351) cpu_start: Compile time:     Jun  6 2024 12:24:15
I (357) cpu_start: ELF file SHA256:  6145af92e...
I (363) cpu_start: ESP-IDF:          v5.2.1-dirty
I (368) cpu_start: Min chip rev:     v0.0
I (373) cpu_start: Max chip rev:     v0.99
I (377) cpu_start: Chip rev:         v0.0
I (382) heap_init: Initializing. RAM available for dynamic allocation:
I (389) heap_init: At 4081DFE0 len 0005E630 (377 KiB): RAM
I (396) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (402) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (409) spi_flash: detected chip: generic
I (413) spi_flash: flash io: dio
W (417) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (430) sleep: Configure to isolate all GPIO pins in sleep state
I (437) sleep: Enable automatic switching of GPIO sleep configuration
I (444) coexist: coex firmware version: 77cd7f8
I (449) coexist: coexist rom version 5b8dcfa
I (454) main_task: Started on CPU0
[... 659 lines cut ...]

I (3944) wifi:pm start, type:1, aid:0x1, trans-BSSID:78:9a:18:f0:76:1f, BSSID[5]:0x1f, mbssid(max-indicator:0, index:0), he:0
I (3944) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (3954) wifi:set rx beacon pti, rx_bcn_pti: 10, bcn_timeout: 25000, mt_pti: 10, mt_time: 10000
I (3964) wifi:[ADDBA]TX addba request, tid:0, dialogtoken:1, bufsize:64, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (3974) wifi:[ADDBA]TX addba request, tid:7, dialogtoken:2, bufsize:64, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x20)
I (3984) wifi:[ADDBA]TX addba request, tid:5, dialogtoken:3, bufsize:64, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (3994) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (4004) wifi:[ADDBA]RX addba response, status:0, tid:0/tb:1(0xa1), bufsize:64, batimeout:0, txa_wnd:64
I (4014) wifi:[ADDBA]RX addba response, status:0, tid:7/tb:1(0xa1), bufsize:64, batimeout:0, txa_wnd:64
I (4024) wifi:[ADDBA]RX addba response, status:0, tid:5/tb:1(0xa1), bufsize:64, batimeout:0, txa_wnd:64
I (4034) WiFi Sta: STA Connected!!
I (5034) WiFi Sta: Got IP:192.168.88.254
I (5034) esp_netif_handlers: sta ip: 192.168.88.254, mask: 255.255.255.0, gw: 192.168.88.1
I (5034) WiFi Sta: connected to ap SSID:RETESTNET password:REDACTED
W (5044) wifi:<ba-add>idx:0, ifx:0, tid:6, TAHI:0x1001f76, TALO:0xf0189a78, (ssn:2, win:64, cur_ssn:2), CONF:0xc0006005
W (5074) wifi:<ba-add>idx:1, ifx:0, tid:0, TAHI:0x1001f76, TALO:0xf0189a78, (ssn:0, win:64, cur_ssn:0), CONF:0xc0000005
I (5264) HTTP: HTTP GET Status = 301, content_length = 219
I (6464) HTTP: HTTP GET Status = 301, content_length = 219
I (7724) HTTP: HTTP GET Status = 301, content_length = 219
I (8954) HTTP: HTTP GET Status = 301, content_length = 219
I (10184) HTTP: HTTP GET Status = 301, content_length = 219
```

Tell me if it's necessary that I do more tests to identify the problem.

AxelLin, 2024-06-07:
E (23624) WiFi Sta: !! STA Disconnected! Reason: 71 (in above log)
What is reason code 71?, I even cannot find it in wifi_err_reason_t.

xuxiao111, 2024-06-07:
@AxelLin 
[image]
The reason coed 71 was newly added in the 11ax protocol. And this reason code is carried in the deauth frame sent by the AP.
@alerighi 
Thank you very much for your response. This information is very helpful for our debugging.
From the log, it can be seen that in "Channel 1 Wi-Fi AX (fails)," the STA is connected in 11AX mode, but our channel bandwidth is 40M. However, since we are an HE 20M only device, due to hardware limitations, we cannot receive HE format packets at 40M bandwidth, which leads to the failure of the case. We will soon add restrictions in the code for this part. In your current usage scenario, could you set the AP's bandwidth to 20M in advance and then check if the problem still exists?

AxelLin, 2024-06-07:
Hi @xuxiao111 @alerighi 

I check the original descriptioon of this issue (https://github.com/espressif/esp-idf/issues/13679#issue-2258597074):
Looks like the original issue is not related to the 40M issue.

What is the expected behavior?
When configuring the board to activate both the AP mode and the STA mode (WIFI_MODE_APSTA) the STA connection to a Wi-Fi AP (in my case an Ubiquity Unifi U6-Pro) does initially succeed (the device gets an IP address), but when making the first network request it fails. Note that the connection to the same AP works correctly if only the STA mode is activated (WIFI_MODE_STA).

Which means AP+STA and STA only has different behavior when connect to the same AP.

And the disconnect reason code is different:
E (9632) WiFi Sta: !! STA Disconnected! Reason: 34 (WIFI_REASON_MISSING_ACKS)

alerighi, 2024-06-07:
Hi, 

I've tried with 20Mhz bandwidth, here is the log:

```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x1804
load:0x4086c410,len:0xe58
load:0x4086e610,len:0x2e24
entry 0x4086c41a
I (23) boot: ESP-IDF v5.2.1-dirty 2nd stage bootloader
I (24) boot: compile time Jun  6 2024 12:24:22
I (24) boot: chip revision: v0.0
I (27) boot.esp32c6: SPI Speed      : 80MHz
I (32) boot.esp32c6: SPI Mode       : DIO
I (36) boot.esp32c6: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=420b0020 size=278a0h (161952) map
I (126) esp_image: segment 1: paddr=000378c8 vaddr=40800000 size=00750h (  1872) load
I (127) esp_image: segment 2: paddr=00038020 vaddr=42000020 size=af7c0h (718784) map
I (280) esp_image: segment 3: paddr=000e77e8 vaddr=40800750 size=15148h ( 86344) load
I (301) esp_image: segment 4: paddr=000fc938 vaddr=408158a0 size=0358ch ( 13708) load
I (310) boot: Loaded app from partition at offset 0x10000
I (310) boot: Disabling RNG early entropy source...
I (322) cpu_start: Unicore app
W (331) clk: esp_perip_clk_init() has not been implemented yet
I (337) cpu_start: Pro cpu start user code
I (338) cpu_start: cpu freq: 160000000 Hz
I (338) cpu_start: Application information:
I (340) cpu_start: Project name:     softap_sta
I (346) cpu_start: App version:      v5.2.1-dirty
I (351) cpu_start: Compile time:     Jun  6 2024 12:24:15
I (357) cpu_start: ELF file SHA256:  6145af92e...
I (362) cpu_start: ESP-IDF:          v5.2.1-dirty
I (368) cpu_start: Min chip rev:     v0.0
I (373) cpu_start: Max chip rev:     v0.99
I (377) cpu_start: Chip rev:         v0.0
I (382) heap_init: Initializing. RAM available for dynamic allocation:
I (389) heap_init: At 4081DFE0 len 0005E630 (377 KiB): RAM
I (395) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (402) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (409) spi_flash: detected chip: generic
I (413) spi_flash: flash io: dio
W (416) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (430) sleep: Configure to isolate all GPIO pins in sleep state
I (436) sleep: Enable automatic switching of GPIO sleep configuration
I (443) coexist: coex firmware version: 77cd7f8
I (449) coexist: coexist rom version 5b8dcfa
I (454) main_task: Started on CPU0
I (454) main_task: Calling app_main()
I (464) pp: pp rom version: 5b8dcfa
I (474) net80211: net80211 rom version: 5b8dcfa
I (484) wifi:wifi driver task: 40826d1c, prio:23, stack:6656, core=0
I (484) wifi:wifi firmware version: a9f5b59
I (484) wifi:wifi certification version: v7.0
I (484) wifi:config NVS flash: enabled
I (484) wifi:config nano formating: disabled
I (494) wifi:mac_version:HAL_MAC_ESP32AX_761,ut_version:N
I (494) wifi:Init data frame dynamic rx buffer num: 32
I (504) wifi:Init static rx mgmt buffer num: 5
[... 84 lines cut ...]
I (3964) WiFi Sta: STA Connected!!
I (4964) WiFi Sta: Got IP:192.168.88.254
I (4964) esp_netif_handlers: sta ip: 192.168.88.254, mask: 255.255.255.0, gw: 192.168.88.1
I (4964) WiFi Sta: connected to ap SSID:RETESTNET password:REDACTED
W (4974) wifi:<ba-add>idx:0, ifx:0, tid:6, TAHI:0x1001f76, TALO:0xf0189a78, (ssn:2, win:64, cur_ssn:2), CONF:0xc0006005
I (5464) wifi:<ba-del>idx:0, tid:6
W (5464) wifi:<ba-add>idx:0, ifx:0, tid:6, TAHI:0x1001f76, TALO:0xf0189a78, (ssn:3, win:64, cur_ssn:3), CONF:0xc0006005
W (5794) wifi:<ba-add>idx:1, ifx:0, tid:0, TAHI:0x1001f76, TALO:0xf0189a78, (ssn:0, win:64, cur_ssn:0), CONF:0xc0000005
W (17074) HTTP_CLIENT: Connection timed out before data was ready!
E (17074) HTTP: HTTP GET request failed: ESP_ERR_HTTP_EAGAIN
I (18274) wifi:state: run -> init (18c0)
I (18274) wifi:ifidx:0, rssi:-61, nf:-97, phytype(0x3, CBW20-SGI), phymode(0x5, 11ax), max_rate:860, he:1
I (18274) wifi:max ampdu length exponent:3(65535 bytes), mmss:0(no restriction)
I (18284) wifi:pm stop, total sleep time: 0 us / 14404824 us

I (18294) wifi:<ba-del>idx:1, tid:0
I (18294) wifi:<ba-del>idx:0, tid:6
I (18294) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,0>, prof:1
I (18304) WiFi Sta: Channel changed from 1 to 1
E (18304) WiFi Sta: !! STA Disconnected! Reason: 24
```

> When configuring the board to activate both the AP mode and the STA mode (WIFI_MODE_APSTA) the STA connection to a Wi-Fi AP (in my case an Ubiquity Unifi U6-Pro) does initially succeed (the device [...]

That is correct, with the ubiquity AP this is the behavior that I'm observing.

xuxiao111, 2024-06-07:
Hi @alerighi,
From the debug log, I can see this "wifi:new**:<1,1>**, old:<1,1>, ap:<1,1>, sta:<1,0>, prof:1", and this means that the device bandwith still unde 40M. In the log where it worked previously, the log is "wifi:new:**<2,0>**, old:<1,1>, ap:<2,1>, sta:<2,0>, prof:1".
<1,1>: the first number is channel number 1, the second number means the snd channel, 0 means second channel none, 1 means second channel above.
If it's convenient, could you upload your modified code?

AxelLin, 2024-06-07:
> If it's convenient, could you upload your modified code?

It looks like the firmware is the same because "Compile time" and "ELF file SHA256" are the same.
Maybe this discovers another issue.

xuxiao111, 2024-06-07:
Hi @alerighi,
I apologize for the confusion. What I meant to say is to modify the bandwidth of the ESP32C6 softAP instead of the AP.

alerighi, 2024-06-07:
Hi @xuxiao111,

by setting the channel as you suggested now the connection works flawlessly both with the Ubiquity and with the Microtik AP. Here is the log, connecting to the Ubiuity AP:

```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x1804
load:0x4086c410,len:0xe58
load:0x4086e610,len:0x2e24
entry 0x4086c41a
I (23) boot: ESP-IDF v5.2.1-dirty 2nd stage bootloader
I (24) boot: compile time Jun  7 2024 12:31:49
I (24) boot: chip revision: v0.0
I (27) boot.esp32c6: SPI Speed      : 80MHz
I (32) boot.esp32c6: SPI Mode       : DIO
I (36) boot.esp32c6: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=420b0020 size=27900h (162048) map
I (126) esp_image: segment 1: paddr=00037928 vaddr=40800000 size=006f0h (  1776) load
I (127) esp_image: segment 2: paddr=00038020 vaddr=42000020 size=af944h (719172) map
I (280) esp_image: segment 3: paddr=000e796c vaddr=408006f0 size=151a8h ( 86440) load
I (301) esp_image: segment 4: paddr=000fcb1c vaddr=408158a0 size=0358ch ( 13708) load
I (310) boot: Loaded app from partition at offset 0x10000
I (311) boot: Disabling RNG early entropy source...
I (322) cpu_start: Unicore app
W (331) clk: esp_perip_clk_init() has not been implemented yet
I (337) cpu_start: Pro cpu start user code
I (338) cpu_start: cpu freq: 160000000 Hz
I (338) cpu_start: Application information:
I (341) cpu_start: Project name:     softap_sta
I (346) cpu_start: App version:      v5.2.1-dirty
I (351) cpu_start: Compile time:     Jun  7 2024 12:36:15
I (357) cpu_start: ELF file SHA256:  6662b91ac...
I (363) cpu_start: ESP-IDF:          v5.2.1-dirty
I (368) cpu_start: Min chip rev:     v0.0
I (373) cpu_start: Max chip rev:     v0.99 
I (377) cpu_start: Chip rev:         v0.0
I (382) heap_init: Initializing. RAM available for dynamic allocation:
I (389) heap_init: At 4081DFE0 len 0005E630 (377 KiB): RAM
I (396) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (402) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (409) spi_flash: detected chip: generic
I (413) spi_flash: flash io: dio
W (417) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (430) sleep: Configure to isolate all GPIO pins in sleep state
I (437) sleep: Enable automatic switching of GPIO sleep configuration
I (444) coexist: coex firmware version: 77cd7f8
I (449) coexist: coexist rom version 5b8dcfa
I (454) main_task: Started on CPU0
I (454) main_task: Calling app_main()
I (474) pp: pp rom version: 5b8dcfa
I (474) net80211: net80211 rom version: 5b8dcfa
I (484) wifi:wifi driver task: 40826da0, prio:23, stack:6656, core=0
I (484) wifi:wifi firmware version: a9f5b59
I (484) wifi:wifi certification version: v7.0
I (484) wifi:config NVS flash: enabled
I (484) wifi:config nano formating: disabled
I (494) wifi:mac_version:HAL_MAC_ESP32AX_761,ut_version:N
I (494) wifi:Init data frame dynamic rx buffer num: 32
I (504) wifi:Init static rx mgmt buffer num: 5
[... 72 lines cut ...]
I (4184) wifi:cipher(pairwise:0x3, group:0x3), pmf:0, security:WPA2-PSK, phy:11ax, rssi:-48
I (4204) wifi:pm start, type: 1, itwt_start:0

I (4204) wifi:pm start, type:1, aid:0x16, trans-BSSID:60:22:32:e4:00:a2, BSSID[5]:0xa2, mbssid(max-indicator:0, index:0), he:1
I (4204) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (4214) wifi:set rx beacon pti, rx_bcn_pti: 10, bcn_timeout: 25000, mt_pti: 10, mt_time: 10000
I (4224) wifi:[ADDBA]TX addba request, tid:0, dialogtoken:1, bufsize:32, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (4234) wifi:[ADDBA]TX addba request, tid:7, dialogtoken:2, bufsize:32, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x20)
I (4244) wifi:[ADDBA]TX addba request, tid:5, dialogtoken:3, bufsize:32, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (4254) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (4264) wifi:[ADDBA]RX addba response, status:0, tid:0/tb:1(0xa1), bufsize:32, batimeout:0, txa_wnd:32
I (4274) wifi:[ADDBA]RX addba response, status:0, tid:7/tb:1(0xa1), bufsize:32, batimeout:0, txa_wnd:32
I (4274) wifi:[ADDBA]RX addba response, status:0, tid:5/tb:1(0xa1), bufsize:32, batimeout:0, txa_wnd:32
I (4284) WiFi Sta: STA Connected!!
W (4304) wifi:<ba-add>idx:0, ifx:0, tid:0, TAHI:0x100a200, TALO:0xe4322260, (ssn:0, win:64, cur_ssn:0), CONF:0xc0000005
I (5294) WiFi Sta: Got IP:10.17.100.225
I (5294) esp_netif_handlers: sta ip: 10.17.100.225, mask: 255.255.0.0, gw: 10.17.0.1
I (5294) WiFi Sta: connected to ap SSID:REDACTED password:REDACTED
W (5304) wifi:<ba-add>idx:1, ifx:0, tid:6, TAHI:0x100a200, TALO:0xe4322260, (ssn:2, win:64, cur_ssn:2), CONF:0xc0006005
I (5454) HTTP: HTTP GET Status = 301, content_length = 219
```

Thanks a lot for the support!

If you don't have further questions, for me we can consider this issue as closed.

xuxiao111, 2024-06-07:
Hi @alerighi,
Thank you very much for helping us identify this issue. If there are no further questions, we can close this issue for now. For the aforementioned problem, we need some time to make internal code modifications and conduct testing. Once completed, we will provide the commit ID here.

alerighi, 2024-06-10:
Thanks! We can close it, this solved the issue in our product.

Maybe you can update the documentation/example code till you fix it in the IDF, as a temporary solution?

Thanks again for the support!

closed 2024-06-10 by alerighi